OpenEtherCATsociety/SOEM #368: Advices about the code written so far with Xenomai and DC

Davidinos, 2020-02-12:
Goodmorning,
I write you to have your opinion about the code reported below. I'm using a Beaglebone black (1GHz single core) with Xenomai Cobalt patch and DC enabled at 1ms.
Do you find anything wrong? I ask you because with our tests we see that sometimes it loses the sync.. Besides I wonder if I can improve the jitter
In attachment the wireshark result with a cycle time 1ms.
[1msCycleTime.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4191107/1msCycleTime.zip)

Thank you.

`/** \file
 * \brief Example code for Simple Open EtherCAT master
 *
 * Usage : red_test [ifname1] [ifname2] [cycletime]
 * ifname is NIC interface, f.e. eth0
 * cycletime in us, f.e. 500
 *
 * This is a redundancy test.
 *
 * (c)Arthur Ketels 2008
 */

#include <stdio.h>
#include <stdbool.h>
#include <stdlib.h>
#include <sys/time.h>
#include <unistd.h>
#include <sched.h>
#include <string.h>
#include <time.h>
#include <pthread.h>
#include <math.h>

#include <sem.h>
#include <signal.h>
#include <sys/mman.h>
#include <alchemy/task.h>
#include <alchemy/timer.h>
#include <alchemy/sem.h>
#include <boilerplate/trace.h>
#include <xenomai/init.h>

//#include "ethercat.h"
#include "includeSOEM/ethercat.h"

#define NSEC_PER_SEC 1000000000
#define EC_TIMEOUTMON 500

struct sched_param schedp;
char IOmap[4096];
RT_TASK thread1, thread2;
RT_SEM sem_desc;
struct timeval tv, t1, t2;
int dorun = 0;
int deltat, tmax = 0;
int64 toff, gl_delta;
int DCdiff;
int os;
uint8 ob;
uint16 ob2;
uint8 *digout = 0;
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
uint8 currentgroup = 0;

int64 maxDelta = 0;

#define SYNC0_SIGNAL_PERIOD_NS 4000000

#define stack64k (64 * 1024)
[... 346 lines cut ...]

        // create thread to update PDOs content
        rt_task_create(&thread1, "SOEM_demo_task", stack64k * 2, 90, 0 );
        // create thread to handle slave error handling in OP
        rt_task_create(&thread2, "SOEM_check_task", stack64k * 4, 10, 0 );
        // Ask Xenomai to warn us upon switches to secondary mode.
        //rt_task_set_mode(0, T_WARNSW, NULL);
        // start cyclic rt_thread
        rt_task_start(&thread1, &ecatthread, (void*) &ctime);
        rt_task_start(&thread2, &ecatcheck, (void*) NULL);

        /* start acyclic part */
        redtest(argv[1],argv[2]);
        //printf("\nType CTRL-C to end this program\n\n" );
        //pause();
   }
   else
   {
      printf("Usage: red_test ifname1 ifname2 cycletime\nifname = eth0 for example\ncycletime in us\n");
   }

   printf("End program\n");

   return (0);
}`

ArthurKetels, 2020-02-13:
Thanks for the info. But actually you ask a xenomai question. There are examples where SOEM runs on bare metal hardware and has maximum jitter below 1us. The issue is with your hardware and OS support. No matter how clever xenomai is written, it can not do much to improve basic design flaws in hardware and drivers. Coding for real time is fundamentally different from what is available in common OS concepts.

Off my soapbox now. What you need to do is instrument your code to find the timing issues. Log your PDO thread timing in relation to the reference slave clock. When possible also log the different time paths from starting the cycle to sending data, to receiving data. The more fine grained the more informative. Store the log in histogram format, that way you can keep it in memory and still catch millions of events.

Davidinos, 2020-02-14:
Hello Arthur, thank you for your answer. 
Your point is very clear and probably we'll move to bare metal implementation to reach our goal.
I have one last question, the only modification that I applied to xenomai implementation regards the osal.c and osal.h since the time base functions are different. None other files needs to be changed right ? (consider we didn't change the nic..).
Thank you for your help.

ArthurKetels, 2020-02-15:
Correct. That is the whole point of osal (operating system abstraction layer). Each target need to implement the optimum OS version of the timing and threading primitives. SOEM is not critical about the specific performance of the implementation, but off course the total performance will suffer from an inefficient implementation.

Davidinos, 2020-02-17:
Thank you very much Arthur.

closed 2020-02-17 by Davidinos