OpenEtherCATsociety/SOEM #369: PDO mapping

KhloodAmin, 2020-02-16:
Hello , i have a basic question about EtherCAT slave , How i make sure that SOEM has made the right mapping for PDOS ? 
Here is the output from slaveInfo - sdo - map 
[image]
[image]
[image]
[image]
[image]

nakarlsson, 2020-02-16:
It looks OK, if you can enter OP it also is a proof it is OK.

KhloodAmin, 2020-02-17:
thank you for your response , i had another question regarding to this point , in the screens above it's mentioned that outputs are mapped to the address 0x000000 and inputs 0x0004 , when i run the simple test with wireshark the captured packets were logical read write from address 0x000000 and there is no mention to the input addresses, how ethercat get the inputs data ? 
[packets.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4213167/packets.txt)

nakarlsson, 2020-02-17:
Its in the same read/write frame, hence the name LRW that combine read/write, the length of the data should be 8 and start at 0x000000.

KhloodAmin, 2020-02-17:
but inputs are mapped to 0x0004 according to - map output so to read inputs state LRD command must be executed at the address 0x0004 , am i wrong ? 
[image]

nakarlsson, 2020-02-17:
You can read inputs with LRD at 0x0004 with length 4

or

You can perform a read & write action with LRW at 0x0000 with length 8

Both will update the inputs

KhloodAmin, 2020-02-18:
got it , thank you

closed 2020-02-18 by KhloodAmin