OpenEtherCATsociety/SOEM #371: Custom PDO mapping for elmo twitters

muenchenstrasse, 2020-02-22:
Hello,

I am trying to create my own PDO mapping using SOEM. I have experience in using TwinCAT to do that and i must say, using twincat to do pdo mapping was very easy. You just need to click on the list of PDOs you want to map and the twincat does everything for you.

So usually my txpdo would consist of 0x1a04 and 0x1a13. I saw that from some other posts that 0x1a02 is fixed on sm2. Is there anyway that i can change it? I mean if twincat can change it, we can as well. :)

I have went through the tutorials, however i did not manage to get information out of it.

A short and brief example could help me a long mile! Thank you!

Mav

nakarlsson, 2020-02-22:
Search for Elmo in Issues and you’ll find examples.

muenchenstrasse, 2020-02-22:
> Search for Elmo in Issues and you’ll find examples.

Hello nakarlsson,
I did exactly that before i started the thread. Unfortunately i do not know which one to start because they are not want i would want.

nakarlsson, 2020-02-22:
https://github.com/OpenEtherCATsociety/SOEM/issues/300 re-configure a drive.

muenchenstrasse, 2020-02-25:
I have tried the example that you sent. In fact, i am trying to change the slaveinfo from the SOEM to see if my PDO mapping is successful.

void slaveinfo(char *ifname)
{
    int cnt, i, j, nSM;
    uint16 ssigen;
    int expectedWKC;

    //int32 ob2;int os;

    printf("Starting slaveinfo\n");

    /* initialise SOEM, bind socket to ifname */
    if (ec_init(ifname))
    {

        printf("ec_init on %s succeeded.\n",ifname);
        /* find and auto-config slaves */
        if ( ec_config_init(FALSE) > 0 )
        {
            for(i=1; i<=ec_slavecount;i++){
                printf("Slave %d has CA? %s\n", i,ec_slave[i].CoEdetails & ECT_COEDET_SDOCA ? "true":"false");
                ec_slave[i].CoEdetails ^= ECT_COEDET_SDOCA;
                printf("New slave %d has CA? %s\n",i, ec_slave[i].CoEdetails & ECT_COEDET_SDOCA ? "true":"false");
            }

            for( int cnt = 1 ; cnt <= ec_slavecount ; cnt++)
            {

                int ret = 0;
                int os;

                uint32 idx_rxpdo = 0x16010001;
                os=sizeof (idx_rxpdo);
                ret += ec_SDOwrite(cnt, 0x1c12, 0, TRUE, os, &idx_rxpdo, EC_TIMEOUTRXM);
                printf("RxPDO mapping object index %d = %04x ret=%d\n", cnt, idx_rxpdo, ret);

                uint32 idx_txpdo = 0x1a020001;
                os=sizeof (idx_txpdo);
                ret += ec_SDOwrite(cnt, 0x1c13, 0, TRUE, os, &idx_txpdo, EC_TIMEOUTRXM);
                printf("TxPDO mapping object index %d = %04x ret=%d\n", cnt, idx_txpdo, ret);

            }
            ec_config_map(&IOmap);

            ec_configdc();
            while(EcatError) printf("%s", ec_elist2string());
            printf("%d slaves found and configured.\n",ec_slavecount);
            expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
            printf("Calculated workcounter %d\n", expectedWKC);
            /* wait for all slaves to reach SAFE_OP state */
            ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 3);
            if (ec_slave[0].state != EC_STATE_SAFE_OP )
            {
                printf("Not all slaves reached safe operational state.\n");
                ec_readstate();
                for(i = 1; i<=ec_slavecount ; i++)
                {
                    if(ec_slave[i].state != EC_STATE_SAFE_OP)
                    {
                        printf("Slave %d State=%2x StatusCode=%4x : %s\n",
                               i, ec_slave[i].state, ec_slave[i].ALstatuscode, ec_ALstatuscode2string(ec_slave[i].ALstatuscode));
                    }
                }
            }

            ec_readstate();
            for( cnt = 1 ; cnt <= ec_slavecount ; cnt++)
            {
                printf("\nSlave:%d\n Name:%s\n Output size: %dbits\n Input size: %dbits\n State: %d\n Delay: %d[ns]\n Has DC: %d\n",
[... 99 lines cut ...]
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L: 256 F:00010064 Type:3
 SM3 A:1c00 L: 256 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
nSM value = 4
rdl value = 1
tSM value = 3
  SM2 outputs
     addr b   index: sub bitl data_type    name
rdl value = 1
tSM value = 4
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program

Maybe I am doing something wrong but could you help me with this?

Mavarick

closed 2020-03-03 by muenchenstrasse