OpenEtherCATsociety/SOEM #373: OS freezes when using SOEM in LUBUNTU

dabhijeet, 2020-03-02:
Hi,

When we run slaveinfo in Lubuntu for the first time, we get the information of the slaves properly but if we run it second time then the OS freezes. The same program runs perfectly in the normal ubuntu. Even when we run redtest to run the motor we don't get any issue in normal ubuntu but if we run it in Lubuntu then for the first time it will show segmentation fault and on the second time the OS freezes. There is no error while building soem in Lubuntu.

nakarlsson, 2020-03-02:
I'm not familiar with LUBUNTU, do you get a stack trace?

dabhijeet, 2020-03-05:
We installed the Lubuntu on another system and it didn't show any problem...
It seems like it is not the Lubuntu problem..

dabhijeet, 2020-03-05:
There is one thing that we have observed that if we turn off the wifi then the slaveinfo won't freeze the OS.  Although the motor is not running when using our modified redtest. Seems like some network driver related issue.

dabhijeet, 2020-03-19:
Hi, we found out the reason why the ethernet port was not working properly.

If the ethernet port is not working well in Linux or if it is freezing the OS then the reason for this is

The default OS Selection of the BIOS is Windows. This setting is suitable for Windows systems. If the system used is LINUX, set it to Intel Linux. I have seen that if OS Selection is in WINDOWS, the LAN of Ubuntu system cannot be connected, but the OS Selection setting is normal on Intel Linux

nakarlsson, 2020-03-19:
good news! I'll close this then

closed 2020-03-19 by nakarlsson