OpenEtherCATsociety/SOEM #377: SOEM with ELMO Platinum Twitters

muenchenstrasse, 2020-03-03:
Hello,
i am having problems with the complete access of the ELMO platinum twitters. I am trying to read EC_Slave.output and EC_Slave.input of the drives.
So i did the slaveinfo and this is what i got:

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s31f6 succeeded.
7 slaves found and configured.
Calculated workcounter 21

Slave:1
 Name:ModuleSlotsDrive
 Output size: 112bits
 Input size: 160bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 0000009a ID: 01100002 Rev: 00120025
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  14 F:00010064 Type:3
 SM3 A:1c00 L:  20 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  14 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000062 Ll:  20 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0

So i got this after doing mapping:

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s31f6 succeeded.
7 slaves found and configured.
Calculated workcounter 21

Slave:1
 Name:ModuleSlotsDrive
 Output size: 112bits
 Input size: 160bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 0000009a ID: 01100002 Rev: 00120025
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  14 F:00010064 Type:3
 SM3 A:1c00 L:  20 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  14 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000062 Ll:  20 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
  [0x0002.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0006.0] 0x60B1:0x00 0x20 INTEGER32    Velocity offset
  [0x000A.0] 0x60FE:0x01 0x20 UNSIGNED32   SubIndex 001
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0062.0] 0x6041:0x00 0x10 UNSIGNED16   Statusword
  [0x0064.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x0068.0] 0x6069:0x00 0x20 INTEGER32    Velocity sensor actual value
  [0x006C.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
  [0x006E.0] 0x60F4:0x00 0x20 INTEGER32    Following error actual value
  [0x0072.0] 0x60FD:0x00 0x20 UNSIGNED32   Digital inputs

So the most obvious step is to create structs like the following:

typedef struct
    {
        uint16 Statusword;
        int32 PositionValue;
        int32 VelocityValue;
        int16 TorqueValue;
        int32 PositionErr;
        uint32 DigitalInputs;
    } _TxPDO;

    typedef struct
    {
        uint16 Controlword;
        int32 TargetPosition;
        int32 VelocityOffset;
        uint32 DigitalOutputs;
    } _RxPDO;

    _RxPDO* m_pRxPDO;
    _TxPDO* m_pTxPDO;

.. and attach them after the ethercat stack is in operational state:

m_pTxPDO = (CiA402::_TxPDO*) ec_slave[m_nDeviceNumber].inputs;
m_pRxPDO = (CiA402::_RxPDO*) ec_slave[m_nDeviceNumber].outputs;

So this is my problem: both my m_pTxPDO pointer and m_pRxPDO are empty. Does anyone has experience which such kind of problems? Be aware that I got a gold twitter/ whistle to run using this method.

Mav

nakarlsson, 2020-03-04:
What do you mean by empty?

Are the inputs/outputs pointers NULL?
Are the content at inputs/outputs all 0s?
How does the data look like in Wireshark?
If you get data and check the memory, does it seem shifted? maybe caused by an alignment issue? You might need to pack the structures.

nakarlsson, 2020-03-04:
It is an TI ESC and require overlapping IOmapping

closed 2020-03-04 by nakarlsson