OpenEtherCATsociety/SOEM #378: The SOEM library usually initializes unsuccessfully, fails to map the PDOs.

alexunder, 2020-03-04:
I used the code below to "initialize" the SOEM, and sometimes fail to map the PDOs. It usually prints "Error:Could not set EC_STATE_SAFE_OP.", then go to the false case, and the failed case is very time-consuming, and the successful case is fast. Any experts can let eliminate the failed case? We used Etherlab before, and it worked well without errors. Thanks!

```c
bool soem_init(char *ifname)
{
    /*  
    int i, j, oloop, iloop, chk;
    needlf = FALSE;
    inOP = FALSE;*/

    /* initialise SOEM, bind socket to ifname */
    if (!ec_init(ifname)) {
        SOEM_ERROR("Could not initialize ethercat driver.\n");
        return false;
    }   

    if ( ec_config_init(FALSE) <= 0 ) { 
        SOEM_ERROR("No slaves found on %s\n", ifname);
        return false;
    }   

    SOEM_INFO("SOEM found and configured %d slaves\n", ec_slavecount);

    unsigned int map_size = ec_slave[0].Obytes + ec_slave[0].Ibytes;
    SOEM_INFO("The actual mapped size:%d from ec_slave.\n", map_size);

    int iomap_size = ec_config_map(IOmap);
    SOEM_INFO("The actual mapped size:%d from ec_config_map.\n", iomap_size);

    ec_configdc();

    // '0' here addresses all slaves
    if (ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE*4) != EC_STATE_SAFE_OP) {
        SOEM_ERROR("Could not set EC_STATE_SAFE_OP.\n");
        return false;
    }   

    /*  
       This section attempts to bring all slaves to operational status. It does so
       by attempting to set the status of all slaves (ec_slave[0]) to operational,
       then proceeding through 40 send/recieve cycles each waiting up to 50 ms for a
       response about the status.
     */
    ec_slave[0].state = EC_STATE_OPERATIONAL;
    ec_send_processdata();
    ec_receive_processdata(EC_TIMEOUTRET);

    ec_writestate(0);
    int chk = 40;
    do {
        ec_send_processdata();
        ec_receive_processdata(EC_TIMEOUTRET);
        ec_statecheck(0, EC_STATE_OPERATIONAL, 50000); // 50 ms wait for state check
    } while (chk-- && (ec_slave[0].state != EC_STATE_OPERATIONAL));

    if(ec_statecheck(0,EC_STATE_OPERATIONAL, EC_TIMEOUTSTATE) != EC_STATE_OPERATIONAL) {
        SOEM_ERROR("OPERATIONAL state not set, exiting\n");
        return false;
    }

    SOEM_INFO("Finished configuration successfully");
    return true;
}
```

ArthurKetels, 2020-03-04:
It would be nice to send two wireshark captures. One OK, and one failing. I need the whole capture, from start to going to OP state. And please try it on a minimal system with the minimum number of slaves that still is showing the issue.

alexunder, 2020-03-05:
[pcap_failed.tar.gz](https://github.com/OpenEtherCATsociety/SOEM/files/4290800/pcap_failed.tar.gz)
[pcap_ok.tar.gz](https://github.com/OpenEtherCATsociety/SOEM/files/4290802/pcap_ok.tar.gz)

I have to say sorry that we have 12 ethercat nodes , and due to the hardware issues, we can't separate them easily. Could you please check the current pcap files? Thanks very much!

ArthurKetels, 2020-03-05:
The posted wireshark files are a mess. There are two etherCAT master instances fighting against each other. There is one with MAC:dc:a6:32:60:ca:1a and one with MAC:01:01:01:01:01:01. Only the last one is correct. Do you have EtherLAB still running in the background?

Also the OK capture is actually not OK. The capture stops at pre-OP. So there is no evidence it made it to at least safe-OP.

My suggestion is to only run the standard slaveinfo test program, and make a capture of that.

alexunder, 2020-03-06:
Yes, we also configured etherlab library on the raspberry board, so I have to remove the etherlab, then do the test?

alexunder, 2020-03-06:
After stopping etherlab service, the soem library works very well. Thanks!

nakarlsson, 2020-03-08:
Fixed? Please close the issue

alexunder, 2020-03-09:
OK! Thanks very much!

closed 2020-03-09 by alexunder