OpenEtherCATsociety/SOEM #379: Uploading TwinCAT 3 configuration via EEPROM results in Invalid input configuration

jtsdana, 2020-03-06:
I have been working on mapping PDO's with a Festo CMMT-ST motordrive. 
I cannot make the any of the example codes work out of the box, it is as if the EEPROM is invalid when booted up. 
So I tried setting it up with TwinCAT3  and it supposedly works, but the configuration is not persistent after a reboot of the drive.
So I saved the hex file from TwinCAT3 and used EEPROM tool to upload the supposedly valid configuration file, but this results in the Invalid input configuration.
I furthermore tried to use ([siitool](https://github.com/synapticon/siitool) to generate a binary image of from the ESI, and upload this image with eepromtool. This lets me run the slave-info script, but no PDO's are found ( even though they should be configured by default.)
I can enable the drive through CiA402 when using the generated binary image from siitool. 

Am I dealing with a non-conforming slave, or is my approach just totally wrong?

nakarlsson, 2020-03-06:
I think your approach is wrong but lets see, send a esi, wireshark and output of slave_info -map.
Comment, if you don't know what you're doing you shouldn't write/fiddle with the EEPROM.

jtsdana, 2020-03-06:
> I think your approach is wrong but lets see, send a esi, wireshark and output of slave_info -map.
> Comment, if you don't know what you're doing you shouldn't write/fiddle with the EEPROM.

First of all, thank you for responding! 

Here is the ESI file
[Festo-CMMT-ST-CiA402-20190702.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4298409/Festo-CMMT-ST-CiA402-20190702.zip)

Wireshark pcap
[slave_info.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4298404/slave_info.zip)

slave_info - map 
```SOEM (Simple Open EtherCAT Master)                                                                                                                                                                                                                                                        
Slaveinfo                                                                                                                                                                                                                                                                                 
Starting slaveinfo                                                                                                                                                                                                                                                                        
ec_init on enx24f5a28cffd7 succeeded.                                                                                                                                                                                                                                                     
1 slaves found and configured.                                                                                                                                                                                                                                                            
Calculated workcounter 3                                                                                                                                                                                                                                                                  
Not all slaves reached safe operational state.                                                                                                                                                                                                                                            
Slave 1 State=12 StatusCode=  1e : Invalid input configuration                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                                          
Slave:1                                                                                                                                                                                                                                                                                   
 Name:CMMT-ST                                                                                                                                                                                                                                                                             
 Output size: 192bits                                                                                                                                                                                                                                                                     
 Input size: 112bits                                                                                                                                                                                                                                                                      
 State: 18                                                                                                                                                                                                                                                                                
 Delay: 0[ns]                                                                                                                                                                                                                                                                             
 Has DC: 1                                                                                                                                                                                                                                                                                
 DCParentport:0                                                                                                                                                                                                                                                                           
[... cut ...]
 FMMU0 Ls:00000000 Ll:  24 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000018 Ll:  14 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program

```

jtsdana, 2020-03-06:
Here is the EEPROM readout from the SII 
[from_fresh_reboot.bin.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4298456/from_fresh_reboot.bin.zip)

nakarlsson, 2020-03-06:
Looks like the online ObjectDictionary is not populated by default as stated in the ESI

Given 1600 and 1A00 is populated correctly check the ESI and populate 
1c12 with 1600
1c13 with 1a00

How to write SDOs in PREOP (important) can be seen here https://github.com/hefloryd/SOEM/commit/22d8f5d

jtsdana, 2020-03-06:
> Looks like the online ObjectDictionary is not populated by default as stated in the ESI
> 
> Given 1600 and 1A00 is populated correctly check the ESI and populate
> 1c12 with 1600
> 1c13 with 1a00
> 
> How to write SDOs in PREOP (important) can be seen here [hefloryd@22d8f5d](https://github.com/hefloryd/SOEM/commit/22d8f5d)

I tried to to set 
1c12 with 1600 and 1x13 with 1a00

The controller reports an EtherCAT related error at
```usedmem = ec_config_map(&IOmap);```
But it still doesn't seem to work.

```
#include <stdbool.h>
#include <stdio.h>
#include <string.h>
#include "ethercat.h"
#include "osal.h"
#include "oshw.h"
// EtherCAT variables:
char IOmap[4096];
int wkc = 0;
unsigned int usedmem;
static int moog_write8(uint16 slave, uint16 index, uint8 subindex,
                       uint8 value) {
  int wkc;

  wkc = ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value,
                    EC_TIMEOUTRXM);
  return wkc;
}
static int moog_write16(uint16 slave, uint16 index, uint8 subindex,
                        uint16 value) {
  int wkc;

  wkc = ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value,
                    EC_TIMEOUTRXM);
  return wkc;
}

static int Map_PDO_CMMT(uint16_t slave) {
    wkc += moog_write8(slave, 0x1C12, 0, 0);
    wkc += moog_write8(slave, 0x1C13, 0, 0);

  wkc += moog_write16(slave, 0x1C12, 1, 0x1600);
    wkc += moog_write16(slave, 0x1C12, 0, 1);

  wkc += moog_write16(slave, 0x1C13, 1, 0x1A00);
    wkc += moog_write16(slave, 0x1C13, 0, 1);
  printf("slave_id: %x", slave);
  return 0;
}

int nic_setup(char* ifname) {
  /* initialise SOEM, bind socket to ifname */
  if (!ec_init(ifname)) {
    return 1;
  }

  /* find and auto-config slaves */
  if (!ec_config_init(FALSE)) {
    return 1;
  }
  printf("%d slaves found and configured.\n", ec_slavecount);
  ec_slave[1].PO2SOconfig = Map_PDO_CMMT;

 ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE);

  usedmem = ec_config_map(&IOmap);
  if (!(usedmem <= sizeof(IOmap))) {
    printf("IO Map not big enough");
  }

  printf("Slaves mapped and configured.\n");

  ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE * 4);

  // send one valid process data to make outputs in slaves happy
  //  ec_send_processdata();
  // wkc = ec_receive_processdata(EC_TIMEOUTRET);
  // wait for all slaves to reach OP state
  // ec_slave[0].state = EC_STATE_OPERATIONAL;
  return 0;
}

int main() {
  // Setup EtherCAT interface:
  if (nic_setup("enx24f5a28cffd7") == 1) {
    // Setup encountered and error and cannot continue.
    return 1;
  }
  // Test remapping:
  // os=sizeof()

  return 0;
}
```

Output of slaveinfo:
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enx24f5a28cffd7 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:CMMT-ST
 Output size: 192bits
 Input size: 112bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000001d ID: 007b5a25 Rev: 00000004
 SM0 A:1200 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1000 L:  24 F:00010064 Type:3
 SM3 A:1100 L:  14 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  24 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000018 Ll:  14 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program
```

nakarlsson, 2020-03-06:
You should end with setting subindex to 1, try that and send a wirshark

'''
    wkc += moog_write8(slave, 0x1C12, 0, 1);
    wkc += moog_write8(slave, 0x1C13, 0, 1);
'''

jtsdana, 2020-03-06:
> You should end with setting subindex to 1, try that and send a wirshark
> 
> '''
> wkc += moog_write8(slave, 0x1C12, 0, 1);
> wkc += moog_write8(slave, 0x1C13, 0, 1);
> '''

I did it the first time, and then ran slave_info, and the correct PDO's showed up without error. I then restarted the drive and tried again, and sadly it didn't work. 
Tried with:
```
static int Map_PDO_CMMT(uint16_t slave) {
    wkc += moog_write8(slave, 0x1C12, 0, 0);
    wkc += moog_write8(slave, 0x1C13, 0, 0);

    wkc += moog_write16(slave, 0x1C12, 1, 0x1600);
    wkc += moog_write16(slave, 0x1C13, 1, 0x1A00);

    wkc += moog_write16(slave, 0x1C12, 0, 1);
    wkc += moog_write16(slave, 0x1C13, 0, 1);

  printf("slave_id: %x", slave);
  return 0;
}
```

Here is the wireshark capture:
[slave_info_reboot.pcapng.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4298771/slave_info_reboot.pcapng.zip)

nakarlsson, 2020-03-06:
moog_write16 should be moog_write8  ,, index 0 is a byte

nakarlsson, 2020-03-06:
'''
static int Map_PDO_CMMT(uint16_t slave) {
    wkc += moog_write8(slave, 0x1C12, 0, 0);
    wkc += moog_write8(slave, 0x1C13, 0, 0);

    wkc += moog_write16(slave, 0x1C12, 1, 0x1600);
    wkc += moog_write16(slave, 0x1C13, 1, 0x1A00);

    wkc += **moog_write8**(slave, 0x1C12, 0, 1);
    wkc += **moog_write8**(slave, 0x1C13, 0, 1);

  printf("slave_id: %x", slave);
  return 0;
}
'''

jtsdana, 2020-03-06:
> ```
> static int Map_PDO_CMMT(uint16_t slave) {
>     wkc += moog_write8(slave, 0x1C12, 0, 0);
>     wkc += moog_write8(slave, 0x1C13, 0, 0);
> 
>     wkc += moog_write16(slave, 0x1C12, 1, 0x1600);
>     wkc += moog_write16(slave, 0x1C13, 1, 0x1A00);
> 
>     wkc += moog_write8(slave, 0x1C12, 0, 1);  <----write8
>     wkc += moog_write8(slave, 0x1C13, 0, 1);  <----write8
> 
>   printf("slave_id: %x", slave);
>   return 0;
> }
> ```
> 
> In slave_info_reboot.pcapng.zip , the above routine is not run

I changed the moog_write16 -> moog_write8 and it works. Here is the code:
```
#include <stdbool.h>
#include <stdio.h>
#include <string.h>
#include "ethercat.h"
#include "osal.h"
#include "oshw.h"
// EtherCAT variables:
char IOmap[4096];
int wkc = 0;
unsigned int usedmem;
static int moog_write8(uint16 slave, uint16 index, uint8 subindex,
                       uint8 value) {
  int wkc;

  wkc = ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value,
                    EC_TIMEOUTRXM);
  return wkc;
}
static int moog_write16(uint16 slave, uint16 index, uint8 subindex,
                        uint16 value) {
  int wkc;

  wkc = ec_SDOwrite(slave, index, subindex, FALSE, sizeof(value), &value,
                    EC_TIMEOUTRXM);
  return wkc;
}

static int Map_PDO_CMMT(uint16_t slave) {
  printf(" MAP PDO ");
  wkc += moog_write8(slave, 0x1C12, 0, 0);
  wkc += moog_write8(slave, 0x1C13, 0, 0);

  wkc += moog_write16(slave, 0x1C12, 1, 0x1600);
  wkc += moog_write16(slave, 0x1C13, 1, 0x1A00);

  wkc += moog_write8(slave, 0x1C12, 0, 1);
  wkc += moog_write8(slave, 0x1C13, 0, 1);

  printf("slave_id: %x", slave);
  return 0;
}

typedef struct PACKED {
  uint16_t controlword;
  int8_t mode_op;
  int32_t target_position;
  uint32_t profile_velocity;
  int16_t target_torque;
  int32_t velocity_offset;
  int16_t torque_offset;
} out_festo_t;

typedef struct PACKED {
  uint16_t status_word;
  int8_t mode_op_display;
  int32_t position_actual_value;
  int32_t velocity_actual_value;
  int16_t torque_actual_value;
} in_festo_t;

in_festo_t *in_festo;
out_festo_t *out_festo;
int nic_setup(char *ifname) {
  /* initialise SOEM, bind socket to ifname */
  if (!ec_init(ifname)) {
    return 1;
  }

  /* find and auto-config slaves */
  if (!ec_config_init(FALSE)) {
    return 1;
  }
  printf("%d slaves found and configured.\n", ec_slavecount);
  ec_slave[1].PO2SOconfig = Map_PDO_CMMT;

  ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE * 4);

  usedmem = ec_config_map(&IOmap);
  if (!(usedmem <= sizeof(IOmap))) {
    printf("IO Map not big enough");
  }

  printf("Slaves mapped and configured.\n");

  ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE * 4);

  printf("segments : %d : %d %d %d %d\n", ec_group[0].nsegments,
         ec_group[0].IOsegment[0], ec_group[0].IOsegment[1],
         ec_group[0].IOsegment[2], ec_group[0].IOsegment[3]);

  out_festo = (out_festo_t *)ec_slave[1].outputs;
  in_festo = (in_festo_t *)ec_slave[1].inputs;

  printf(" status: %x \n mode op: %x \n positon_actual: %x \n ",
         in_festo->status_word, in_festo->mode_op_display,
         in_festo->position_actual_value);
  // send one valid process data to make outputs in slaves happy
  ec_send_processdata();
   wkc = ec_receive_processdata(EC_TIMEOUTRET);
  // wait for all slaves to reach OP state
   ec_slave[0].state = EC_STATE_OPERATIONAL;

  return 0;
}

int main() {
  // Setup EtherCAT interface:
  if (nic_setup("enx24f5a28cffd7") == 1) {
    // Setup encountered and error and cannot continue.
    return 1;
  }
  // Test remapping:
  // os=sizeof()

  return 0;
}
```

nakarlsson, 2020-03-06:
nice, please close the issue

PS: I'm not sure the slave is conformant since it don't come up using CoE with default settings. Instead it use the EEPROM PDO data, I suspect without digging in the data, without success.

jtsdana, 2020-03-06:
> nice, please close the issue
> 
> PS: I'm not sure the slave is conformant since it don't come up using CoE with default settings. Instead it use the EEPROM PDO data, I suspect without digging in the data, without success.

I will refer to this issue, in the email I'm sending to Festo about this issue, hopefully this will be resolved in the future :) 
And thanks aaaalot for the assistance!!!

closed 2020-03-06 by jtsdana

lzrhcj, 2023-01-10:
> Looks like the online ObjectDictionary is not populated by default as stated in the ESI
>
> Given 1600 and 1A00 is populated correctly check the ESI and populate 1c12 with 1600 1c13 with 1a00
>
> How to [...]

> Looks like the online ObjectDictionary is not populated by default as stated in the ESI
>
> Given 1600 and 1A00 is populated correctly check the ESI and populate 1c12 with 1600 1c13 with 1a00
>
> How to [...]

Thank you very much! Can I ask for more detailed information? Because I also have this problem when using other types of motors, I hope I can refer to the solution of this problem to solve the same problem. I have repeatedly checked your solution to this problem, but I don't understand this point. May I ask how you can see from this esi file how to populate it correctly. Assuming I'm a beginner, thanks!!!!