OpenEtherCATsociety/SOEM #380: COE Test file for SOEM

nshubham5741, 2020-03-07:
I am new to EtherCAT & CAN over Ethernet. I am trying to implement EtherCAT master on a ARM processor with linux booted on it. Can someone please provide COE test example..

nakarlsson, 2020-03-08:
The method is generic but the data to uses is drive specific.

Check #379 for the method.

nshubham5741, 2020-03-11:
Thank you.. I now got a better idea about SOEM library.. 

I am referring Erika documentation : http://www.erika-enterprise.com/wiki/index.php?title=EtherCAT_Master for controlling a drive.

Can you please let me know the correct general procedure for PDO mapping with SDO, and sending/ getting the input & output parameters to the drive?

nakarlsson, 2020-03-11:
I guess you can combine the ERIKA documentation with #379 and you'll get close enough.
In the end SOEM is just  a library, you write the EtherCAT Master when it come to executing the controller loops and tasks.

nakarlsson, 2020-04-21:
I'll close it

closed 2020-04-21 by nakarlsson