OpenEtherCATsociety/SOEM #381: general question to "Accessing data through IOmap" from tutorial.txt

AlSa-91, 2020-03-15:
First of all I want to thank for this great work!
Working through the tutorial.txt (simpletest, red_test, slaveinfo) and closed/open issues I don't understand what the meanings of "module index as index internal to the slave in case more than one channel" and in this context "slave_EL4001_1.out1" (which is an integer used in set_output_int16) from the tutorial.txt are.
Can someone help me? Thanks alot

nakarlsson, 2020-03-16:
It’s based on https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/rtk/main.c.

And the set function is a utility function. Say that you have a slave_EL4002, I think you have 2x uint16 channels, then the second channel would be module index 1. So, using the set/get function set/get a single channels within a single slave. SOEM only have IOmap data pointer per slave and leaves it to the Application to encode/decode channel based slaves.

AlSa-91, 2020-03-17:
Ok, It's clear. Thank you!

closed 2020-03-17 by AlSa-91