OpenEtherCATsociety/SOEM #383: Blocked at ec_receive_processdata(), unable to receive processdata

KevinHwangAi, 2020-03-18:
I run the simple_test in ubuntu 18.04, when runs to ec_receive_processdata() ,the program is blocked there,and the slave's processdata is no longer received.

`SOEM (Simple Open EtherCAT Master)
Starting to configure robot
ec_init on eth0 succeeded.
ec_config_init slave 1
1 slaves found and configured.
In PRE_OP
Motor drive setup
Motor drive wkc1: 2
Motor drive wkc3: 6
Motor drive wkc4: 10
Motor drive wkc5: 14
Slaves mapped, state to SAFE_OP.
segments : 1 : 16 0 0 0
Calculated workcounter 3
get into ec_receive_processdata_group
get into ecx_waitinframe

`
Please tell me where the problem may be?

ArthurKetels, 2020-03-18:
Please send a zipped wireshark capture. ec_receive_processdata(EC_TIMEOUTRET); is in principle not blocking. The program should continue.

KevinHwangAi, 2020-03-18:
Here is the Wireshk capture.
[soem_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4349160/soem_test.txt)

KevinHwangAi, 2020-03-18:
When a message is sent to slave through ec_send_processdata(), there is no response from ec_receive_processdata. It may be that slave sends back a message after sending the message, but does XX miss the receiving time

ArthurKetels, 2020-03-18:
Please do not send wireshark text files, the raw data files are much easier to parse for me with the help of wireshark. So a) capture, b) save (not save-as), c) zip, d) post here.

But nevertheless, the file looks ok. No problems with the slave or on the wire. So there must be something with your implementation or driver.  Please try on another computer.

KevinHwangAi, 2020-03-18:
When a message is sent to slave through **ec_send_processdata()**, there is no response from **ec_receive_processdata()**. I think it may be the **slave** sends back a message , but **ec_receive_processdata()** miss the receiving time?

ArthurKetels, 2020-03-18:
What I see is a return time for the LRD and LWR packets of a little over 190us. This is quite long for only one slave. So my guess is that there is some packet coalescing going on in your linux NIC driver. See the  drvcomment.txt file in the root dir of SOEM.

Even with this time the default timeout setting for SOEM is 2000us. Did you change these values?

Then check the return value of ec_receive_processdata(). A negative value will indicate a timeout. A positive value is the working counter. It should be 3 in your case.

closed 2020-03-21 by KevinHwangAi