OpenEtherCATsociety/SOEM #384: Use of a Faulhaber MC5005 with (orocos) SOEM

gborghesan, 2020-03-19:
Dear all,
I am trying to interface the motor control MC5005 in orocos.
Before that, i tried the slaveinfo command to get some insight, but it sends the controller in error state.
The same applies if I run with the orocos soem Master component.
After sending a wireshark tracing to the technical support, he pointed out the following

> the step PreOp to SafeOp don’t work.
Your AL- Error  is StatusCode=26.
>Only the following combinations of settings is valid. Everything else will produce this error message:
  >  SM0 = 0 & SM1 = 1 (Freerun)
   > SM0 = 1 & SM1 = 22h & ADO 0981 (SyncOut Unit) bits [3,1,0] := 000b (SYNC0 deactivated)
   > SM0 = 2 & SM1 = 2 & ADO 0981 (SyncOut Unit) bits [3,1,0] := 011b (SYNC0 is active)

> SOEM master can only handle distributed clocks DC.
> That means SM0 = 2 & SM1 = 2.
> The Bit-combination in Register 0981 is wrong.

Since my experience is limited (so far only adapted drivers btw similar devices), I have no clue on how to make it working. Any direction appreciated.
Cheers, Gianni.

ArthurKetels, 2020-03-20:
Please post the wireshark trace here too. I want to make sure I have all relevant data before postulating an answer. SOEM can do all operation modes, freerun, SM sync, and SYNC0 sync.

gborghesan, 2020-03-20:
Dear Arthur,
I zipped 3 traces in [slave_info_MC5005.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4359647/slave_info_MC5005.zip)

- slaveinfo.pcapng:  using the command slaveinfo without options 
- slaveinfo_map.pcapng:   slaveinfo with map option
- slaveinfo_sdo.pcapng:   slaveinfo with sdo option

Below the outcome of the commands.

**$ sudo ./slaveinfo enxe4b97ab11172**
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enxe4b97ab11172 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  26 : Inconsistent settings

Slave:1
 Name:MC5005
 Output size: 48bits
 Input size: 144bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000147 ID: 00003521 Rev: 0000000a
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:1c00 L:  18 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:  18 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

**$ sudo ./slaveinfo  enxe4b97ab11172 -map**
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enxe4b97ab11172 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  26 : Inconsistent settings

Slave:1
 Name:MC5005
 Output size: 48bits
 Input size: 144bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000147 ID: 00003521 Rev: 0000000a
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:1c00 L:  18 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:  18 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10
  [0x0002.0] 0x0000:0x00 0x00
End slaveinfo, close socket
End program

**$ sudo ./slaveinfo  enxe4b97ab11172 -sdo**
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enxe4b97ab11172 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  26 : Inconsistent settings

Slave:1
 Name:MC5005
 Output size: 48bits
 Input size: 144bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000147 ID: 00003521 Rev: 0000000a
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:1c00 L:  18 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:  18 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
Time:    2983.210 MBX slave:1 error:0004 The service is no supported
Time:    2983.211 MBX slave:1 error:0004 The service is no supported
... (the same message with increasing time)
Time:    2983.308 MBX slave:1 error:0004 The service is no supported
End slaveinfo, close socket
End program

ArthurKetels, 2020-03-20:
Thanks for the wireshark data. The problem is clear.

The slave maps objects 0x1a00, 0x1a01, 0x1a02 and 0x1a03 for outputs. Those objects are mutually exclusive. This slave can only have one active at the same time. 

The simple answer is, this slave has a bug in the PDO mapping, go to Faulhaber and ask them to fix the bug. The long answer is that you can also change the mapping yourself. Look for other posts here that show how to do that.

Second problem with this slave is that it does not support the object dictionary. But this does not prevent SOEM from mapping the slave correctly.

gborghesan, 2020-03-21:
Dear Arthur, 
Thanks for your insights - I forwarded your comments to Faulhaber. 
I will look on how to do the remapping as you suggested.

closed 2020-03-21 by gborghesan

zhangxue122, 2021-05-15:
@gborghesan Dear gborghesan,

Recently, I am trying to use the SOEM to control the faulhaber motor. Adapted motor drive in my project is MC5004. I also meet the same problem with "Slave 1 State=12 StatusCode=  26 : Inconsistent settings". Since I am new to SOEM, May I ask if you have solve this problem?

Cheers, Xue.

gborghesan, 2021-05-16:
Unfortunately, I had not the time to solve it, and a colleague used the
motor by means of LabVIEW.a

zhangxue122, 2021-05-17:
Got it. Thank you for your kind reply. :)