OpenEtherCATsociety/SOEM #385: Porting the SOEM to a new platform

Davidinos, 2020-03-20:
Goodmorning,
I write you since I'm trying to port SOEM to TIRTOS platform. More specifically I created new oshw and osal implementations and ported the redtest code in my application. 
As result I get "No slave found".
I have the doubt that the problem could be caused by the transmitted data content or the timing (which is a weak point in TIRTOS).
In attachment you can find the wireshark capture, can you please check it and tell me if the frame is correct at least from the transmitted data perspective?
[ethercat board redtest 10ms.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4359156/ethercat.board.redtest.10ms.zip)

Thank you.

Regards,
Davide

ArthurKetels, 2020-03-20:
Your transmitted packets look OK. The slave responds with the correct information. It looks like the frames are not received by your port. This can be seen from the repeated tries.

nakarlsson, 2020-04-21:
Can this be closed?

closed 2020-05-06 by nakarlsson