OpenEtherCATsociety/SOEM #386: Cannot enter to State OP with NanoTec C5-E motor driver

AlSa-91, 2020-03-20:
dear SOEM devs,

First of all sorry for my bad following english!
I'm certainly not the first one here to complain that he won't get into the State Op, as the solved forum posts show, so I hope you can help me too.
[Study Behaviour of Reaching Op State.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4361568/Study.Behaviour.of.Reaching.Op.State.zip)

My goal is to put a NanoTec C5-E engine control unit into OP mode in order to operate it later via PDO, which I obviously cannot do and I have been stagnating for a long time despite reading other open and solved problems here in the forum. 

I can see from codeblocks in debug mode that when I execute my program at the first ec_writestate(0); there must be a malfunction, because after execution two red lights on my controller go on and indicate errors. 
The one lamp, which is the operation LED, indicates error voltage (according to the manual), which I neglect at first approximation, because otherwise there are no problems (e.g. when executing simpletest or slaveinfo).
The other LED (ERR) is one of two "EtherCAT LEDs" (RUN and ERR). ERR (Process Data Watchdog, Timeout/ EtherCAT Watchdog) indicates an error, but I don't know how to interpret the error.

For example, I have tried:
    #357 ecx_context.slavelist[1].FMMUunused = 1; => same problem
    #218 or #142 where SOEM gives me no error, so I don't assume that the DC is my problem here
    #224 timeouts changed => same problem

Attached are wiresharks, my source code, its results as well as an ESI- file, calls to slaveinfo -sdo + map and eepromtool and notes about my operating system. I have packed the data into a zip or would it have been better to break down the data here individually?
OP_State_CppCode is my own code, which is based on simpletest or tutorial.txt (I also tried code snippets from redtest, but it felt like it opened more construction sites).

I hope someone can help me out of this problem and thank you in advance!

ArthurKetels, 2020-03-20:
The curious thing is that from your wireshark trace the slave goes to OP without any problem.

The issue might not be with EtherCAT but just your slave configuration/setting. It might throw an error condition because wrong encoder / voltage / home run / etc. etc. Again from an EtherCAT perspective all is well.

The slave does not support object dictionary over CoE, that is why the map and sdo list feature does not work. Not nice, but you can complain about that to the manufacturer.

AlSa-91, 2020-03-21:
Many thanks for the quick response @ArthurKetels.
With reference to your answer I have some questions:

1) How can you tell from the Wireshark that the slave is going into OP? Does this also apply to the Wireshark of my program or only to the slaveinfo? I can see from the result that at some point State OP was reached, an explicit output of "ec_slave[1].state" (=2) at the end tells me that it must be in Pre Op...

2) I have now additionally configured motor data such as voltage encoder etc. via sdo and send them to the controller before the IOmap command in the hope that this might solve the problem. But this problem remains.
A config file that takes this into account is actually already on the controller, because I configured the device via USB according to the manual. Is an additional initialization necessary despite the stored data on the controller?

3) Do you recognize that the device does not have an object directory via COE at the slaveinfo -sdo command or what exactly? 

4) I got two xml files from the manufacturer. The one I uploaded is the ESI file according to the manufacturer. The other one is an object dictionary 
[od_structure.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4363152/od_structure.txt)
, so I can use their control program (I save it here as txt for upload). Is this not the required object directory? Do I have to implement it into the slave via SOEM myself? In the manual there is only a short instruction for TwinCAT how to integrate the ESI file. However, I have not yet worked with TwinCAT and I do not actually intend to do so.

5) Interestingly, error lamps light up when I run my code through an Infineon XMC4800 or another motor controller (Trinamics 8670 EvalKit). It always seems that the first "ec_writestate(0);" in my program causes the problem. As described in the tutorial.txt I only want to send some data to make the slave happy... Is it really my code that doesn't really do much different than slaveinfo or simpletest?

6) You indicated that I can complain to the vendor about CoE because of the object dictionary. Is that what the EtherCAT standard says?

7) Why does slaveinfo not always show me the same result after several calls in a row when I write my program only up to "ec_config_map(&IOmap);" and run it? By the way, the error lamps do not light up during this process! Sometimes it shows my changed PDO values, sometimes the default values and sometimes no values at all...How can this arbitrariness be explained?

Please forgive the amount of questions...

Thanks again!

AlSa-91, 2020-03-23:
Additional to my last post, I asked the manufacturer for the problem with sdo. He told me, that the drive supports object dictionary over CoE and asked me if I integrate the ESI-File in the master. 
Which makes me wonder what the slaveinfo -sdo command actually does. Does it simply read the possible sdo commands, which are also in the ESI-file? I don't quite understand why -sdo (and -map) command doesn't give me anything or only these error messages ("error:0004 The service is no supported").

In the forum I often read that ETG (if EtherCAT Group is meant) should provide more information or a better understanding of EtherCAT. The problem is that as a student I can't register there, which I found out after a phone call.
Thanks again!

ArthurKetels, 2020-03-23:
Sorry to be so blunt, but this "issues" section of the project is not the right place to educate people about the workings of EtherCAT. When you are a student you can ask you school to become an ETG member. It is free of charge. Many schools are ETG member.

If other visitors of this site want to make some effort to explain things to you, thanks very much.

AlSa-91, 2020-03-23:
Thank you again @ArthurKetels!
The problems are that the school is closed (COVID-19) and so the operation does not take place normally at the moment, which is obviously not your problem.
would you be so kind as to tell me whether this "missing object directory" should normally be implemented by the manufacturer or why is my slaveinfo -sdo so empty? i would not like to be fobbed off by the manufacturer by pointing out that the ESI file can be integrated into TwinCAT and that it works there...

ArthurKetels, 2020-03-23:
CoE Object Dictionary is not mandatory for slaves. A slave vendor could opt to only describe the CoE objects in the ESI (EtherCAT Slave Information) file. And obviously this would work with TwinCAT or other masters that keep an extensive slave library. 

SOEM however is aimed at all targets, also embedded platforms that have not the luxury of large databases and large mount of memory. A very nice feature of EtherCAT is that almost all information can be stored in the slaves. So opting to not using that feature is in my opinion very shortsighted of the slave vendor in question.

Then again, you are not forced to buy any of those slaves. Vote with your money, and support slave vendors that do understand the needs of all clients.

There are SOEM users that have implemented ESI file parsing. SOEM is after all only a library, and anyone can extend it to their specific needs.

AlSa-91, 2020-03-23:
i have already seen that other manufacturers show a different behaviour in this respect with my other slaves. i can understand your estimation, but as a customer you really have to be in the ethercat material to be able to consider this when buying the components. Thank you the last time for this issue ;)

closed 2020-03-23 by AlSa-91