OpenEtherCATsociety/SOEM #387: Why is PO2SOconfig deprecated and what is the better alternative?

ghusera, 2020-03-24:
"https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html" recommends setting a configuration hook ("ec_slave[slc].PO2SOconfig = EL7031setup;") to ensure that slave configuration happens at the right point.

But then, "https://openethercatsociety.github.io/doc/soem/structec__slavet.html#afc6aba2853562175217d30ba3a2b22a4", explains "registered configuration function PO->SO, (DEPRECATED)", meaning PO2SOconfig's use should be avoided ("https://techterms.com/definition/deprecated").

So, how should the slave configuration be called ideally?

Thank you!

nakarlsson, 2020-03-24:
A few years back we added definitions EC_VER1 and EC_VER2, were the later added the context concept to be able to support running multiple instances of SOEM.  But, most of the legacy examples still implement/use EC_VER1 while new functionality might not support EC_VER2.

So, the short answer is going for EC_VER2 and use PO2SOconfigx. Currently you'll see how it is used in https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/intime/ec_master/ec_master.c. And https://github.com/rtlabs-com/fsoe-soem-demo/blob/master/test/linux/fsoe_sample/fsoe_sample.c

The concepts and most of the functionality is the same for EC_VER1 and EC_VER2, guess that’s why we're lagging behind when it comes to porting the examples.

ghusera, 2020-04-12:
Thank you, but I just don't see ... there is none of "EC_VER" or "PO2SOconfig" on either of the links you included in your reply.

nakarlsson, 2020-04-12:
Ok, use PO2SOconfig then and just try it out, I think you'll figure out what you need to know along the way.

nakarlsson, 2020-05-09:
Try it out and re-open if needed

closed 2020-05-09 by nakarlsson

ghusera, 2020-05-11:
I did try PO2SOconfig. It works just fine for my one slave setup.
I learned nothing about EC_VER1 and EC_VER2 or PO2SOconfigx.
For the sake of those who will find this thread via some search function in the future:
Please enlighten us!