OpenEtherCATsociety/SOEM #388: Update to using SOEM with Intime

SimonWakley, 2020-03-24:
I am trying to use SOEM 1.4.0 with Intime 4.2 and Intime 6.3.  With the existing CMAKE setup, I could not see how to make it work with Intime and had to edit a few files adding nicdrv.h to get it to compile as well as remove some initialization from ec_master.c to get it to compile.  I feel like I am missing some documentation etc.  Does anyone have updated data or documents on this.  I am currently using the Hilscher Card under Intime to provide EtherCAT master, but I would like to see if using SOEM is a viable alternative.
Is anyone else working on this and could maybe provide some pointers.
There are a couple of older posts that are somewhat similar but the same issues seem to exist 3 years later and I was wondering if there was a branch I should get etc.
Thanks.
Simon

nakarlsson, 2020-03-24:
What you see is what we got for free.

SimonWakley, 2020-03-24:
Ok Totally understood - for free this is AMAZING.  I would like to see if I can get it to work with Intime and is there anyone else you know of who is working on this.  I am totally happy to post changes etc and help out.  Any user groups etc that you know of?

nakarlsson, 2020-03-25:
Not that I'm aware of.

But to give it a try, why not just add the SOEM\OSAL\OSHW for IntTime to your InTime project, compile and fix any issues.

In the end, it is the nicdrv/gettime and mutex(if threading is required) support that is required

SimonWakley, 2020-04-16:
I have been working on this here and there and have it working and compiling with my Intime project.  Right now it is seeing the adapter for me but it is not running SlaveInfo or SimpleTest fully.  Would it be useful to do a push with the changes at some point as there is definitely some code changing needed to get it to work with Intime more fully.  The guys in the UK whom I work with have had more success than I and maybe they could do the push?

nakarlsson, 2020-04-16:
You should try to get it to work first, we currently don't have access to the InTime environment so any review would be purely academic, given the changes are big.

If you got/get some partial success and can provide a Wireshark, we could analyze that.

nakarlsson, 2020-05-09:
@SimonWakley, any progress?

SimonWakley, 2020-05-09:
Hi,  Thanks for checking in!  I have made some limited progress with talking to different drives, I have worked with Elmo, Schneider, Moog and Kollmorgen drives and I believe these will all work well, but I have been distracted from the project and have not gotten it working with Intime yet.  I can detect the card but it will not see Slaves, but the Crew in the UK got SimpleTest (or a version thereof) to work, but with Intel card and I only have Realtek.  I am hoping to go to the office next week where I have Intel cards and give it another try.  No other people working with SOEM under Intime as far as we know?

SimonWakley, 2020-05-13:
Hi,  So the issue was with the Realtek Card not working but with an Intel card it does.  So I now have a motor moving under my control via SOEM on Intime, so that's a big step forward.  It's an Elmo drive and I have set the PDOs to my liking.  Having issues with other Slaves at the moment such as the Robot I am using, it's much easier to work under windows even though it is not terribly real-time the debugging tools are better for me.  I'm going to see if I can work out what is wrong with this Realtek card and if it is something I can sort out or if I have to get Tenasys to modify their driver.  Cyclic Position Mode only.  (I assume that's what most people would be using).

SimonWakley, 2020-05-13:
With some slight mods when I run slaveinfo with rtl1g0. (The adapters are not found, they need to be specified) I get
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Available adapters
#1:  name: lo0
#2:  name: lo0
#3:  name: lo0
#4:  name: lo0
Starting slaveinfo
SOEM: hpeOpen rtl1g0 with options
INF_3437|rtl1g0:|Vers=001002001|Date=Apr 18 2018 23:42:22 istack:|Vers=001002001|Date=Apr 18 2018 23:43:02
INF_3475|End dev allocation
WRN_506|PCI extcap 3 not found
INF_803|62:00.0 10ec/8168/17aa/38d4 80 Realtek 8168 PCIe GBE Family Controller
Using Memory Mapping!
DBG_443|MsiOffset != 0, assuming MSI works
DBG_443|MsiOffset != 0, assuming MSI works
Using 1 MSI message
INF_1286|IRQ alloc *rid=1:MSI
version:1.94.01

The intel card prints out a lot more info
[SOEM_IntimeIntelWorks.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4619611/SOEM_IntimeIntelWorks.txt)

HW_revision: 54100000
Ethernet address: 9c:5a:44:6b:22:6b

This product is covered by one or more of the following patents:
US6,570,884, US6,115,776, and US6,327,625.
INF_1124|MAC address: 9c:5a:44:6b:22:6b:
INF_1754|Polling mode, tick=1
INF_853|Attached rtl1g0:Realtek 8168 PCIe GBE Family Controller
DBG_293|params=23970 TX sem=0000 239e0  RX sem=0000 23a20
ec_init on rtl1g0 succeeded.
No slaves found!
End slaveinfo, close socket
INF_1972|Irtl1g0 terminating
INF_2071|Jrtl1g0 terminating
End program

But from a similar machine with an intel card and the same slave

nakarlsson, 2021-02-09:
@SimonWakley , Can we close this issue?

nakarlsson, 2021-07-21:
We’ll close due to inactivity

closed 2021-07-21 by nakarlsson