OpenEtherCATsociety/SOEM #390: Library of Setup Steps for EtherCAT drives

SimonWakley, 2020-04-16:
Different drives need different set ups to get them to work with EtherCAT and this seems to be something of a complicated step requiring reading through manuals and trying to decipher what the (usually poor) translations actually mean.  I am working with a Copley XEL and an Elmo Gold and when I get them working is there a separate repo for the working Config() code?  Obviously it varies from user to user/use to use - so there is no right and wrong code but it might be useful to have access to a library of functions.

nakarlsson, 2020-04-16:
Hi Simon, currently there are no such repo that I know of. It would be a good feature for the future.
I guess you have a fork of SOEM, why not provide an example how it would look like?

nakarlsson, 2022-01-14:
close due to inactivity

closed 2022-01-14 by nakarlsson