OpenEtherCATsociety/SOEM #391: Sync problem

casinia, 2020-04-17:
Hi, all.
I wonder if the master is synchronized with the slave, so I print out toff, ec_DCtime, ts and (ts - ec_DCtime) for each cycle. As shown in the figures, (ts - ec_DCtime) was nearly 1ms at the beginning, but then the value became smaller and smaller, and then increased in the reverse direction. And DCtime changes about 1ms every cycle, ts also changes about 1ms. In this case,  are the master and slave synchronized?

[image]
[image]
[image]
[image]
[image]

ArthurKetels, 2020-04-17:
Could you post the whole list as a zipped csv file here?

casinia, 2020-04-20:
[Uploading log_data.zip…]()
I post data here.

casinia, 2020-04-20:
[log_data.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4500965/log_data.zip)
I post data here.

casinia, 2020-04-20:
ec_receive_processdata(EC_TIMEOUTRET);

if (ec_slave[0].hasdc)
{
	ec_sync(ec_DCtime, cycletime, &toff);
}
ec_send_processdata();

clock_gettime(CLOCK_REALTIME, &ts);
tv.tv_sec = ts.tv_sec;
tv.tv_usec = ts.tv_nsec / 1000;
tv.tv_sec -= 946684800UL;
pc_time = (((uint64)tv.tv_sec * 1000000) + (uint64)tv.tv_usec) * 1000;

The above code is how I get the ts.The third column in logged_data.txt is pc_time.

ArthurKetels, 2020-04-20:
Thanks for the data. It is clear you have no synchronization between master and slave.

I do not know you whole RT loop code. Above is missing some important pieces. How do you use toff in your timing? Because this is the feedback from ec_sync. 

When your pc clock is too far of then the simple PI controller in ec_sync can not correct anymore. Please change the P and I gains to suit your hardware.

And also, NEVER use CLOCK_REALTIME, because it is not real time and can be adjusted by your system time server. And by the looks of your data it is adjusting time like crazy. Use CLOCK_MONOTONIC. 

When you look at your pc time adjustment (left column) it integrates to exactly the same difference as between pc and slave (right column).

casinia, 2020-04-21:
Since the system I use does not support clock_nanosleep (CLOCK_MONOTONIC, TIMER_ABSTIME, & ts, & tleft), so I can only use a hardware timer to delay, and when the delay is completed, start the RT task with a semaphore:

for(;;)
{
    timer_delay = cycletime+ toff;
    set_harware_timer_delayvalue(timer_delay );
    start_harware_timer();

    /*hardware timer delay is completed, i can take a semaphore*/    
    rt_semaphore_take(SemId, WAITING, 0);
    if (dorun>0)
    {
         wkc = ec_receive_processdata(EC_TIMEOUTRET);
         dorun++;
         if (ec_slave[0].hasdc)
         {
            /* calulate toff to get linux time and DC synced */
            ec_sync(ec_DCtime, cycletime, &toff);
         }
         ec_send_processdata();
    }
}

/* PI calculation to get linux time synced to DC time */
void ec_sync(int64 reftime, int64 cycletime , int64 *offsettime)
{
   static int64 integral = 0;
   int64 delta;
   /* set linux sync point 50us later than DC sync, just as example */
   delta = (reftime - 50000) % cycletime;
   if(delta> (cycletime / 2)) { delta= delta - cycletime; }
   if(delta>0){ integral++; }
   if(delta<0){ integral--; }
   *offsettime = -(delta / 100) - (integral / 20);
   gl_delta = delta;
}

casinia, 2020-04-21:
Is my logic about hardware timer delay correct?
And how i change  the appropriate  P and I gains?

casinia, 2020-04-23:
If (ts - ec_DCtime) is 1ms, is it in sync?

casinia, 2020-04-26:
Still here?
I was wondering if I was testing the wrong difference.
The result I tested is the ec_DCtime shows 500us after sync 0 event which is in the middle of the cycle. Does this mean that the synchronization was successful?

ArthurKetels, 2020-04-27:
A random time sample says nothing about being "synchronized". Only a long term fixed relation between two clocks in offset and drift signals sync. When you compare ec_DCtime with your system clock it is clear your system clock is not well behaving. 

So my suggestion is to taken another platform (linux box) and do the same tests. From there you can learn about the details of clock synchronization. If the hardware you test now is not suitable then no amount of software tricks will help. Repair what is broken, do not try to duct-tape it.

Changing P and I gain is trivial. 
`*offsettime = -(delta / 100) - (integral / 20);`
1/100 is the P gain, 1/20 is the I gain.
When you do not know how a PI controller works : https://en.wikipedia.org/wiki/PID_controller

nakarlsson, 2020-07-14:
@casinia , can we close this ?

closed 2020-08-14 by nakarlsson