OpenEtherCATsociety/SOEM #392: MacOS Catalina Slave Power Cycle Problem

berkaydeniz, 2020-04-17:
Every time I power cycle the slave the assertion on line 443 of nicdrv.c is triggered
```
Assertion failed: (0), function ecx_inframe, file /Users/denizilhan/ghost/sandbox/foc_float/ble-tool/SOEM/oshw/macosx/nicdrv.c, line 443.
```
and thus the software crashes. If I comment out the line, `//assert(0);` things work just fine. Obviously this assertion is in place for a purpose. Are there any general rules for handling a slave power cycle that I am missing or is this a bug.

Currently using commit version `7fe05b5b`.

Note that the same software work perfectly fine with no crashes when built under Ubuntu. So this is specifically triggered on MacOS Catalina.

ArthurKetels, 2020-04-17:
Looks like a frame is duplicated by the OS network stack. Can you make a wireshark trace?

nakarlsson, 2020-04-18:
It think the assert can be removed. The frame have been abandoned.

Why it happens is strange, specially in a single threaded SOEM application. Best guess, the frame get buffered in some send Q when the power cycle occur. During the power cycle SOEM try to receive it and fails, marking the buffer EMPTY.
But again, think the assert is to drastic, linux don’t assert. We don’t assert in other plattforms either.

nakarlsson, 2020-05-09:
Fixed in #399

closed 2020-05-09 by nakarlsson