OpenEtherCATsociety/SOEM #394: Using SOEM on Arduino Due

robin843, 2020-04-20:
Hi,

does somebody know if it is possible to use SOEM on an Arduino Due with an Ethernet Shield? I already tried to write osal/oshw/nicdrv files. When using **ec_config_init()** to detect all slaves it somehow run into problems. It detects all slaves but never finishes this method. It stucks at random positions in code after detecting slaves with **ecx_detect_slaves()** .

So I'm asking myself if it is in general possible to run SOEM on Arduino?

Thank you for your help!

Best regards,

Robin

ArthurKetels, 2020-04-20:
512K Flash and 100K RAM are enough for SOEM, also the CortexM3 is more than powerful enough. But you should reduce the structure size a bit. This means less slaves and less deep buffers. For the rest there are no reasons why it should not work.

nakarlsson, 2020-05-06:
Can we close this?

robin843, 2020-05-09:
yes! Thank you very much!

closed 2020-05-09 by robin843