OpenEtherCATsociety/SOEM #395: Diagnosis Message

antoinericha, 2020-04-29:
I am using SOEM to run two EL7201 servo drivers along with EL 2809 and EL1008.
I always get a sudden Error in the EL7201. I read the 10F3 CoE to get the diagnosis messages.
I got the following message:
0x00 0xffffffe0 0x21 0x1c 0x02 0x00 0x05 0xffffff81 0xfffffff4 0xfffffff3 0xffffffb6 0xffffffba 0x4f 0x7e 0x77 0x2e 0x06 0x00 0x00 0x00 0x06 0x00 0x00 0x00 0x05 0x00 0x00 0x00

Can Someone help me in decoding this message. I tried to follow the documentation of Error Correction but i could not relate to a valid text ID

ArthurKetels, 2020-05-03:
How exactly did you read 0x10f3 object? What subindex did you use? It is always best to post the wireshark capture here to avoid mistakes.

Anyway, I assume you read bytes and print them out as signed 32 bits (should have printed unsigned 8bits) .

Then the messageid is : 0x8105

Looking up in the documentation :

> 0x8105 Error GeneralPD-Watchdog
Communication between the fieldbus and the outputstage is secured by a Watchdog. The axis is stopped automatically if the fieldbus communication is interrupted.- The EtherCAT connection was interrupted during operation- The Master was switched to Config mode during operation 

Conclusion : You are not running your PDO real-time and the slave watchdog times out. Again a wireshark trace could have told you that without reading the diag from the slave.

nakarlsson, 2020-05-09:
@antoinericha , can we close the issue?

antoinericha, 2020-05-09:
Sorry for the late reply. 
Concerning how i read the message, I used the same function used in slaveinfo (char* SDO2string) with dtype octet string(000a).
With this function, I read  0x10f3 with subindex of the latest message received that has an octet string type.

closed 2020-05-20 by antoinericha