OpenEtherCATsociety/SOEM #396: `firm_update.c` doesn’t disable the SyncManagers before reconfiguring them

gralpli, 2020-05-04:
I'm using the Beckhoff Slave Stack Code which initializes and enables the SyncManagers at bootup. `firm_update.c` tries to reconfigure SyncManager 1 to a different start address, but fails in doing so, because the SyncManager is already enabled. It needs to be disabled first, otherwise the new configuration is simply ignored.

Pull request #56 solves this issue.

ArthurKetels, 2020-05-06:
The SSC should not initialize any SM by itself. This is a protocol violation.

All the sample programs need to be updated to the current VER2 spec of SOEM. This implies using context and the ecx_ versions of the functions. But for me it doesn't itch enough, so no scratching expected for the near term ;-)  When other users feel obliged to give the good example, by all means.

gralpli, 2020-05-06:
Right, as we are in INIT state after bootup, there should be no SM enabled. I will check why this appears to be the case on my slave.

Updating all the sample programs looks like a nice task that would make a good first contribution from me 😇 I think I will do this in the near future. Together with SOEM these sample programs were really useful for me while developing my slave, and still are for testing!

nakarlsson, 2020-07-14:
Can we close this one?

gralpli, 2020-07-14:
No, the firmware update program from the master branch still doesn’t work for me.

SOEM brings the device to PREOP, so it’s not the Beckhoff SSC that enables the SMs. Then, the `firm_update.c` goes back to INIT and tries to reconfigure the SMs which, if I remember correctly, fails on my device, because the SMs have their enabled bit still set and don’t accept changes in this state.

nakarlsson, 2021-07-21:
Close due to inactivity. Also, doesn’t seem to be an SOEM issue.

closed 2021-07-21 by nakarlsson