OpenEtherCATsociety/SOEM #397: Synchronization

Zago86, 2020-05-06:
I'd like to know about synchronization: 
the ec_send_processata() must always be paired by ec_receive_processdata(), or you can call two or more consecutive ec_send_processdata() and after the ec_receive_processdata(). if that, what would happen?

thanks.

ArthurKetels, 2020-05-06:
Your question is not about synchronization but about ordering. 

ec_send_processdata() forms an ordered pair with ec_receive_processdata(). So first a send and then a receive. You can concatenate as many pairs as you like, send-receive send-receive send-receive.....etc, etc.

The only exception is when using ecx_send_processdata() and friends for multiple groups. You can assign each individual slave to a group and have their processdata mapping separate per group. For each group there needs to be a send-receive pair.  It is not needed to have the same update rate for each group (that is actually the whole point in using groups).

Zago86, 2020-05-07:
Thanks for the reply.

I would like to know what is the best way to understand the cause of "wkc <expectedWKC" after sending the command wkc = ec_receive_processdata (EC_TIMEOUTRET))

It is not clear to me whether ec_readstate () returns the current state or not.

nakarlsson, 2020-05-07:
You can check simple_test -> ecatcheck for an idea.

ec_readstate updates the local ec_slave array with the current states/al status of all slaves in addition to returning the state of the currently lowest found slave state in the network, 
From there you can check all slaves status and status code using the local cached values. 

Do refresh of the local values when needed using ec_readstate.

nakarlsson, 2020-07-14:
@Zago86 , can we close this?

closed 2020-08-14 by nakarlsson