OpenEtherCATsociety/SOEM #398: Setting up multiple slave groups

davoodmoradi, 2020-05-07:
Hi, 
I am trying to set up two slave group with different update rate on my EtherCAT slaves. I can make it work when i set only one group, but having problem when try to set up for two groups. Is there an example on how to set up multiple groups? Which parameters i need to modify? 
Regards

ArthurKetels, 2020-05-07:
No example available yet.

Steps:
ec_config_init()
ec_slave[i].group = 1 // i is list of slaves for group 1
ec_slave[j].group = 2 // j is list of slaves for group 2
ec_config_map_group(&groupio1, 1) // maps slaves in group 1 to groupio1 buffer
ec_config_map_group(&groupio2, 2) // maps slaves in group 2 to groupio2 buffer

PDO loop 1 at x Hz
ec_send_processdata_group(1)
ec_receive_processdata_group[1, timeout)

PDO loop 2 at y Hz
ec_send_processdata_group(2)
ec_receive_processdata_group[2, timeout)

Slaves that have not been reassigned to a group have group 0. 
ec_send_processdata() is the same as ec_send_processdata_group(0)

davoodmoradi, 2020-05-07:
Hi Arthur,
Thanks for your reply. 
Let me explain a little about my test project. I have 10 slaves and IOmap size is 47. If I don't use group and put all the slaves in group 0, every thing works perfect, no issue.
I am trying to to set 2 groups, slave #1 (servo drive) belongs to group 1 and the rest belong to group 2 (some beckhoff IOs). The IOmap size for group 1 is 24 bytes and for group 2 is 23 byts.
I modified the EC_MAXGROUP from 2 to 3 and recompiled the the library.
I am following your guide to configure them. Following is my code:
if (ec_init(ecat_ifname))
	{
		if ( ec_config_init(FALSE) > 0 )
		{
			for(x=1; x<=ec_slavecount; x++)
			{
				if(x<=1) ec_slave[x].group = 1;
				else ec_slave[x].group = 2;
			}
			x = ec_config_map_group(&IOmap1, 1);
			rt_printf("IOmapSize1 = %d\n",x);
			
			x = ec_config_map_group(&IOmap2, 2);
			rt_printf("IOmapSize1 = %d\n",x);
........................................................
when running this code, the first ec_config_map_group returns the right IOmapSize which is 24, but the second one instead of 23, returns 35. 
If I swap them and run:
                        x = ec_config_map_group(&IOmap2, 2);
			rt_printf("IOmapSize1 = %d\n",x);
                        x = ec_config_map_group(&IOmap1, 1);
			rt_printf("IOmapSize1 = %d\n",x);
again, the first ec_config_map_group returns the correct IOmapSize which is 23, but the second one instead of 24, returns 37. 
any idea what I am doing wrong?

ArthurKetels, 2020-05-07:
I made one mistake. Before ec_config_init() you must set the logical start address of each group.
ec_group[1].logstartaddr = 0x10000;
ec_group[2].logstartaddr = 0x20000; 
This assigns 64K IO space for each group. You can choose any location in 32bit address space below 0x8000000. As long as group mappings do not overlap. 

If the problem persists, please post a zipped wireshark capture.

davoodmoradi, 2020-05-07:
Still having the same problem. Preparing the wireshark capture and sent it to you asap.
Thanks for your help.

davoodmoradi, 2020-05-07:
Hi Arthur,
attached is zipped wireshark capture. Please let me know if you have any question.
[ECatWireShark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4595880/ECatWireShark.zip)

ArthurKetels, 2020-05-08:
Have you pulled the latest git for SOEM? The wireshark trace shows packets being send that do not align with the latest code base. Or you have added code in your test program that is more than you have stated above. Slaves are being put in safe-OP before a valid mailbox has been programmed.

Anyway, I have made a similar set-up and that works with my code suggestion. No issues found. Two groups map independent and can do independent PDO loops. Moving slaves from one group to the other behaves as expected.

davoodmoradi, 2020-05-09:
Hi Arthur,
I downloaded the latest version and installed it. After a couple of hours I managed to make it work. Thanks a lot for your help.

closed 2020-05-09 by ArthurKetels