OpenEtherCATsociety/SOEM #400: No PDO mapping

GuiyangXin, 2020-05-08:
Hi there, 
I am trying to use SOEM to connect a Copley drive. I can use SDO to send command, for example, velocity command  ec_SDOwrite(1, 0x60FF, 0x00, false, sizeof(speed), &speed, EC_TIMEOUTRXM);

I want to use PDO to send this command. I don't know how to map PDO. Can you please write down several lines to map PDO. I got the slaveinfo as follows,

```
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:IES
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000000ab ID: 00002050 Rev: 00010003
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1100 L: 256 F:00010022 Type:2
 SM2 A:1200 L: 256 F:00010064 Type:3
 SM3 A:1800 L: 256 F:00010020 Type:4
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0c
 CoE details: 0d FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket

```

Here is the Copley drive PDF document for object dictionary
http://www.copleycontrols.com/wp-content/uploads/2018/02/All-CANopen_Programmers_Manual-Manual.pdf  

Thanks in advance.

nakarlsson, 2020-05-09:
Please post a wireshark log

GuiyangXin, 2020-05-09:
Hi @nakarlsson ,

Thank you for your reply. Here is the wireshark log file I recorded when running slaveinfo test.
https://drive.google.com/file/d/1z90NE2RoWXSqJlD1k0G50FH1T1yqJXfv/view?usp=sharing

I forgot to add -map argument last time. After adding that, I got more message as follows,

```
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket

```
But this means there is no PDO as what the output and input size mean "Output size: 0bits
 Input size: 0bits".

I also tried to do PDO mapping for velocity command and feedback as shown in the tutorial.txt.

```
/* Map velocity PDO assignment via Complete Access*/
    uint16 map_1c12[2] = {0x0001, 0x60FF};
    uint16 map_1c13[2] = {0x0001, 0x606C};

    retval += ec_SDOwrite(slave, 0x1c12, 0x00, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
    retval += ec_SDOwrite(slave, 0x1c13, 0x00, TRUE, sizeof(map_1c13), &map_1c13, EC_TIMEOUTSAFE);
```
But this doesn't work. The result of ec_SDOwrite is 0. I think this means ec_SDOwrite failed to write value into 0x1c12 and 0x1c13.

ArthurKetels, 2020-05-09:
The wireshark trace shows no evidence of downloading PDO mapping to the slave.  Are you sure the code is executed before the mapping? And when not absolutely sure the slave can handle complete access, it is better to use the single index method.

Unless you write a valid PDO mapping in this slave it will not operate.

GuiyangXin, 2020-05-10:
> The wireshark trace shows no evidence of downloading PDO mapping to the slave. Are you sure the code is executed before the mapping? And when not absolutely sure the slave can handle complete [...]

Hi @ArthurKetels, the wireshark log is recorded during running slaveinfo.cpp without 
```
retval += ec_SDOwrite(slave, 0x1c12, 0x00, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
retval += ec_SDOwrite(slave, 0x1c13, 0x00, TRUE, sizeof(map_1c13), &map_1c13, EC_TIMEOUTSAFE);
```
Do you want me to record a wireshark log file during PDO remapping?

What do you mean using single index method? Is it like
```
/* Map velocity PDO assignment via Complete Access*/
    uint16 map_1c12 = 0x60FF;
    uint16 map_1c13 = 0x606C;

    retval += ec_SDOwrite(slave, 0x1c12, 0x00, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
    retval += ec_SDOwrite(slave, 0x1c13, 0x00, TRUE, sizeof(map_1c13), &map_1c13, EC_TIMEOUTSAFE);

```
According to the page 45 of Copley CANopen manual, the sub-index 0x 00 of 0x1c12 should contain the number of mapped PDOs under 0x1c12. The other sub-indexes are used to map PDOs, and their types are unsigned 32. The 32 bits are composed of 
0-7 Size (in bits) of the object being mapped. Must match the actual object size as defined in the object dictionary.
8-15 Sub-index of the object to be mapped.
16-31 Index of the object to be mapped.

So I think if I want to map 0x60FF to 0x1c12, I need to do 
```
  uint32 map_1c12 = 0x200060FF;

    retval += ec_SDOwrite(slave, 0x1c12, 0x01, TRUE, sizeof(map_1c12), &map_1c12, EC_TIMEOUTSAFE);
```
But this doesn't work neither.

ArthurKetels, 2020-05-10:
Searching this forum is too much work I guess. There are multiple posts about Copley drives.

Anyway, an example (not tested) for the PO2SOconfig hook:
`void AEPsetup(uint16_t slave)
 {
    uint8_t u8val;
    uint16_t u16val;

    // assign RxPDO mapping index 0x1700 to output SM2  
    u8val = 0;
    ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u16val = 0x1700;
    ec_SDOwrite(slave, 0x1c12, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
    u8val = 1;
    ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    // assign TxPDO mapping index 0x1b00 to input SM3
    u8val = 0;
    ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u16val = 0x1b00;
    ec_SDOwrite(slave, 0x1c13, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
    u8val = 1;
    ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    // set mode, 1 = profile position mode
    u8val = 1;
    ec_SDOwrite(slave, 0x6060, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

    // desired state, 30 = position controlled by EtherCAT
    u16val = 30;
    ec_SDOwrite(slave, 0x2300, 0x00, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);

    // set ESC watchdogdivider
    u16val = 2498;
    ec_FPWR(ec_slave[slave].configadr, 0x0400, sizeof(u16val), &u16val, EC_TIMEOUTRET);
    // set ESC watchdog counter
    u16val = 1000;
    ec_FPWR(ec_slave[slave].configadr, 0x0410, sizeof(u16val), &u16val, EC_TIMEOUTRET);
    // set ESC watchdog counter
    u16val = 1000;
    ec_FPWR(ec_slave[slave].configadr, 0x0420, sizeof(u16val), &u16val, EC_TIMEOUTRET);
  }`

GuiyangXin, 2020-05-10:
Your suggestion works. It successfully made the PDO mapping. Thank you very much. This channel can be closed.

closed 2020-05-10 by ArthurKetels