OpenEtherCATsociety/SOEM #401: Cannot reach OP with no error

WangKaimeng, 2020-05-21:
Hi,
I had the same issue as #138 . However, when I finished reading the #138, I still didn't know how to modify the SOEM code to deal with the wrong alignment of SM and FMMU.

I really would prefer a solution to this problem! Thanks in advanced!

ArthurKetels, 2020-05-21:
The problem is not with SOEM but with the slave firmware. Object 0x1c00 has the wrong paramters. So fix the slave.

WangKaimeng, 2020-05-22:
> slave firmware

slave firmware refers to xml file in EEPROM or slave code generated by SSC tool?

ArthurKetels, 2020-05-22:
The code generated by SSC tool. Your EEPROM is most likely OK but you CoE not. SOEM does not use the XML ESI file. You have to manually change the content of object 0x1c00. The RxPDO should be defined as output and the TxPDO as input. Not the other way around.

nakarlsson, 2020-07-14:
@WangKaimeng , can we close this?

arkorobotics, 2020-08-17:
Although this issue was closed years ago, we stumbled across the same problem when using SSC Tool V5.12 with the LAN9252 and found another working solution:

We had an issue where the SM2 and SM3 types would appear swapped when scanning for slave info.  SM2 type enumerated as an input (4) and SM3 type enumerated as an output (3), instead of the expected types SM2 = 3 [output] and SM3 = 4 [input]. After digging in deeper, we also noticed that the SM0 and SM1 types were swapped as well.

After running across this thread and a lot of head scratching, we finally compared the SyncManager code generated by V5.11 against V5.12. The issue we found isn't entirely obvious and the solution we came up with... works... but aren't exactly sure why. Our best guess is that it has something to do with how an array of UINT16's are parsed and sent as byte to the LAN9252.

**Using SSC Tool V5.11:**
"coeappl.c" structures and packs the SyncManager type info like this:

```
typedef struct OBJ_STRUCT_PACKED_START {
UINT16 u16SubIndex0; /**< \brief SubIndex 0*/
UINT8 aEntries[4]; /**< \brief Entry buffer (one for each SyncManager)*/
} OBJ_STRUCT_PACKED_END
TOBJ1C00;

TOBJ1C00 sSyncmanagertype = {0x04, {0x01, 0x02, 0x03, 0x04}};

OBJCONST TSDOINFOENTRYDESC    OBJMEM asEntryDesc0x1C00[2] = {
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ},
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ}};
```

**Using SSC Tool V5.12** 
"coeappl.h" structures the SyncManager type info like this:
```
typedef struct OBJ_STRUCT_PACKED_START {
UINT16 u16SubIndex0; /**< \brief SubIndex 0*/
//*ECATCHANGE_START(V5.12)*/
UINT16 aEntries[2]; /**< \brief Entry buffer (one for two SyncManagers)*/
//*ECATCHANGE_END(V5.12)*/
} OBJ_STRUCT_PACKED_END
TOBJ1C00;
```

"coeappl.c" packs the SyncManager type info like this:
```
TOBJ1C00 sSyncmanagertype = {0x04, {0x0102, 0x0304}};

OBJCONST TSDOINFOENTRYDESC    OBJMEM asEntryDesc0x1C00[] = {
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ},
   {DEFTYPE_UNSIGNED8, 0x08, ACCESS_READ}};
```

The solution we found is to replace the V5.12 SyncManager structures and packing code with the code generated by SSC V5.11. After doing that and recompiling, the SyncManager types enumerate correctly. 

FYI: We chose not to use the V5.11 code is because by the time we found this issue, we had already invested a good deal of effort into porting the auto-generated code from V5.12 to play nice with the STM32 libraries we are using (specifically a custom HAL for the STM32F373). The level of effort required to port the code again seemed more difficult than trying to debug and replace a few lines of code.

Hope this helps someone in the future.

ArthurKetels, 2020-08-18:
Thanks for finding the difference between SSC 5.11 and 5.12. To me the bug is clear.

`TOBJ1C00 sSyncmanagertype = {0x04, {0x0102, 0x0304}};`

The change from uint8_t aEntries[4] to uint16_t aEntries[2] created a endianness problem. The two are only the same in memory presentation on a big endian processor. If your slave microcontroller is little endian the byte order in each aEntries[] is reversed.

The proper solution would be:
`TOBJ1C00 sSyncmanagertype = {0x04, {htoes(0x0102), htoes(0x0304)}};`

Of course the macro htoes() should be properly defined in the target environment.

The problem with your patch is that it can go wrong with CompleteAccess of the object 0x1c00. If I am not mistaken the change from uint8 to uint16 is to convert the object easier to CA representation with automatic padding of entry 0.

arkorobotics, 2020-08-18:
Ah! Great point about the CA representation. I had a hunch it was endianness issue and I definitely like your solution more. 

I've modified my code to use `__builtin_bswap16` and switched the type back to UINT16:
`TOBJ1C00 sSyncmanagertype = {0x04, {__builtin_bswap16(0x0102), __builtin_bswap16(0x0304)}};`

P.S. - Do you mean `htons()` instead of `htoes()`?

ArthurKetels, 2020-08-18:
> P.S. - Do you mean htons() instead of htoes()?
Indeed, I was mixed up with the conversion used in SOEM HostToEthercatShort htoes(). Here you need the other one, HostToNetworkShort. EtherCAT is little endian, network is big endian.

Using the buildin feature will fail on some targets, and does certainly fail on big endian controllers. A good system lib will use the buildin functions for htons() where appropriate. And non of the uses in SSC are time critical, so why care. Portable code is by far more important.

nakarlsson, 2020-10-22:
close due to inactive

closed 2020-10-22 by nakarlsson

cheeryguo, 2025-01-10:
> We stumbled across the same problem when using SSC Tool V5.12 with the LAN9252 and found another working solution:
>
> We had an issue where the SM2 and SM3 types would appear swapped when scanning for [...]

Yes, You did. It helped me in 2025. Thanks very much!