OpenEtherCATsociety/SOEM #402: index:0016.00 error:06020000 The object does not exist in the object directory

joaosbastos, 2020-05-22:
Hello, i'm starting to use SOEM on windows and had success configuring a remote I/O from Helmholz. Now i'm trying to add a servo drive from FESTO, although it should be more simple because it doesn't have modules it is giving me a hard time. I need some hint / comments / solutions, so i attach the ESI, the result of slaveinfo -sdo and the result of slaveinfo -map. 
Thank you.

[Festo-CMMT-ST-CiA402-20200113.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4669776/Festo-CMMT-ST-CiA402-20200113.txt)
[sdo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4669779/sdo.txt)
[map.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4669786/map.txt)

ArthurKetels, 2020-05-22:
The problem is with object 0x1c12 (SM2 assignment) and 0x1c13 (SM3 assignment). The reference to the PDO mapping objects is a uint16_t but the bytes are swapped. Object 0x0016 indeed does not exists, but object 0x1600 does, and was the intended object.

2 solutions : a) ask Festo for a firmware update to correct this error. b) rewrite the objects (they are writable) in the PO2SO hook (search in issues how to do that)

joaosbastos, 2020-05-25:
Thank you @ArthurKetels.
I inverted the bytes on the index of SMs and variables, and some errors disappeared. I also disconnected the other slaves, now only the problematic one is on the network. I attach the new log, with the remaining errors.
[log2.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4676526/log2.txt)

ArthurKetels, 2020-05-25:
From the log I can see that actually the objects 0x1c12 and 0x1c13 are not rewritten to the correct mapping. From your log:

>  Index: 1c12 Datatype: 0006 Objectcode: 08 Name: Sync Manager 2 PDO Assignment
  Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 000f Name: SubIndex 000
          Value :0x01 1
  Sub: 01 Datatype: 0006 Bitlength: 0010 Obj.access: 000f Name: SubIndex 001
          Value :0x0000 0
 Index: 1c13 Datatype: 0006 Objectcode: 08 Name: Sync Manager 3 PDO Assignment
  Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 000f Name: SubIndex 000
          Value :0x01 1
  Sub: 01 Datatype: 0006 Bitlength: 0010 Obj.access: 000f Name: SubIndex 001
          Value :0x0000 0

As you can see both objects have value 0x0000. And of course this leads to a config error in the slave.

joaosbastos, 2020-05-26:
I made some more tests
[https://github.com/Apollo3zehn/EtherCAT.NET/issues/49](url)

ArthurKetels, 2020-05-26:
Why make it so difficult for yourself? Just search the subject here. Others had the same issue and solved it. #379

nakarlsson, 2020-07-14:
@joaosbastos , can we close this?

closed 2020-07-15 by joaosbastos