OpenEtherCATsociety/SOEM #403: rtelligent r60 ethercat stepper driver

Houman-HM, 2020-05-27:
Dear all, 

I want to use rtelligent r60 EtherCAT stepper driver to control a stepper motor.
When I try running the sample test code, it does not bring the EtherCAT into operational state. And spits out the following:
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 22 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x0025 : Invalid output mapping
Request init state for all slaves
End simple test, close socket.

I'm a newbie when it comes to working with EtherCATs and I have no clue how to go further.
I was wondering if anyone can help me get this working.

Thanks in advance.

ArthurKetels, 2020-05-28:
The slave has no output PDO defined. And a stepper controller with no outputs is quite useless.

Please give some more info. The ESI file of the slave (xml), the output of slaveinfo -map and slaveinfo -sdo

Houman-HM, 2020-05-28:
Dear Arthur,
Thanks for the quick reply.
I have uploaded the information you asked in the following link.
https://github.com/Houman-HM/ethercat_control/tree/master/Documentation
I really appreciate if you could help me get this running. 
Thanks.

ArthurKetels, 2020-05-29:
The actual mapping in the slave is different from the one in the ESI file. This is a recipe for disaster. Did you modify the PDO mapping? If not then the default settings in the slave are wrong. You have to reset the PDO mapping according to the ESI file. You can search this forum how to do that. It does require some detailed EtherCAT programming knowledge.

Houman-HM, 2020-05-30:
Thanks for the info and help Arthur. 
Well, I don't know if I have managed to change the PDO mapping while tryig different stuff. However,  I got the error about Invalid output mapping in the very first time I ran the simple_test code, after which I started fiddling around with it.  Having that said, is there any piece of software, whether for Linux or Windows, which would get the ESI file and remap the PDOs in the right way?

Houman-HM, 2020-06-11:
Dear Arthur,
I tested the EtherCAT driver using EC-Engineer software and it works. I have not made any modifications in the EtherCAT's configuration. I was wondering if you could let me know if EC-Engineer software does some sort of an automatic output remapping, which would result in the EtherCAT driver reaching the operational mode and working properly.

ArthurKetels, 2020-06-11:
SOEM is different from most masters in that it does not use the ESI xml file to configure the slave but instead reads the configuration from the slave eeprom and CoE data. EC-Engineer reads the ESI file, configures the slave accordingly and then it works. As most slaves are only tested this way it seems they are fine. But the on-slave data is also mandatory, and should be exactly similar to the xml file. When not then of course the slave can not configure correctly with the default SOEM algorithm. 

This is not a bug in SOEM but in the slave. You should ask for an update from the vendor to rectify this bug. The other solution is to leave it as is and instead manually configure the slave as per the xml data. Search for other posts here how to do that. You are not the first, and I guess not the last facing this issue.

nakarlsson, 2020-07-14:
@Houman-HM , can we close this?

Houman-HM, 2020-07-14:
@nakarlsson Yes, thanks to @ArthurKetels, I managed to figure out where the problem is and trying to fix it.

closed 2020-07-14 by Houman-HM