OpenEtherCATsociety/SOEM #404: PDO Receiving works but sending not

robin843, 2020-05-29:
Hello,

I have a little test setup with multiple motors as slaves configured. When I run slaveinfo it shows me the correct PDO mapping (default one) and also when I try to receive packages it works and I got information from the motors.

But when I try to send data to the slaves nothing happens. I also looked at the wireshark file and the data to send is in there. That is why I'm confused. The manufacturer of the motors delivers also an app where i can check if a slave received packages and also what the current values are (which I want to change). There is says it receives 0 packages.

Maybe someone more experienced does see my error. I've already searched in issues but did not find this case. Thank you for your help! I have attached the slaveinfo and a wireshark file.

Best regards
Robin

[capture-output.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4701571/capture-output.zip)

This is the slaveinfo output:

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth0 succeeded.
9 slaves found and configured.
Calculated workcounter 27

Slave:1
 Name:MOVI-C MOVIDRIVE CiA402
 Output size: 112bits
 Input size: 72bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000059 ID: 0000000b Rev: 00000002
 SM0 A:1000 L:1024 F:00010026 Type:1
 SM1 A:1400 L:1024 F:00010022 Type:2
 SM2 A:1800 L:  14 F:00010064 Type:3
 SM3 A:1c00 L:   9 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  14 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000007e Ll:   9 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 1024 rd: 1024 MBX protocols : 0c
 CoE details: 0f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6060:0x00 0x08 INTEGER8     Modes of operation
  [0x0001.0] 0x0000:0x00 0x08
  [0x0002.0] 0x6040:0x00 0x10 UNSIGNED16   Control word
  [0x0004.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0008.0] 0x60FF:0x00 0x20 INTEGER32    Target velocity
  [0x000C.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x007E.0] 0x6061:0x00 0x08 INTEGER8     Modes of operation display
  [0x007F.0] 0x0000:0x00 0x00
  [0x007F.0] 0x6041:0x00 0x10 UNSIGNED16   Status word
  [0x0081.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x0085.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value

Slave:2
 Name:MOVI-C MOVIDRIVE CiA402
 Output size: 112bits
 Input size: 72bits
 State: 4
 Delay: 800[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
[... 279 lines cut ...]
 FMMU0 Ls:00000070 Ll:  14 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:000000c6 Ll:   9 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 1024 rd: 1024 MBX protocols : 0c
 CoE details: 0f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0070.0] 0x6060:0x00 0x08 INTEGER8     Modes of operation
  [0x0071.0] 0x0000:0x00 0x08
  [0x0072.0] 0x6040:0x00 0x10 UNSIGNED16   Control word
  [0x0074.0] 0x607A:0x00 0x20 INTEGER32    Target position
  [0x0078.0] 0x60FF:0x00 0x20 INTEGER32    Target velocity
  [0x007C.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x00C6.0] 0x6061:0x00 0x08 INTEGER8     Modes of operation display
  [0x00C7.0] 0x0000:0x00 0x00
  [0x00C7.0] 0x6041:0x00 0x10 UNSIGNED16   Status word
  [0x00C9.0] 0x6064:0x00 0x20 INTEGER32    Position actual value
  [0x00CD.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
End slaveinfo, close socket
End program

ArthurKetels, 2020-05-29:
Slaveinfo does not put the slave in Operational state (OP). This means inputs are working but outputs are disabled. After going to OP you also need to set Modes of Operation following the guidelines of DS402. 

May I ask if you have any experience with simpler EtherCAT slaves, or is this your first?

nakarlsson, 2020-07-14:
Inactive

closed 2020-07-14 by nakarlsson