OpenEtherCATsociety/SOEM #405: Non-blocking SDO queries

brinkap, 2020-06-03:
I've been using SOEM for a few months now and have been very happy with the performance and features available by the current release. Thanks for making the effort open-source!

My EtherCat application has some non-PDO mappable data objects that I'd like to access through COE SDOread() and SDOwrite() calls. During the real-time process loop, I cannot make use of these functions since they are blocking functions. Even if one or two SDOread() functions can make my process deadline, I'm afraid it will not scale to support an arbitrary number of devices on the EtherCat bus. 

Is there an existing mechanism in SOEM to asynchronously read and write SDOs? If not, I'd be willing to help contribute to this functionality with guidance from the SOEM maintenance team. Thanks!

nakarlsson, 2020-06-03:
Not in the current master branch. It might work by reading the SDOs in a low prio thread/task, depends on your application requirements.

ArthurKetels, 2020-06-03:
There is a reason the SDOread/write functions are blocking. They are not supposed to work in OP mode. SDO transmissions are meant to configure the slave. That it sometimes seems to work all right is not an excuse.

All real time data should be mapped to PDO. If that is not possible with your slaves then that is the problem. Improve the slave, Going on a path to include real time SDO is bound to fail and ultimately will not scale.

brinkap, 2020-06-04:
@nakarlsson I'd be afraid that this a) would create many pthread instances (one per SDO per device!) that could hurt performance from context switching and b) that the SDO functions would not be thread safe with the IOmap process exchange.

@ArthurKetels Thanks for your perspective. I agree blocking SDO read/writes greatly simplifies parameter setting during SafeOp to PreOp compared to an async SDO mechanism I've used in another EtherCat Master. My intent was not to gain real-time performance using SDO read/writes but to query SDOs at low frequency (e.g. 1 Hz) or perhaps after an infrequent event (e.g. get error code after faulting) Does your recommendation change in light of this use case?

If I follow the code through the blocking sections: SDOwrite() calls ecx_mbxreceive() which loops over a ecx_FPRD() until the timeout is reached. If I issue a ecx_mbxsend() that triggers the SDO request and later read the mailbox entry with ecx_mbxreceive(), could this be a possible work around? Admittedly, this would have high latency but that is acceptable for these types of queries.

ArthurKetels, 2020-06-07:
I am guessing your application but probably the best solution for now would be to follow nakarlsson advice. SOEM is build to be thread safe (except for PDO). You could make one thread that handles SDO transfers. Add a queue so that multiple requests can be handled. Synchronize the SDO task from the PDO task in such a way that only one SDO request can be handled per PDO cycle. By serializing this way you could prevent bandwidth over use by the SDO task,

nakarlsson, 2020-07-14:
@brinkap , can we close this?

brinkap, 2020-07-14:
Yes, I followed your recommendations have async SDO RW working for my application, performance is great! Thanks for the support!

In case others have the same question I did, I implemented the SDO read/write a separate thread that wakes up on a pthread_cond_signal sent by the PDO thread when the application needs to issue a new SDOread/write device instruction. The PDO thread fills out a 'SDO request' struct that contains the device id and relevant SDO read/write arguments and pushes this request struct to a locking circular queue. Once the message is pushed, the PDO thread signals the SDO thread and continues spinning. When the blocking SDO function returns, an 'SDO response' struct is populated and pushed to a separate response circular queue, where it will eventually be serviced by the main PDO thread that polls the response queue for new messages. The SDO thread goes back to sleep and waits for the next request to come on the request queue.

closed 2020-07-14 by brinkap