OpenEtherCATsociety/SOEM #406: SOEM license is problematic.

jeremiah, 2020-06-03:
The license file for this software (https://github.com/OpenEtherCATsociety/SOEM/blob/master/LICENSE) combines a proprietary IP regime with the GPL v2 in a manner that violates the GPLv2. The GPL does not allow any additional restrictions to be placed on the program. From the GPLv2, section 6;

"You may not impose any further restrictions on the recipients' exercise of the rights granted herein."

With the requirement to have an additional EtherCAT Master License, you violate the GPL's terms.

ArthurKetels, 2020-06-03:
You are confusing copyright law and patent law. GPL is about copyright. The EtherCAT master license is about patents. If in doubt, ask the legal department of Beckhoff GmbH.

jeremiah, 2020-06-04:
I don't think I am. GPL v2 has an implicit patent grant. Here's a good explainer: https://copyleft.org/guide/comprehensive-gpl-guidech7.html

The SOEM license as it stands is a violation of GPL in my opinion and I would strongly advise you seek a third-party review. If you are in violation of the GPL you lose your right to use GPL'd software which is a lot of software, the Linux kernel included.

ArthurKetels, 2020-06-04:
I know that article, but thanks for reminding me again. The point here is that we talk about patents that cover the code in question and are held by the licensor.  In this case the patents in question are first not property of of the copyright owners of SOEM and second do not cover the functions of SOEM. The statement about the EtherCAT master license is only there to make users of SOEM aware of possible patent implications when using the software (SOEM) in combination with other hardware (slaves). The software (SOEM) itself is free to use according to GPL v2 and is only amended to grant extra freedoms.

As an example : using Linux on a computer that controls a chemical assembly machine will not grant you a license to make patented drugs.

jeremiah, 2020-06-05:
Okay, but perhaps the potential patent infringement and subsequent penalty might be moved somewhere else like the README? It is confusing to have this in the LICENSE file. I say this because I'm someone who is responsible for open source usage in various companies and a colleague wants to use this software and I'm going to have a hard time convincing our lawyers that it's okay.

ArthurKetels, 2020-06-06:
Fair point. I am going to look into this.
Can we then close this issue?

jeremiah, 2020-06-08:
Yes, of course.

closed 2020-06-08 by jeremiah