OpenEtherCATsociety/SOEM #407: Configuring the DC when we have multiple slave groups

davoodmoradi, 2020-06-08:
Hi,
I am having problem configuring the DC synchronization when i have multiple slaves groups. 
If i set all my slaves in one group (group0) i can activate the DC option and everything works great. In my case, I have two groups, group 1 includes a servo drive (slave 1) which is DC capable and cycle time is 2 ms. Group two has some Beckhoff digital and analog IOs (no DC capable module) with 1 ms cycle time. I don't care about the group two, but want to synch the servo drive. 
Is there any procedure i need to follow to make it happen?
I appreciate your help in advance.
Davood Moradi

ArthurKetels, 2020-06-08:
DC works over all groups. This means that all slaves synchronize to the first DC capable slave. Only one packet is in control of that. The FRMW command (configured read multiple write) of the slave clock. That slaves are logically mapped in different groups does not matter.

Please send a wireshark trace of your setup if you cannot get it to work properly.

nakarlsson, 2020-07-14:
@davoodmoradi , can this be closed?

davoodmoradi, 2020-07-14:
Hi,
Still having the problem, is there an example that shows how to do it? 
Regards
Davood Moradi

nakarlsson, 2022-01-14:
No example available, close due to inactivity

closed 2022-01-14 by nakarlsson