OpenEtherCATsociety/SOEM #408: SOEM & BOSCH Servo Drive

davoodmoradi, 2020-06-09:
Hi, 
I am having some problem running a Bosch servo drive (CSB02 with multi Ethernet). I can make it work and everything seem to be OK, i can run it, but every 2-3 minutes i am receiving the wrong WKC. the working counter supposed to be 3 and every 2-3 minutes i am getting a couple of packets with working counter 2. i tested it with and without DC option, same result. 
the same system works great with a couple of other servos (Schneider LXM32, ABB ACSM1).
any idea what can cause this issue?
best regards
Davood Moradi

nakarlsson, 2020-06-09:
Does the slave enter SAFEOP + E with an AL StatusCode?

perhaps you can share a wireshark capturing the issue.

A loss of WKC 3->2 seems strange, like the SM3 would be deactivated?

davoodmoradi, 2020-06-09:
Hi,
Thanks for your reply, 
The drive goes all the way to operational state and i can run it, no issue. But after a minute or more I am getting the wrong WKC and fault on the drive (the WKC must be 24 and it gets 23). Attached is the wireshark capture. It starts getting the wrong WKC at around 50 second from start.
I appreciate your help in advance
Davood Moradi
[WireShark1.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4753189/WireShark1.zip)

ArthurKetels, 2020-06-09:
Thanks for the wireshark trace.

The reason you could even see WKC 23 (reduction of only one) is in the SM configuration.
 
>Frame 3056: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface 0
Ethernet II, Src: 03:01:01:01:01:01 (03:01:01:01:01:01), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
EtherCAT frame header
EtherCAT datagram(s): 'FPWR': Len: 8, Adp 0x1001, Ado 0x818, Wc 1 
    EtherCAT datagram: Cmd: 'FPWR' (5), Len: 8, Adp 0x1001, Ado 0x818, Cnt 1
        Header
        SyncManager: 0011060062000100
            SM Start: 0x1100
            SM Length: 0x0006
            SM Ctrl/Status: 0x0062, OpMode: 1 buffer, Access: Read, PDI IRQ, Watchdog trigger, 3 buffer state: 1. buffer
                .... .... .... ..10 = OpMode: 1 buffer (0x2)
                .... .... .... 00.. = Access: Read (0x0)
                .... .... ...0 .... = ECAT IRQ: False
                .... .... ..1. .... = PDI IRQ: True
                .... .... .1.. .... = Watchdog trigger: True
                .... ...0 .... .... = IRQ write: False
                .... ..0. .... .... = IRQ read: False
                .... 0... .... .... = 1 buffer state: Read
                ..00 .... .... .... = 3 buffer state: 1. buffer (0x0)
            SM Enable: 0x0001, Enable
        Working Cnt: 1
Pad bytes: 000000000000000000000000000000000000000000000000

As you can see SM4 of the first slave (x0x1001) is programmed as 1-buffer read. This is normally exclusively used for mailboxes. The semantics of a 1-buffer are different from a 3-buffer. As long the slave is busy writing the buffer or the master has already read the buffer once the SM is disabled from the master side. As a result the WKC is not increased by 1. This is also proven by the PDO data at the WKC=23, there is no input data visible from slave 0x1001.

Remedies:
a) Nothing, this is expected behavior from the slave. 
b) The SM config bits are not written correctly in the slave SII. Actually as 3-buffer is intended. Ask for a firmware fix.
c) Vendor is not responding, fix it yourself by writing the correct config bits for SM4 in the PO2SOhook. The SM configuration is stored in the ec_slave struct.

davoodmoradi, 2020-06-09:
Thanks Arthur,
I will try to fix it by configuring the SM4 and let you know the result.
Regards
Davood Moradi

nakarlsson, 2020-06-11:
here is an example that change SM settings, https://github.com/hefloryd/SOEM/commit/22d8f5d

ArthurKetels, 2020-06-11:
Correction on my side. I intended to say 4th SM, this is SM[3] counting from zero.

davoodmoradi, 2020-06-12:
Hi Arthur,
Thanks a lot, I managed to modify the SM configuration. 
where can i find the detailed description about the SM flags?
what is the OpMode for 3 buffer read?
Regards
Davood Moradi

nakarlsson, 2020-06-12:
You'll find info here: https://download.beckhoff.com/download/Document/io/ethercat-development-products/ethercat_esc_datasheet_sec2_registers_2i8.pdf

SyncManager settings  adress 0x804-..
Control, Status, Actiate, PDI

davoodmoradi, 2020-06-12:
Hi Arthur,
Thanks you so much, I modified the SyncManager and resolved the issue. It seems everything works OK. No more ether-cat communication error or wrong WKC.
Appreciate your help
Davood Moradi

closed 2020-06-12 by davoodmoradi

ArthurKetels, 2020-06-12:
It would be nice if you could post the code snippet that made it work. That way others can learn too.