OpenEtherCATsociety/SOEM #409: expectedWKC formula

balazskiss90, 2020-06-16:
Hello,

I started dealing with Ingenia Everest XCR motor drives, and I was able to run a successful customized simple_test with an actual position reading, but I needed to do the following modifications:
For some reason the modules started communicating with the master only after I blocked the LRW commands. 
However, in that case the other issue was the value of the expectedWKC. Therefore, I modified the formula according to the following:

`expectedWKC = (ec_group[0].outputsWKC) + ec_group[0].inputsWKC; /* changed from (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC; */`

As I found it in Beckhoff´s documentation the LRW command in the case of successful read and write adds +3 to the WKC/slave. Because I´m not using LRW, only LRD/LWR commands, which only increments the WKC (+1)/slave, I changed the formula as it is in the code above. 

Am I right in this case or should I look for an issue somewhere else?

Please, consider the fact that I am new to the EtherCAT communication protocol. I appreciate any help in the issue above. Thank you in advance!

Best regards,
Balazs

nakarlsson, 2020-06-16:
Hi, it makes sense, the application should calulcate the expectedWKC based on knowing what LRW / LRD /LWR is used, 

The info used by SOEM is context->grouplist[group].blockLRW

balazskiss90, 2020-06-17:
Hi, thank you for your quick response. I´ll make it as a condition then. I just wanted to make that I´m on the right track.

closed 2020-06-17 by balazskiss90

stevedanomodolor, 2020-07-22:
Were you able to get the ingenia drivers to work, I am currently working on the same drivers and seem to get the same problem, when I try to block the LRW command, it does not seem to go to operational state

balazskiss90, 2020-07-23:
Hi, Yes, at my side it is working. The Ingenia drives can reach OP_STATE. Some tips:

- make sure that your PDO mapping is correct;
- if you block the LRW commands, you should change the wkc calculation as it is above;
- for receiving process data I changed the time out:
`wkc = ec_receive_processdata(EC_TIMEOUTRET * 100);`

stevedanomodolor, 2020-07-23:
@balazskiss90  Why wkc = ec_receive_processdata(EC_TIMEOUTRET * 100)? is there a specific reason for multiplying it by 10 ?

balazskiss90, 2020-07-23:
@stevedanomodolor I just added extra time, because currently I am using a network adapter, which is not really suitable for RT purposes.

stevedanomodolor, 2020-07-23:
@balazskiss90  Did you have issue with the sm2 and sm3 type reversed, I am having the same issue with ingenia drivers and I cant read the data correctly, what firmware version are you using, 

 Name:Servo Drives
 Output size: 104bits
 Input size: 120bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
```
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  15 F:00010064 Type:4
 SM3 A:1c00 L:  13 F:00010020 Type:3
```
 FMMU0 Ls:00000000 Ll:  13 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000d Ll:  15 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
 ` SM2 inputs`
     addr b   index: sub bitl data_type    name
  [0x000D.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000F.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  [0x0010.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  [0x0012.0] 0x60B2:0x00 0x10 INTEGER16    Torque offset
  [0x0014.0] 0x21EF:0x00 0x20 REAL32       Interpolation time mantissa
  [0x0018.0] 0x21F0:0x00 0x20 REAL32       Interpolation time exponent
`  SM3 outputs`
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
  [0x000B.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
End slaveinfo, close socket
End program

balazskiss90, 2020-07-24:
@stevedanomodolor I am using the latest EtherCAT firmware, and I did not get this issue.  Unfortunately, I cannot help you with that, as I stated above I am also new to EtherCAT, therefore I recommend you to create a ticket here, or ask help from 'support@ingeniamc.com'.

stevedanomodolor, 2020-07-24:
I know, me too, but can you tell me the firmware version you downloaded?,
the latest one has an issue with the sm and type.

balazskiss90, 2020-07-27:
@stevedanomodolor I am using the 1.6.0 EtherCAT firmware, but I did not have this issue, that you described, and I also did not find it on their website. 
https://doc.ingeniamc.com/eve-xcr/downloads/firmware-1-6-0