OpenEtherCATsociety/SOEM #413: PDO dimension problem

AlexCorrado, 2020-06-26:
Goodmorning to all,

I'm using a Pc with Ubuntu as Ethercat master and an ethercat slave which uses a LAN9252. The slave is connected to a pc with window which is used to load the SOES program and once the software has been loaded, it remains connected in debug mode for checking the variables.  I'm using the last version of Ethercat SOES. I have a problem with the PDO lenght. The problem is that if I define in the EEPROM of the slave a PDO with 128 fields where each field has dimension 1 Byte, when i run the program simpletest.c everything works properly, but if I define a PDO with 128 fields where each field has dimension 4 Byte, it doesn't work. I'm doing this because the EEPROM doesn't support PDO with more than 128 fields hence I have increased the size of each field. I need a PDO with 512 Byte. Someone can help me? Is it possible to do this? I share the screen of command window when i run the simpletest program and the wireshark file.

NOTE:

- To solve the reading/writing problem on the SOES I thought I could manually change the 
   byte size of input/output to 512 even if the fields are 128, however the fact remains that 
   I don't know how to change the SOEM program.

Thank You!

[image]

[image]

[image]

[wire.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4835746/wire.zip)

AlexCorrado, 2020-06-26:
This is the screen of the slaveinfo.c with -map

[image]

nakarlsson, 2020-06-26:
Please close https://github.com/OpenEtherCATsociety/SOEM/issues/410 or address the question in https://github.com/OpenEtherCATsociety/SOEM/issues/410.

The slave implementation is unclear, if you have SDO issues you should sort that out before digging in PDO issues.

AlexCorrado, 2020-06-29:
@nakarlsson sorry for the 2 posts, in the last implementation I used a PDO with size 128 Byte and there was no problems but now, before proceeding with SDO management, I need to increase the size of the PDO. My priority has momentarily changed. The SOES run on a STM32 F4 and we have done a personalized driver for the LAN9252, but it start ok.

AlexCorrado, 2020-06-30:
I defined a PDO with 128 fields where each field has dimension 4 Byte (unint32 variables) and the slaveInfo.c program now work and the result is presented in following screen:

https://lh3.googleusercontent.com/-FrZ4L136TR4/XvtRxF31ZWI/AAAAAAAAAC0/_SbUhOn6o046qYB73cY4_E2YEhK7kngMgCK8BGAsYHg/s0/2020-06-30.png
...
[image]
...
It's possible to see that the Ibytes/Obytes variables indicate the right dimension of the PDO or rather 512 Byte. The problem is that the simpletest program doesn't work again. Is this a problem of SOEM that doesn't accept PDOs with these dimensions? Or something else is wrong?
For the slave program i used the value 512 as the number of byte for the input and the output as mentioned above.

nakarlsson, 2020-07-11:
The above Application isn’t a SOES implementation, right? SOES only support a traditional SM0/1 as mailboxes setup. ProcessData at SM2...

AlexCorrado, 2020-07-13:
The above application is based on SOES but I have used only PDO function until now. 
I noticed that with a PDO data size greater than 128 Bytes, the simpletest.c doesn't work.
I close this topic and I'm going to continue the discussion on the topic #410.
Thank you.

closed 2020-07-13 by AlexCorrado

nakarlsson, 2020-07-13:
Yes, try to create a standard slave, it’ll save you time in the end. 

About the above issue, you need to consider that the processdata SMs make use 3xsize of RAM. Having 512 Bytes of data will need SM physical addresses to be at offset size x3.

AlexCorrado, 2020-07-13:
Thank you, I tried as you said with SM0 at index 0x1000 and SM1 at index 0x1600 and it works. This problem is solved, now remains to solve the SDO problem with the 4 SM. But thank you!