OpenEtherCATsociety/SOEM #414: Master application with SCHED_DEADLINE

alessio76, 2020-06-29:
Hello everybody,
I'm dealing with a strange issue when I try to run my master application under SCHED_DEADLINE policy (I'm using the PREEMPT_RT patch on Ubuntu kernel version 4.19.100 with rt41 patch), so I don't know if this is the right place to submit this issue; if so I apologize.
I've successfully written an EtherCAT master based on the red_test example to drive a brushless DC motor: the master has a main thread that handles the setup of the slave and just after the slave enters in OP state this thread spawns the RT one that takes care of the process data communication. 
At first I used the SCHED_FIFO policy for the RT-thread and here everything is fine. Then I tried to use the SCHED_DEADLINE policy but randomly my system crashes and the only way to recover from this state is to reboot my PC with the power button. 
The core of the RT-thread is:

#define DEADLINE 1000000
#define RUNTIME 900000
...
OSAL_THREAD_FUNC ecatthread(){
int i=0;
struct sched_attr attr;
attr.size=sizeof(attr);
attr.sched_nice = 0;
attr.sched_priority = 0;
attr.sched_policy   = SCHED_DEADLINE;
attr.sched_runtime  =  RUNTIME;
attr.sched_period = DEADLINE;
attr.sched_deadline = DEADLINE;
attr.sched_flags = 0;
ret = sched_setattr(0, &attr, 0);
        
        if (ret < 0) {
            perror("sched_setattr failed to set ");
            exit(-1);
           }
/* this cycle takes the servo to OPERATION ENABLED state in order to enable the movement of the axis */
while(check!=1){
	   time1=ec_DCtime;
           //reset the synchronized deadline
	   refresh_deadline(&attr,(uint64)(DEADLINE+toff));	
           //wait for next cycle	
	   sched_yield();
	   wkc=ec_receive_processdata(EC_TIMEOUTRET);
           //set the right command to the slave
	   check=Servo_state_machine(MOVE);
           //synchronize Linux clock and DC
	   ec_sync(ec_DCtime, DEADLINE, &toff);	
	   ec_send_processdata();
           time2=ec_DCtime;
           cycle=time2-time1;
       }

ec_send_processdata();
while(i<TRAJECTORY_SAMPLES){
      time1=ec_DCtime;
      //reset the synchronized deadline
      refresh_deadline(&attr,(uint64)(DEADLINE+toff));	 
      //wait for next cycle
      sched_yield();
      wkc = ec_receive_processdata(EC_TIMEOUTRET);
      //synchronize Linux clock with DC
      ec_sync(ec_DCtime, DEADLINE, &toff);
      misured_position[i]=in_EPOS->position_actual_value;
      misured_current[i]=in_EPOS->current_actual_value;
      misured_velocity[i]=in_EPOS->velocity_actual_value;
      misured_torque[i]=in_EPOS->torque_actual_value;
      tv[i]=ec_DCtime;
      out_EPOS->target_position=target_position_abs[i];
      ec_send_processdata();
      time2=ec_DCtime;
      cycle=time2-time1;
      cicli[i]=cycle;
     }

Here the refresh_deadline function:

void refresh_deadline(struct sched_attr *attr, uint64 addtime){
     attr->sched_runtime=RUNTIME;
     attr->sched_deadline=addtime;
     attr->sched_period=addtime;
		
	 if(sched_setattr(0,attr,0)<0){
			 perror("sched_setattr failed");
			 exit(-1);
			 }
}

I don't read errors from `sched_setattr()` and I tried to run this thread even without calling `ec_sync()` and `refresh_deadline()`, but the PC continued to crash. So I tried to set the DEADLINE to 2000000 and RUNTIME to 1500000, but this doesn't solve the problem.
Do you have any idea about what might cause these crashes?

ArthurKetels, 2020-06-29:
Interesting exercise, but I am not sure this is worth pursuing. Synchronizing with EtherCAT is about absolute time, SCHED_DEADLINE works with relative times. SCHED_DEADLINE does not improve latency over SCHED_FIFO. Most of the time the latency problems are more in the NIC driver layer and not the EtherCAT rt-task. 

First read up on SCHED_DEADLINE : https://www.kernel.org/doc/Documentation/scheduler/sched-deadline.txt

You can not mix relative time with absolute time : ec_sync(ec_DCtime, DEADLINE, &toff); is wrong.

You abuse the SCHED_DEADLINE principles. Your runtime is crazy high. Fixing it up with sched_yield() is not proper.

A nice article on how to reduce network latency : https://blog.cloudflare.com/how-to-achieve-low-latency/

alessio76, 2020-06-30:
Thank you for your useful advices. In fact, in one of the "lucky" test of the program I noticed that even with this new policy the calculated jitter wasn't much better than the one with SCHED_FIFO.

closed 2020-06-30 by alessio76