OpenEtherCATsociety/SOEM #417: Question about master transition to OP when some devices doesn't support OP state

asor, 2020-07-13:
Hello, 

I have one principal question about SOEM master. If there are slave devices on the bus which could not switch its state to SAFE_OP and OP is it possible master to go to OP state without error leaving these devices in PRE_OP and switching other to OP?

Thank you in advance!

nakarlsson, 2020-07-13:
Sure, the master can continue to run with other slaves in OP. How to handle the ”error” is up to you, the expected working counter would be lower than expected, still up to you what is expected. The simple_test would try to act on this scenario as an error, but thats how it is written.

asor, 2020-07-20:
Thank you for the answer.

nakarlsson, 2020-10-23:
I'll close , write a new issue if you got more questions

closed 2020-10-23 by nakarlsson