OpenEtherCATsociety/SOEM #418: Difference between ec_SDOwrite and ec_SDOxwrite?

stevedanomodolor, 2020-07-14:
I am trying to use setup some of the parameters for the Pdo mapping but cannot see the main difference between the ec_SDOwrite and ec_SDOxwrite, does the first map objects from the CoE object dictinary?, what difference does it make using the other one?

nakarlsson, 2020-07-14:
Do you mean ecx_SDOwrite? Instead of ec_SDOxwrite. 

If, they do the same, the ecx_ version make use of the SOEM ver 2 API using a user instatiated context.

Either of the SDOWrite should work to write PDO config data.

stevedanomodolor, 2020-07-14:
Yes, the ecx_sdowrite, please what do you mean by user instantiated context?

stevedanomodolor, 2020-07-14:
Are there any examples of how to implement that context?

nakarlsson, 2020-07-14:
You can check the SOEM ver 1 API instance in ethercatmain.c
Look for ecx_contextt  ecx_context and how it is setup.

Or

This FSoE sample, https://github.com/rtlabs-com/fsoe-soem-demo/tree/master/test/linux/fsoe_sample

closed 2020-07-15 by stevedanomodolor