OpenEtherCATsociety/SOEM #421: EoE potentially invalid check data_offset > eoedatasize in ethercateoe.c

mangodan2003, 2020-07-17:
I have been experimenting with the tests/linux/eoe_test/eoe_test.c file and foudn that to make the GetIP work I had to comment out the check 

> if(data_offset > eoedatasize)

around line 300

The SSC 5.12 Doesn't seem to expect this to match.

Note that there is also a bug in the SSC re reading the IP address from aIpAdd in the event where MBX_16BIT_ACCESS is not set, see post in ETG Dev Forum.

nakarlsson, 2020-07-17:
Can you share a Wireshark and also print the values in the check please.

mangodan2003, 2020-07-21:
I'm unable to do this at present but I think it should be pretty obvious from the two code bases. 
In SOEM ethercateoe.c line 300 it checks for (data_offset > eoedatasize), where data_offset is the cumulative size of all the parameters that **might** be in the  payload. Where as eoedatasize is set from mbxheader.length calculated by the slave (SSC 5.12 eoeappl.c) , a fixed amount (10) and the parameters it **actually** included (MAC Address and IP only) which is 10 + 6 +  4.
I suspect the `data_offset += 4;` (or similar as appropriate) should be inside the scope of the check against flags for each possible parameter.

nakarlsson, 2020-07-21:
Sure, I agree that makes sense. Still, I’d like to get a reference from SSC if/when possible.

SetIP/GetIP functions was a bit fuzzy when we add them. Wireshark couldn’t dissect frames that just included partial data and so on. TwinCAT also added some wired empty data.

So, if you can include some reference data we can have another go at it and make it work on all sides.

nakarlsson, 2020-07-21:
You can perhaps try to move the "data_offset += x" in respective scope, to verify? I don't have any uptodate SSC device to test with, the one I got works.

PR #431 (2020-08-17): EoE and Distributed clock fixes
EoE fix
Obvious fix, the optional IP parameters length should
only be included in total length if they are included.
fixes #421

DC fix
The correct factor for seconds should be 1000000000
fixes #432
commit bae37b90 (2020-08-17): Don't always include optional IP parameter lengths

Obvious fix, the optional IP parameters length should
only be included in total length if they are included.

fixes #421
PR #436 (2020-08-18): Don't always include optional IP parameter lengths
Re-apply the EoE fix fom the reverted PR /revert-431-master

Obvious fix, the optional IP parameters length should
only be included in total length if they are included.

fixes #421
commit 101ac54a (2020-08-18): Don't always include optional IP parameter lengths

Obvious fix, the optional IP parameters length should
only be included in total length if they are included.

fixes #421
closed 2020-08-18 by ArthurKetels