OpenEtherCATsociety/SOEM #422: compiling code multiple time causes one object to produce error -

stevedanomodolor, 2020-07-20:
I am currenlty doing the PDO mapping for an ingenia mc Ethercat driver Everest XCR driver. Everything does well except when I try to modify one object. When i compile the code multiple times, after 5 times for examples, it produces this error: Is there a reason why and would this a problem?

Time:   22541.080 SDO slave:1 index:2106.00 error:08000020 Data cannot be transferred or stored to the application

Code

Code added to slaveinfo example 
slaveinfo function

```
 //PDO mapping for each slaves from vendor ID and product code
            for (int slc = 1; slc <= ec_slavecount; slc++)
            {
                // Printing drive info
                printf("%x\t%x\n", ec_slave[slc].eep_man, ec_slave[slc].eep_id);

                if ((ec_slave[slc].eep_man == 0x0000029c) && (ec_slave[slc].eep_id == 0x02c30001))
                {
                    printf("Found %s at position %d\n", ec_slave[slc].name, slc);
                    // ec_slave[slc].PO2SOconfig = &servo_drive1_PDO_MAP;
                    int revl = servo_drive1_setup(slc);
                    printf("Revl: %d\n", revl);
                }
            }
            // configure IOMap
            int iomap_size = ec_config_map(IOmap);
            printf("SOEM IOMap size: %d\n", iomap_size);
```

 ```

int servo_drive1_setup(uint16 slave)
{

    int ret = 0;
    // int retRead = 0;
    uint8_t dType_8;
    uint32_t dType_32 = 0;
    uint16_t dType_16 = 0;
    float dType_float = 0.0;
    //int size_8 = sizeof(dType_8);
    // int size_32 = sizeof(dType_32);
    // int size_16 = sizeof(dType_16);
    // int size_float = sizeof(dType_float);

    /* Basic motor configuration */
    /* Motor configuration */
    // Motor pair
    // Value - 1
    dType_16 = 1;
    ret += ec_SDOwrite(slave, 0x2106, 0x00, FALSE, sizeof(dType_16), &dType_16, EC_TIMEOUTSAFE);
    // retRead += ec_SDOread(slave, 0x2106, 0x00, FALSE, &size_16, &dType_16, EC_TIMEOUTSAFE);
    // printf("Motor pair: %d\n", dType_16);
    // Rated current
    // Value - 2.3
    dType_float = 2.3;
    ret += ec_SDOwrite(slave, 0x2100, 0x00, FALSE, sizeof(dType_float), &dType_float, EC_TIMEOUTSAFE);
    // retRead += ec_SDOread(slave, 0x2100, 0x00, FALSE, &size_float, &dType_float, EC_TIMEOUTSAFE);

    // Current direct set-point
    // Value - 0
    dType_float = 0.0;
    ret += ec_SDOwrite(slave, 0x201B, 0x00, FALSE, sizeof(dType_float), &dType_float, EC_TIMEOUTSAFE);
    // retRead += ec_SDOread(slave, 0x201B, 0x00, FALSE, &size_float, &dType_float, EC_TIMEOUTSAFE);
    // printf("Current direct set-point: %f\n", dType_float);
    // Current direct loop input offset
    // Value - 0
    dType_float = 0.0;
    ret += ec_SDOwrite(slave, 0x207D, 0x00, FALSE, sizeof(dType_float), &dType_float, EC_TIMEOUTSAFE);
    // retRead += ec_SDOread(slave, 0x207D, 0x00, FALSE, &size_float, &dType_float, EC_TIMEOUTSAFE);
[... 219 lines cut ...]
 FMMU0 Ls:00000000 Ll:  17 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000011 Ll:   5 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x0011.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x0013.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  [0x0014.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
  [0x000B.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
  [0x000D.0] 0x60F4:0x00 0x20 INTEGER32    Position following error
End slaveinfo, close socket
End program

```

nakarlsson, 2020-07-20:
By compile I guess you mean 5 runs?

Please, post a Wireshark when it fails.

stevedanomodolor, 2020-07-20:
Yes, running it 5 times, what do you mean by wireshark and how do I get the it?

nakarlsson, 2020-07-20:
How about asking the drive manufacturer for help? As mentioned, the drive produce the massage, they’ll know the reason if not stated in the documentation.

stevedanomodolor, 2020-07-20:
ok, I will do that now 
thank you

closed 2020-07-20 by stevedanomodolor