OpenEtherCATsociety/SOEM #423: Reading value for object is zero

stevedanomodolor, 2020-07-22:
After mapping the pdo and making sure it is correct using the slaveinfo example, I created a short program that 3 initial 3 threads, one for the pdo read/write, one for the error check and the last for the data printing. But I never seemed to set the ethercat CIA 402 state to operation mode, I noticed that the values of every objects was zero even the position reading. I modified the code to only read the position value before going to the cia operational state(for testing) but i seemed to get zero everytime. Is there a possible reason why this is happening?

[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4958306/wireshark.zip)

PDO mapping slaveinfo
```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth2 succeeded.
Slaves matched correctly
29c	2c30001
Found Servo Drives at position 1
Slave 1 configured correcty
Revl: 1
SOEM IOMap size: 28
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Servo Drives
 Output size: 104bits
 Input size: 120bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  15 F:00010064 Type:4
 SM3 A:1c00 L:  13 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:  13 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000d Ll:  15 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x000D.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000F.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  [0x0010.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  [0x0012.0] 0x60B2:0x00 0x10 INTEGER16    Torque offset
  [0x0014.0] 0x21EF:0x00 0x20 REAL32       Interpolation time mantissa
  [0x0018.0] 0x21F0:0x00 0x20 REAL32       Interpolation time exponent
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
  [0x000B.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
End slaveinfo, close socket
End program

```
This is the result of the program 

```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth2 succeeded.
1 slaves found and configured.
All slaves are present as expected!
[... 446 lines cut ...]
            }

            // printf("\nRequest init state for all slaves\n");
            // ec_slave[0].state = EC_STATE_INIT;
            // /* request INIT state for all slaves */
            // ec_writestate(0);
        }
        else
        {
            printf("No slaves found!\n");
            inOP = FALSE;
        }
        // printf("End simple test, close socket\n");
        // /* stop SOEM, close socket */
        // ec_close();
    }
    else
    {
        printf("No socket connection on %s\nExcecute as root\n", ecat_ifname);
        inOP = FALSE;
    }

    return inOP;
}
```

stevedanomodolor, 2020-07-22:
Reading the bytes directly from the inputs gives all zero 
00000000
00000000
00000000
00000000
00000000
00000000
00000000
00000000

ArthurKetels, 2020-07-22:
Thanks for all the data. Now it is easy to dig in and find out what is going on. That it is too much is not a problem, I can pick out the relevant pieces.

Actually we have a bug in the slave eeprom data. This is a well know bug in the Beckhoff SSC (see ethercat.org forum). The SM2 and SM3 type are reversed. From your slaveinfo output:

>  Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  15 F:00010064 Type:4
 SM3 A:1c00 L:  13 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:  13 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000d Ll:  15 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0

As we can see SM2 is defined as type 4 (inputs), but the flags have 00010064 (3 buffer write). SM3 is defined as type 3 (outputs), but the flags have 00010020 (3 buffer read).

SOEM programs the FMMUs according to the specified SM types. So it assigns FMMU0 to SM3 and sets the FMMU to write. It assigns FMMU1 to SM2 and sets it to read.

SOEM then brings the slave to OP.
During a PDO cyle following happens:
FMMU0 is triggered by the logical address 0000, PDO objects from the master (outputs) are transferred to SM3. But SM3 is actually configured in the read direction (from slave to master). Therefore the SM refuses the transfer and the WKC is not incremented.
FMMU1 is triggered by the logical address 000d, PDO objects from SM2 (inputs) are transferred to the master. But SM2 is actually configured in the write direction (from master to slave). Therefore the SM refuses the transfer and the WKC is not incremented.

So this is why there is no data transfer from master to slave and from slave to master, even while the slave remains in OP. 

The remedy is to change the SM types as defined in the eeprom SII to the correct values. SM2 type 3, SM3 type 4. Contact engenia to get your slave updated with the correct data. 

As a stop-gap you could change the SM type in your program.
connect a function to ec_slave[slave].PO2SOconfig
in that function do
ec_slave[slave].SMtype[2] = 3;  
ec_slave[slave].SMtype[3] = 4;

ArthurKetels, 2020-07-22:
For your (and engenia) reference about the SSC bug:
https://github.com/OpenEtherCATsociety/SOEM/issues/224#issuecomment-540666564

stevedanomodolor, 2020-07-23:
Thank you very much for your quick response, I have contacted with ingenia about the drivers to see wether there can fix it. 

I tried the stop-gap method, it seem to have no effect so far

```
int fix_SM(uint16 slave)
{
    ec_slave[slave].SMtype[2] = 3;
    ec_slave[slave].SMtype[3] = 4;
    return 1;
}

           for (int cnt = 1; cnt <= ec_slavecount; cnt++)
            {
                ec_slave[cnt].PO2SOconfig = &fix_SM;
                // ec_slave[slave].SMtype[2] = 3;
                // ec_slave[slave].SMtype[3] = 4;
            }

            int iomap_size = ec_config_map(IOmap);
```

Running the red_test, i still get the following result 

```
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.Ctime 3348286152297056495, dt         6495, O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00 00 00
OK : all slaves resumed OPERATIONAL.
processdata cycle 22545 , Wck   0, DCtime 3348286152518051715, dt         1715, O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00 00 00
```
And the type does not seem to change 

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Slaves matched correctly
SOEM IOMap size: 22
1 slaves found and configured.
Calculated workcounter 3
```

Slave:1
 Name:Servo Drives
 Output size: 88bits
 Input size: 88bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:4
 SM3 A:1c00 L:  11 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
WKibviabvabvoaovboabvoabvoabvo: 1
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000D.0] 0x607A:0x00 0x20 INTEGER32    Position set-point
  [0x0011.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
  [0x0015.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
WKibviabvabvoaovboabvoabvoabvo: 1
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
End slaveinfo, close socket
End program
```

stevedanomodolor, 2020-07-23:
UPDATE, I tried the fix in the issue you sent by modifying the thercatcoe file, and it seems to correct the Type ISSUE
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Slaves matched correctly
SOEM IOMap size: 22
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Servo Drives
 Output size: 88bits
 Input size: 88bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  11 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
WKibviabvabvoaovboabvoabvoabvo: 1
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000D.0] 0x607A:0x00 0x20 INTEGER32    Position set-point
  [0x0011.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
  [0x0015.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
WKibviabvabvoaovboabvoabvoabvo: 1
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
End slaveinfo, close socket
End program

ArthurKetels, 2020-07-23:
Could be that the PDO updates (LRW datagrams) during the SO->OP transition are too slow.
Please send a wireshark trace.

stevedanomodolor, 2020-07-23:
Thank you so much for answering me so far, really gratefull. 
What seems to work was when I blocked the LRW commands. What I do not understand is should I not reduce the workcounter to 2 since the lRW command is not longer there? 
         expectedWKC = (ec_group[0].outputsWKC ) + ec_group[0].inputsWKC;
This is the wireshark of thee working red_test
[wireshark3.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4964315/wireshark3.zip)

```
         for (int i = 1; i <= ec_slavecount; i++)
         {
            ec_slave[i].blockLRW = 1;
         }

         // for (int cnt = 1; cnt <= ec_slavecount; cnt++)
         // {
         //    ec_slave[cnt].PO2SOconfig = &fix_SM;
         // }

         ec_config_map(&IOmap);

         ec_configdc();
         printf("%d slaves found and configured.\n", ec_slavecount);
         /* wait for all slaves to reach SAFE_OP state */
         ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE);

         /* configure DC options for every DC capable slave found in the list */
         // ec_configdc();

         /* read indevidual slave state and store in ec_slave[] */
         ec_readstate();
         for (cnt = 1; cnt <= ec_slavecount; cnt++)
         {
            printf("Slave:%d Name:%s Output size:%3dbits Input size:%3dbits State:%2d delay:%d.%d\n",
                   cnt, ec_slave[cnt].name, ec_slave[cnt].Obits, ec_slave[cnt].Ibits,
                   ec_slave[cnt].state, (int)ec_slave[cnt].pdelay, ec_slave[cnt].hasdc);
            printf("         Out:%8.8x,%4d In:%8.8x,%4d\n",
                   *ec_slave[cnt].outputs, ec_slave[cnt].Obytes, *ec_slave[cnt].inputs, ec_slave[cnt].Ibytes);
            /* check for EL2004 or EL2008 */
            if (!digout && ((ec_slave[cnt].eep_id == 0x0af83052) || (ec_slave[cnt].eep_id == 0x07d83052)))
            {
               digout = ec_slave[cnt].outputs;
            }
         }
         expectedWKC = (ec_group[0].outputsWKC *2) + ec_group[0].inputsWKC;
```

Result 

```
SOEM (Simple Open EtherCAT Master)
Redundancy test
Starting Redundant test
ec_init on eth1 succeeded.
1 slaves found and configured.
1 slaves found and configured.
Slave:1 Name:Servo Drives Output size: 88bits Input size: 88bits State: 4 delay:0.1
         Out:00000000,  11 In:00000000,  11
Calculated workcounter 3
Request operational state for all slaves
Operational state reached for all slaves.
^Cocessdata cycle 29247 , Wck   3, DCtime 3348287776112542105, dt        -7895, O: 00 00 00 00 00 00 00 00 I: 50 02 01 69 04 00 00 00
```

stevedanomodolor, 2020-07-23:
Sorry for the amount of questions but, why do I keep getting this, does this mean that the connection was lost. I am still able to read the data now but keep getting this 

OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.

ArthurKetels, 2020-07-23:
This means the calculated workcounter is not correct. When CalculatedWKC != returnedWKC then the check function is called. It reads the slave state, sees nothing wrong prints the message and returns back to the PDO cycle, and on and on.

For your previous question about adjusted WKC for LRD/LWR, in the SOEM code : 

>     else if(rxbuf[idx][EC_CMDOFFSET]==EC_CMD_LWR)
         {
            if(idxstack->dcoffset[pos] > 0)
            {
               memcpy(&le_wkc, &(rxbuf[idx][EC_HEADERSIZE + idxstack->length[pos]]), EC_WKCSIZE);
               /* output WKC counts 2 times when using LRW, emulate the same for LWR */
               wkc = etohs(le_wkc) * 2;
               memcpy(&le_DCtime, &(rxbuf[idx][idxstack->dcoffset[pos]]), sizeof(le_DCtime));
               *(context->DCtime) = etohll(le_DCtime);
            }
            else
            {
               /* output WKC counts 2 times when using LRW, emulate the same for LWR */
               wkc += wkc2 * 2;
            }
            valid_wkc = 1;
         }

The wkc for LWR is compensated to behave similar as the LRW case. This simplifies the user application code.

stevedanomodolor, 2020-07-23:
The reading value works fine now, not tried the sending pdo but I assumed it should work, thank you very much to be honest,
```

502b7d45000
 position value: 382135
502b7d45000
 position value: 382135
502b7d45000
 position value: 386230
502b6e4501116
 position value: 410720
50260446000
 position value: 410723
50263446000
 position value: 410723
50263446000
```

stevedanomodolor, 2020-07-23:
@ArthurKetels  After changing and the applying the fix similar to the other issue, the type seems correct and I am able to read data but I am having issues writing commands, it does not modify the bits atall. I noticed that the size of the input and output are inverted. That should not be like that should it?  This is the fix I did because the (https://github.com/OpenEtherCATsociety/SOEM/issues/224#issuecomment-525872643) because using the one you mentioned alone without modifying the ethercatcoe code, the ethercat could not enter operational mode. 
```
  for (int cnt = 1; cnt <= ec_slavecount; cnt++)
            {
                ec_slave[cnt].PO2SOconfig = &fix_SM;
                // ec_slave[cnt].SMtype[2] = 3;
                // ec_slave[cnt].SMtype[3] = 4;
            }

```
int fix_SM(uint16 slave)
{
    ec_slave[slave].SMtype[2] = 3;
    ec_slave[slave].SMtype[3] = 4;
    return 1;
}
```

```
```
Slave:1
 Name:Servo Drives
 Output size: 152bits
 Input size: 104bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  19 F:00010064 Type:3
 SM3 A:1c00 L:  13 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  19 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000013 Ll:  13 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:1
PDO mapping according to CoE :
WKibviabvabvoaovboabvoabvoabvo: 1
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x0013.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x0015.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  [0x0016.0] 0x6071:0x00 0x10 INTEGER16    Target torque
  [0x0018.0] 0x60B2:0x00 0x10 INTEGER16    Torque offset
  [0x001A.0] 0x21EF:0x00 0x20 REAL32       Interpolation time mantissa
  [0x001E.0] 0x21F0:0x00 0x20 REAL32       Interpolation time exponent
  [0x0022.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
WKibviabvabvoaovboabvoabvoabvo: 1
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
  [0x000B.0] 0x6077:0x00 0x10 INTEGER16    Torque actual value
End slaveinfo, close socket
End program
```

ArthurKetels, 2020-07-23:
Again you wireshark logs come to the rescue, the problem is that the CoE object 1c00 defines the SM types too (redundant with SII eeprom). You have worked around the eeprom problem, but the CoE version is still wrong. The datatypes for the SM0..3 are 2,1,4,3. They should be 1,2,3,4. Same bug in the slave firmware, but other spot.  The partial correction causes to the wrong mapping calculation in SOEM.

You could also try to fix this in SOEM, but the best solution would still be a slave firmware update that corrects all bugs.

stevedanomodolor, 2020-07-24:
I was able to get it to work, as a temporal fix but waiting on the company's response. Thank you very much for your help so far amd fast response.

closed 2020-07-24 by stevedanomodolor

balazskiss90, 2020-08-10:
Hello @stevedanomodolor,

Previously we had a brief discussion, where I couldn't help you unfortunately, since I'm also new to EtherCAT. 
Earlier, I was able to reach OP state (in EtherCAT state machine) with the modifications, that I mentioned, otherwise I used only ec_SDOread in OP state, which worked quite well. 
Otherwise, lately I would like to use PDO(IOmap) access, and I have the same issue that you described here:

1. If I modified the ethercatcoe.c file, I couldn't reach operational state anymore with the simpletest, but the slaveinfo looks okay;
2. I added the SMtype fix function, but it seems that it has no effect, and if I attach that function to PO2SOconfig hook, after PDO mapping it seems that the PDO mapping function won't be called;

Could you help me, please, what modifications did you do and where did you apply them in order to reach OP state, and be able to read the PDO data? 

Thank you for your help in advance!

stevedanomodolor, 2020-08-10:
The p02so config does not indeed produce any effect. 
Maybe, you did not modify the ethercatcoe file correclty, 

This is basically what i did

For the main file 
```
 if (ec_config_init(FALSE) > 0) {
.................

                for (int i = 1; i <= ec_slavecount; i++) {
                    ec_slave[i].blockLRW = 1; /* blocking LRW command */
                }

..............
                    int usedmem = ec_config_map(&IOmap);
                    printf("SOEM used IOMap size: %d\n", usedmem);
                    /* configure DC options for every DC capable slave found in the list */
                    ec_configdc();

```

for the ethercatcoe file 

```

int ecx_readPDOmap(ecx_contextt *context, uint16 Slave, int *Osize, int *Isize)
{
   int wkc, rdl;
   int retVal = 0;
   uint8 nSM, iSM, tSM;
   int Tsize;
   uint8 SMt_bug_add;

   *Isize = 0;
   *Osize = 0;
   SMt_bug_add = 0;
   rdl = sizeof(nSM);
   nSM = 0;
   /* read SyncManager Communication Type object count */
   wkc = ecx_SDOread(context, Slave, ECT_SDO_SMCOMMTYPE, 0x00, FALSE, &rdl, &nSM, EC_TIMEOUTRXM);
   /* positive result from slave ? */
   if ((wkc > 0) && (nSM > 2))
   {
      /* limit to maximum number of SM defined, if true the slave can't be configured */
      if (nSM > EC_MAXSM)
         nSM = EC_MAXSM;
      /* iterate for every SM type defined */
      for (iSM = 2; iSM < nSM; iSM++)
      {
         rdl = sizeof(tSM);
         tSM = 0;
         /* read SyncManager Communication Type */
         wkc = ecx_SDOread(context, Slave, ECT_SDO_SMCOMMTYPE, iSM + 1, FALSE, &rdl, &tSM, EC_TIMEOUTRXM);
         if (wkc > 0)
         {
            // start slave bug prevention code, remove if possible
            if ((iSM == 2) && (tSM == 2)) // SM2 has type 2 == mailbox out, this is a bug in the slave!
            {
               SMt_bug_add = 1; // try to correct, this works if the types are 0 1 2 3 and should be 1 2 3 4
            }
            else if ((iSM == 2) && (tSM == 4))
            {
               tSM = 3;
            }
            else if ((iSM == 3) && (tSM == 3))
            {
               tSM = 4;
            }
            if (tSM)
[... 122 lines cut ...]
                  // printf("SMSMNSBHBVSHBVISBISBOISBOSBOBSOBSOBSOBOSBOS: output: %d\n", iSM);

                  /* we are doing outputs */
                  *Osize += Tsize;
               }
               else
               {
                  /* we are doing inputs */
                  *Isize += Tsize;
               }
            }
         }
      }
   }

   /* found some I/O bits ? */
   if ((*Isize > 0) || (*Osize > 0))
   {
      retVal = 1;
   }
   return retVal;
}
```

Ps, if you try to store the parameter with the store object, let me know because, I am still not able to do so.

balazskiss90, 2020-08-10:
Thank you for your quick response, I do appreciate it. 
I did the same modifications and in the mean time I could make it work, at least the data readings for now. 
I'm just trying now to bring the CANopen state machine into operational state and move the motor. If I can work it out, I'll let you know.

stevedanomodolor, 2020-08-10:
Send a message to ingenia to fix this issue, i sent it to thwm and i have
not gotten any response from them.

balazskiss90, 2020-08-10:
I'll send them a message also. Otherwise, they usually reply in 24 hours. If the problem is more complicated, (I assume this will be) it usually tooks them a couple days. If you don't get any response a kind reminder couldn't hurt. :)

stevedanomodolor, 2020-08-10:
Another thing to know, is that you cannot store pdo mapping, if you try to do so without actually modifying any other pdo, you get an error, just figured out after many tries

balazskiss90, 2020-08-11:
Thank you for your advice, otherwise I never store the pdo mapping, instead I attach an initialization function to the PO2SOconfig hook and call it at the beginning of the program:

`int XCR_setup(uint16 slave){
	int retval;
	uint8 u8val;
	uint16 u16val;

	retval = 0;
	
	uint32 u32Map;

	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u32Map = 0x60400010; // controlword
	retval += ec_SDOwrite(slave, 0x1600, 0x01, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x607A0020; // position set-point
	retval += ec_SDOwrite(slave, 0x1600, 0x02, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x60FF0020; // velocity set-point
	retval += ec_SDOwrite(slave, 0x1600, 0x03, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x60600008; // operation mode
	retval += ec_SDOwrite(slave, 0x1600, 0x04, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u8val = 4;
	retval += ec_SDOwrite(slave, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	
	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1A00, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u32Map = 0x60410010; // statusword
	retval += ec_SDOwrite(slave, 0x1A00, 0x01, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x60640020; // actual position
	retval += ec_SDOwrite(slave, 0x1A00, 0x02, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x606C0020; // actual velocity
	retval += ec_SDOwrite(slave, 0x1A00, 0x03, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x60610008; // operation mode display
	retval += ec_SDOwrite(slave, 0x1A00, 0x04, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u8val = 4;
	retval += ec_SDOwrite(slave, 0x1A00, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1601, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u32Map = 0x60400010; // controlword
	retval += ec_SDOwrite(slave, 0x1601, 0x01, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x607A0020; // position set-point
	retval += ec_SDOwrite(slave, 0x1601, 0x02, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u8val = 2;
	retval += ec_SDOwrite(slave, 0x1601, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	
	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1A01, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u32Map = 0x60410010; // statusword
	retval += ec_SDOwrite(slave, 0x1A01, 0x01, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x60640020; // actual position
	retval += ec_SDOwrite(slave, 0x1A01, 0x02, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u8val = 2;
	retval += ec_SDOwrite(slave, 0x1A01, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1602, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u32Map = 0x60400010; // controlword
	retval += ec_SDOwrite(slave, 0x1602, 0x01, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x607A0020; // position set-point
	retval += ec_SDOwrite(slave, 0x1602, 0x02, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u8val = 2;
	retval += ec_SDOwrite(slave, 0x1602, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	
	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1A02, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u32Map = 0x60410010; // statusword
	retval += ec_SDOwrite(slave, 0x1A02, 0x01, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u32Map = 0x606C0020; // actual velocity
	retval += ec_SDOwrite(slave, 0x1A02, 0x02, FALSE, sizeof(u32Map), &u32Map, EC_TIMEOUTRXM);
	u8val = 2;
	retval += ec_SDOwrite(slave, 0x1A02, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u16val = 0x1600;
	retval += ec_SDOwrite(slave, 0x1c12, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
	u8val = 1;
	retval += ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u8val = 0;
	retval += ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	u16val = 0x1a00;
	retval += ec_SDOwrite(slave, 0x1c13, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTRXM);
	u8val = 1;
	retval += ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	ec_slave[slave].SMtype[2] = 3;
        ec_slave[slave].SMtype[3] = 4;
	printf("SM type fix done for slave: %d\n", slave);
	
	while (EcatError) printf("%s", ec_elist2string());

	printf("Everest slave %d set, retval = %d\n", slave, retval);
	return 1;
}`

```
...
/* link slave specific setup to preop->safeop hook */
ec_slave[slc].PO2SOconfig = &XCR_setup;
...
```

It's just an example. you can map any pdo mappable register to 0x1600 or 0x1A00:
https://doc.ingeniamc.com/eve-xcr/manuals/everest-xcr-reference-manual/ethercat-canopen-registers

stevedanomodolor, 2020-08-11:
I know, i have been able to map data and make the motor move, and I do call the pdo mapping function every single time, although sdo configuration objects, i dont change the value everytime, but store it in the nvm memory because it is faster to execute. First, i check to make sure that the data is correct if not, store it.

balazskiss90, 2020-08-11:
Okay, I see. I think a misunderstood your comment and previously you mentioned that you had issue with pdo mapping. 
Otherwise, thank you for your tip.

balazskiss90, 2020-08-12:
@stevedanomodolor I got back an answer from Ingenia. They are aware of the mapping issue and it will be fixed in the upcoming 1.6.1. firmware.

stevedanomodolor, 2020-08-12:
I think the person I am talking too in on vacation, i am talking to
antonio, can you ask them also whether there can make it possible to store
pdo mapping, i have also sent them the same thing.

Thank you btw

stevedanomodolor, 2020-08-15:
Did you encounter this error when running multiple drivers at the same time?
When I run only one driver, everything works perfectly.
SOEM (Simple Open EtherCAT Master)
Simple test
Initializing Ethercat communcation
6 slaves found and configured.
All slaves are present as expected!
Slave in preop state, initicializing data configuration
ret: 1
ret: 1
ret: 1
ret: 1
ret: 1
ret: 1
All slaved pdo was mapped correctly
SOEM used IOMap size: 324
Slave:1 Name:Servo Drives Output size:136bits Input size:296bits State: 4 delay:0.1
Found Servo Drives at position 1
Slave:2 Name:Servo Drives Output size:136bits Input size:296bits State: 4 delay:1090.1
Found Servo Drives at position 2
Slave:3 Name:Servo Drives Output size:136bits Input size:296bits State: 4 delay:2172.1
Found Servo Drives at position 3
Slave:4 Name:Servo Drives Output size:136bits Input size:296bits State: 4 delay:3259.1
Found Servo Drives at position 4
Slave:5 Name:Servo Drives Output size:136bits Input size:296bits State: 4 delay:4341.1
Found Servo Drives at position 5
Slave:6 Name:Servo Drives Output size:136bits Input size:296bits State: 4 delay:5411.1
Found Servo Drives at position 6
Calculated workcounter 18
Request operational state for all slaves
Not all slaves reached operational state.
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog
Slave 2 State=0x14 StatusCode=0x001b : Sync manager watchdog
Slave 3 State=0x04 StatusCode=0x0000 : No error
Slave 4 State=0x14 StatusCode=0x001b : Sync manager watchdog
Slave 5 State=0x14 StatusCode=0x001b : Sync manager watchdog
Slave 6 State=0x14 StatusCode=0x001b : Sync manager watchdog
Problem intializing ethecat communication!!!
End program