OpenEtherCATsociety/SOEM #424: Example for EtherCAT SoE

TBartzsch, 2020-07-27:
Hello,

I want to set up a connection between a Raspberry Pi with SOEM and an IndraDrive from Bosch Rexroth via EtherCAT.
Since I only need to set the position to which the motor shall drive, I thought Servodrive-Profile over EtherCAT (SoE) is the best choice.

I am new to EtherCAT, so I am searching for an example how to use SoE with SOEM.
Does anyone know a website where such an example is shown, or does anyone has one?
Of course I already looked for an example, but could not find one.

Thank you,
TBartzsch

ArthurKetels, 2020-07-27:
Hmm, taking an unusual configuration both in slave (SoE) and master (RPi) and being new to EtherCAT might not be the best combination. 

My sincere suggestion would be to:
- Read up on EtherCAT and its protocols (available at www.ethercat.org).
- Read the Indradrive manuals.
- First try SOEM on a standard laptop with linux (easier to code and debug).
- Then try a simple slave configuration with some digital IO and get that working stable.

As far as I know the Indradrive slaves need master synchronization, this is another level of complexity (specially on a RPi).

I assume you want to use a RPi4? Otherwise real time performance from the NIC is mediocre.

If you then must continue with SoE see the ecatconfig.c on how SOEM does SoE to figure out the slave mapping. Configuring the slave with SoE is than not so difficult anymore.

TBartzsch, 2020-07-27:
Thanks for your answer.
To get real time performance, I use a Wiznet Ethernet Controller as described here:
http://www.simplerobot.net/2018/06/rpi-ethercat-master-picat.html
I will continue as you suggested and write again if I get the system to work (to close this issue) or if further problems occure.

ArthurKetels, 2020-07-27:
As you can read on the simplerobot.net webpage another user there is also implementing the Indradrive with Wiznet and RPi. Perhaps you can contact him to figure out how to get everything running?

closed 2020-09-29 by TBartzsch

ArthurKetels, 2020-09-29:
And, did you get the system to work?

TBartzsch, 2020-09-30:
Hi,

unfortunately not, I could not get the Bosch Rexroth controller to run.

The controller never changed to safe op, it always stopped in pre op.

Maybe I will try again later, when I have more time.

But I thought it was time to close the issue, so it does not rest opened all the time.

 

Greetings,

Thorben Bartzsch

 
 

Gesendet: Dienstag, 29. September 2020 um 16:27 Uhr