OpenEtherCATsociety/SOEM #425: The dreaded 0 sized output/input

cjastro, 2020-07-28:
I have been developing a driver on qnx for quite a while with the Elmo Gold Cello drive.  Because its QNX, I've been using soem 1.3.1, as that is what version has a qnx port.  Recently, because of covid, I've had to switch to a different drive for development. THis drive is a Copley BEL drive.  Using my software I'm , after I map IO, i'm getting 0 sized output and input..  So i ran the simple_test and slaveinfo programs...  same thing.  I switched to linux, and a newer vesion of SOEM from the soem repo, and I get the same results.  This stuff was working with the elmo.

Attached are she output captures for simple_test and slaveinfo -map console outputs, with their associated wireshark captures.
[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4990048/simple_test.txt)
[simple_test-wshark.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4990049/simple_test-wshark.txt)
[slaveinfo..txt](https://github.com/OpenEtherCATsociety/SOEM/files/4990050/slaveinfo.txt)
[slaveinfo-wshark.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4990051/slaveinfo-wshark.txt)

Some things worth noting:  
1) When the Elmo drive powers up, the servo is OFF...  no power.  motor is free to turn, and the motor stiffens up when I elvate to the correct state.   When this copley powers up, the servo is already on and holding still.(another member of the team tuned the inner velocity loop..  is it possible he left the drive in the wrong mode, making ethercat only partially functional?).. Of course without output and input, I cant do much to change states.

2)  When you send your rx and tx pdo mappings, the elmo allows me to send everything in 16 bit values, including the number of values being sent.  In the copley I must send the number of values as an 8 bit.

Is there a document somewhere that I might use to help port the most recent version of soem(1.4.x) to QNX?  I see that more recent versions have thread support, and some other features(im not using any of these features, but its newer code)...  QNX has no cmake, so, being the hacky bastard that I am,  I tried generating makefiles in the soem 1.4 tree,  then tried moving over some of the qnx directories from 1.3.1...  the build just sort of vomited all over itself because of missing headers...  I found similar headers in other 1.4 diretories and they had the pthread stuff in them...  it looks like there is a bit of work to do, here...  I might try putting it in the cross compiler with cmake and see if it helps, any.

THanx for any help,
CJ

ArthurKetels, 2020-07-28:
Could you please repost the wireshark data in their native (.pcap) format? These text files are impossible to parse. And, please zip them...

cjastro, 2020-07-28:
okay... these are the pcapng files, zipped

[simple_test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4990915/simple_test.zip)
[slaveinfo.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4990916/slaveinfo.zip)

ArthurKetels, 2020-07-28:
Thanks for the files. Now it is clear that there is no PDO mapping programmed in the Copley.
Both object 0x1c12:00 and 0x1c13:00 have zero as number of PDO entries. 
So first figure out which objects you want to have mapped (and are mapable according to the manual), set up a PO2SO hook function for the slave with the appropriate mapping and then try again.

cjastro, 2020-07-29:
ahhh... I appologize for wasting your time there.   My code runs on qnx and I can't wireshark from there, easily so rather than move that code over to linux I just ran simple_test...  I assumed that the simple_test stuffed in some default pdo value,  I didn't bother looking.  

So...  I grabbed some code from my qnx source and crammed it into simple_test.  Here is the code:

 //CYC_SYNC_VELOCITY_MODE                
                unsigned char buf8;
                unsigned short buf16;
                int eight = 8;
                int sixteen = 16;
                //unsigned int buf32;   
                buf8=9;
                ec_SDOwrite(1,0x6060,0, FALSE, 1, &buf8, EC_TIMEOUTRXM);

                printf("############ made it 1 ####################\n");
                //RX-PDO                
                buf8=0;
                ec_SDOwrite(1,0x1c12,0, FALSE, 1, &buf8, EC_TIMEOUTRXM);
                buf16=0x1601;
                ec_SDOwrite(1,0x1c12,1, FALSE, 2, &buf16, EC_TIMEOUTRXM);
                buf8=1;
                ec_SDOwrite(1,0x1c12,0, FALSE, 1, &buf8, EC_TIMEOUTRXM);

                printf("############ made it 2 ####################\n");
                //TX-PDO
                buf8=0;
                ec_SDOwrite(1,0x1c13,0, FALSE, 1, &buf8, EC_TIMEOUTRXM);
                buf16=0x1a03;
                ec_SDOwrite(1,0x1c13,1, FALSE, 2, &buf16, EC_TIMEOUTRXM);
                buf8=1;
                ec_SDOwrite(1,0x1c13,0, FALSE, 1, &buf8, EC_TIMEOUTRXM);

                ec_SDOread(1, 0x6060, 0, FALSE, &eight, &buf8, EC_TIMEOUTRXM);
                printf("mode=%i\n", buf8);

                ec_SDOread(1, 0x1c12, 0, FALSE, &eight, &buf8, EC_TIMEOUTRXM);
                ec_SDOread(1, 0x1c12, 1, FALSE, &sixteen, &buf16, EC_TIMEOUTRXM);
                printf("rxpdo:0=%i rxpdo:1=%i\n", buf8, buf16);

                ec_SDOread(1, 0x1c13, 0, FALSE, &eight, &buf8, EC_TIMEOUTRXM);
                ec_SDOread(1, 0x1c13, 1, FALSE, &sixteen, &buf16, EC_TIMEOUTRXM);
                printf("txpdo:0=%i txpdo:1=%i\n", buf8, buf16);

         int usedmem = ec_config_map(&IOmap);
         printf("usedmem=%i\n", usedmem);
         ec_configdc();

i reset the drive, then ran the modified simple_test, followed by slaveinfo -map.  attached are the outputs:

[simple_test.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4991884/simple_test.txt)
[simple_test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4991885/simple_test.zip)
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/4991886/slaveinfo.txt)
[slaveinfo.zip](https://github.com/OpenEtherCATsociety/SOEM/files/4991887/slaveinfo.zip)

Again, I apologize for not catching that.  I should have checked.

CJ

ArthurKetels, 2020-07-29:
Well, you code worked flawless, but I doubt it is what you wanted. You successfully assigned object 0x1601 to SM2 and object 0x1a03 to SM3. When doing the actual mapping SOEM finds out that there are no entries in both assign object. Result is that there is still an empty set. 

Again a case of : RTFM.

As the Copley manual states : 

> Fixed PDOs 
First, the bad news: Fixed means that the user can’t change them.Next, the good news: Fixed means they are saved the the CPU flash memory and are optimized for speed. They take less CPU time to execute, leaving more time for other tasks. Receive and Transmit PDOs, relatively speaking...From this it is clear that transmit & receive must always be discussed in the context of the producer and/or the consumer of the data. Master Slave    Types of Data Transmit Receive Control word, target position/velocity/torque, torque/velocity offset Receive Transmit Status word, actual position/velocity/torque, position (following) error 
Fixed RxPDOs 
For CSP (Cyclic Synchronous Position) 0x6060 Mode of Operation = 8:
Receive PDO 4: 0x1700 
0x6040 Control Word 
0x607A Profile Target position 
0x60B1 Velocity Offset 
0x60B2 Torque Offset 
For CSV (Cyclic Synchronous Velocity) 0x6060 Mode of Operation = 9:
Receive PDO 5: 0x1701 
0x6040 Control Word 
0x60FF Target Velocity 
0x60B2 Torque Offset 
For CST (Cyclic Synchronous Torque) 0x6060 Mode of Operation = 10:
Receive PDO 6: 0x1702 
0x6040 Control Word 
0x60FF Target Velocity 
0x60B2 Torque Offset 
Fixed TxPDO 
Transmit PDO 5: 0x1B00 
0x6041 Status Word 
0x6064 Actual Motor Position 
0x60F4 Position Loop Error (following-error) 
0x606F Actual Motor Velocity 
0x6077 Torque Actual Value

> Not-Fixed, or User Programmable PDOs 
These are only used when the user defines their contents. They run slower and take more CPU time than the fixed PDO. This should be kept in mind when defining their contents, keeping the amount of data moved to the minimum required for tasks. Contents of these PDOs are not linked to any mode of operation as are the fixed PDOs.  
Un-Fixed RxPDOs 
Receive PDO 0: 0x1600 
Receive PDO 1: 0x1601 
Receive PDO 2: 0x1602 
Receive PDO 3: 0x1603 
Un-Fixed TxPDOs 
Transmit PDO 1: 0x1A00 
Transmit PDO 2: 0x1A01 
Transmit PDO 3: 0x1A02 
Transmit PDO 4: 0x1A03

So I would suggest to try 0x1701 and 0x1b00 first.

You also could have searched this forum for "copley" and would have found some solutions too.

cjastro, 2020-07-29:
ahhhhh...  now I feel like a jerk.  The elmo works with that old code...  I bet they have some default values in those pdo's

thanx

cjastro, 2020-07-30:
yup...  im a moron.  all working.  
thanx to ArthurKetels

cj

closed 2020-07-30 by cjastro