OpenEtherCATsociety/SOEM #427: Store new pdo mapping

stevedanomodolor, 2020-08-07:
Is it possible to store the pdo mapping to the nvm memory so that I don't have to mapping everytime I compile my program, this way preventing the pdo maping from resetting after every power cycle? 
The pdo mapping maps correctly but after I run it again, it gives this error, but the mapping is still done correctly, but when I remove the mapping part of the program and turn off and on the driver again, it looses the the pdo mapping I did. 
(This is the result of the slaveinfo example)

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
 RXPDO1 configured incorrectly
 TXPDO1 configured incorrectly
 RxPDO assign configured incorreclty
 TxPDO assign configured incorreclty
mapping succesfullTime:   39833.833 SDO slave:1 index:1600.01 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.833 SDO slave:1 index:1600.02 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.834 SDO slave:1 index:1600.03 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.834 SDO slave:1 index:1600.04 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.835 SDO slave:1 index:1600.05 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.835 SDO slave:1 index:1600.06 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.836 SDO slave:1 index:1600.07 error:06010003 Subindex can not be written, SI0 must be 0 for write access
Time:   39833.846 SDO slave:1 index:1c12.01 error:06010003 Subindex can not be written, SI0 must be 0 for write access
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  25 : Invalid output mapping

Slave:1
 Name:Servo Drives
 Output size: 88bits
 Input size: 424bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  53 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000b Ll:  53 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:1
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x0002.0] 0x607A:0x00 0x20 INTEGER32    Position set-point
  [0x0006.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
  [0x000A.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  [0x000B.0] 0x0000:0x00 0x00
  [0x000B.0] 0x0000:0x00 0x00
  [0x000B.0] 0x0000:0x00 0x00
ISMssssssssssssssssssssssssssssssssssssssssssssssssss3
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x000D.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
  [0x000E.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0012.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x0016.0] 0x203B:0x00 0x20 REAL32       Current quadrature value
  [0x001A.0] 0x2060:0x00 0x20 REAL32       Bus voltage value
  [0x001E.0] 0x2067:0x00 0x20 REAL32       Power stage temperature
  [0x0022.0] 0x206F:0x00 0x20 REAL32       Voltage quadrature command
  [0x0026.0] 0x2072:0x00 0x20 REAL32       Current quadrature demand
[... 492 lines cut ...]
    ret += ec_SDOwrite(slave, TXPDO_MAP_ASSIGN, 0x01, FALSE, sizeof(dType_U16), &dType_U16,
                       EC_TIMEOUTSAFE);
    // printf("ret : %d\n", ret);

    // Set number of assigned PDO
    dType_U8 = 0x01;
    ret += ec_SDOwrite(slave, TXPDO_MAP_ASSIGN, 0x00, FALSE, sizeof(dType_U8), &dType_U8,
                       EC_TIMEOUTSAFE);
    if (ret != 42) {
        printf(" TxPDO assign configured incorreclty\n");
        return 0;
    }

    /**********************************TPDO************************************************************/

    // TODO: CASK supervisor whether it is recommend to sotre this parameters to the non-volatile memory
    // // saving configuration

    // dType_32 = 0x73;
    // ret += ec_SDOwrite(slave, 0x1010, 0x01, FALSE, sizeof(dType_32), &dType_32,
    // EC_TIMEOUTSAFE);

    // printf("Slave %d was not configured correcty\n", slave);
    return 1;
}

ArthurKetels, 2020-08-07:
In below part of your code:
>    /*********************TPDO mapping***************************************/
>
>    /* Unmap all registers from PDO by setting Subindex 0x00 to zero */
    dType_U32 = 0x00000000;
    ret += ec_SDOwrite(slave, TXPDO_MAP_OBJECT, 0x00, FALSE, sizeof(dType_U32), &dType_U32,
                       EC_TIMEOUTSAFE);`

Objects of type array or record have the first subindex (0) defined as U8. A wireshark trace will show that above SDO download will fail. Then following SDO downloads will fail too because the first subindex was not successfully reset to zero.

When something is not working as expected check the return value of each SOEM function for errors, and do not lazy skip over them as in your above code. BTW, RXPDO has the same issue.

Searching in this forum for configuration code examples would have showed you the same.

ArthurKetels, 2020-08-07:
Your other question about storing the configuration in the slave NVM depends on the slave vendor in question. You have to look it up in the slave manual.

Some slaves have NVM to store configuration others do not. And the philosophy behind the policy is debatable for both ends. Do you push configuration down in the tree or keep it high up? What if you have a mixed slave combination where some have NVM and others do not? Many PhD thesis is written on this subject.

For me in my professional work use the following rules:
a) Use NVM for storing configuration in a slave when available, this is for speed of system configuration.
b) Check each slave for correct configuration against your application requirements.
c) When above check fails, reprogram the slave and try to store in NVM again (could have been a slave swap).
d) When it fails a second time, bail out with an error report.

stevedanomodolor, 2020-08-07:
Thank you very much, i will look it, i did not notice it, sorry

stevedanomodolor, 2020-08-07:
I was able to fix it

stevedanomodolor, 2020-08-07:
I was able to store basic configuration, the code I sent you does not reflect that though, the issue I had was with the pdo mapping, and how do you check this "Check each slave for correct configuration against your application requirements." to make sure the configuration set before has not changed?, is this done by reading the values individually with the sdo or is there a more faster way?

Thank you

ArthurKetels, 2020-08-07:
Ehh, if what you posted is not what you intended then please post what you intend to do?

Your questions seem confused and possibly reflect your state of mind at this moment. There seem to be no issues with SOEM and yet you keep asking questions about your slave and your code here. If the problem is with your slave then ask the questions to your slave vendor (did you look to object 0x1010 for example?). If you do not understand the basics of EtherCAT enough, please read the available documentation on www.ethercat.org (it also handles the different ways of how a slave can store state, and how to check it, also it describes CompleteAccess SDO functions that Ingenia supports). If there is a problem with your coding ask yourself how to solve those issues.

You ask about "faster ways", but what are your requirements about configuration speed? "Make sure" can be done in many ways, not only programming.

I don't want to be rude, but keep focused on your goals. Gather knowledge, make a plan, execute, test, start over when needed.

stevedanomodolor, 2020-08-07:
Thank you very much, i will look into it.

stevedanomodolor, 2020-08-08:
Thank you for your time and sorry for the constant disturbance. 
About the storing, I was able to do so(sorry, i didnot pay attention to that). My issue is still with setting the mapping and assign object subindex 0. My objective now is to make sure I can pdo map the data correctly. 
After multiply try, I was still am not able to write to txpdo assign object, but after the second try, i was able to was able to write to the rpdo mapping parameter. 

(Also, this is out of curiosity, why is it that using the ec_sdoread does not return any value only after I have configured the pdo mapping with the ec_configmap function?)

[wireshark_NUM_try.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5045428/wireshark_NUM_try.zip)
udo ./slaveinfo_modified eth1 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Could not get write access(rxpdo map1), pdo already mapped?
Could not get write access(txpdo assign), pdo already mapped?
Check: 2
Try 1 : Wrong pdo configuration for slave: 1
Trying configuration again
Could not get write access(txpdo assign), pdo already mapped?
Check: 3
Try 2 : Wrong pdo configuration for slave: 1
Trying configuration again
Could not get write access(txpdo assign), pdo already mapped?
Check: 3
Try 3 : Wrong pdo configuration for slave: 1
Trying configuration again
Could not get write access(txpdo assign), pdo already mapped?
Check: 3
Try 4 : Wrong pdo configuration for slave: 1
Trying configuration again
Could not get write access(txpdo assign), pdo already mapped?
Check: 3
Try 5 : Wrong pdo configuration for slave: 1
Trying configuration again
Could not get write access(txpdo assign), pdo already mapped?
Check: 3
Try 6 : Wrong pdo configuration for slave: 1
Trying configuration again
End slaveinfo, close socket
End program

```

 [RXPDO_ASSIGN_SUBINDX0] = {.index = 0x1C12, .subindex = 0x00, .size = BIT8},
    [RXPDO_ASSIGN_SUBINDX1] = {.index = 0x1C12, .subindex = 0x01, .size = BIT16},
    [TXPDO_ASSIGN_SUBINDX0] = {.index = 0x1C13, .subindex = 0x00, .size = BIT8},
    [TXPDO_ASSIGN_SUBINDX1] = {.index = 0x1C13, .subindex = 0x01, .size = BIT16},
    // [TXPDO_ASSIGN_SUBINDX2] = {.index = 0x1C13, .subindex = 0x02, .size = NOT_SPECIFIC},
    [STORE]                 = {.index = 0x1010, .subindex = 0x01, .size = BIT32},
    [DIG_INPUT_VALUES]      = {.index = 0x2600, .subindex = 0x00, .size = BIT32}};

```
This is the rpdo mapping part 

```
dType_U8 = 0;
    ret += ec_SDOwrite(slave, obj[RPDO1_MAPPING].index, 0x00, FALSE, sizeof(dType_U8), &dType_U8,
                       EC_TIMEOUTRXM);

    // printf("%x\n", obj[RPDO1_MAPPING].index);
    // printf("return value rxpdo : %d\n", ret);

    if (ret == 1) {
        ret = 0;
        // Subindex 1
        // Adding control word

        // dType_U32 = BIT_SHIFT(obj[CONTROLWORD].index, obj[CONTROLWORD].subindex, obj[CONTROLWORD].size);
        // ret += ec_SDOwrite(slave, obj[RPDO1_MAPPING].index, 0x01, FALSE, sizeof(dType_U32), &dType_U32,
        //                    EC_TIMEOUTRXM);

[... 114 lines cut ...]
                            mapping_check[i - 1] = pdo_mapping_setup(i);
                            printf("Check: %d\n", mapping_check[i - 1]);
                            if (mapping_check[i - 1] != 4) {
                                printf("Try %d : Wrong pdo configuration for slave: %d\n", try, i);
                                printf("Trying configuration again\n");
                            } else {
                                printf("Try %d :pdo configuration correct for slave: %d\n", try, i);

                                config_correct[i - 1] = TRUE;
                            }
                        }

                        if (try == NUM_TRY) {
                            for (int i = 0; i < NUM_SLAVES; i++) {
                                if (config_correct[i] == FALSE) {
// goto: bad practise remove later
                                                goto wrong_configuration;
                                }
                            }
                            break;
                        }
                    }
                }

```

stevedanomodolor, 2020-08-08:
Is it normal that the return value for the rxpdo mapping parameters always gives zero at first try and later works? 

[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5046570/wireshark.zip)

```

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
ret rxpdo(map): 0
Could not get write access(rxpdo map1), pdo already mapped?
ret txpdo(map): 1
Data value: 60410010
Data value: 20760020
Data value: 203b0020
Data value: 20670020
Data value: 238b0020
Data value: 60640020
Data value: 606c0020
Data value: 26000020
Data value: 60610008
Data value: 60770010
ret rxpdo(assign) : 1
ret tpdo(assign): 3
Check: 3
Try 1 : Wrong pdo configuration for slave: 1
Trying configuration again
ret rxpdo(map): 1
Data value: 60400010
Data value: 60710010
Data value: 60ff0020
Data value: 21ef0020
Data value: 21f00020
Data value: 60600008
ret txpdo(map): 1
Data value: 60410010
Data value: 20760020
Data value: 203b0020
Data value: 20670020
Data value: 238b0020
Data value: 60640020
Data value: 606c0020
Data value: 26000020
Data value: 60610008
Data value: 60770010
ret rxpdo(assign) : 1
ret tpdo(assign): 3
Check: 4
Try 2 :pdo configuration correct for slave: 1
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Servo Drives
 Output size: 136bits
 Input size: 296bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  17 F:00010064 Type:3
 SM3 A:1c00 L:  37 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  17 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000011 Ll:  37 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
[... 111 lines cut ...]
        dType_U32 = BIT_SHIFT(obj[OP_MODE].index, obj[OP_MODE].subindex, obj[OP_MODE].size);
        ret += ec_SDOwrite(slave, obj[RPDO1_MAPPING].index, 0x06, FALSE, sizeof(dType_U32), &dType_U32,
                           EC_TIMEOUTRXM);

        printf("Data value: %x\n", dType_U32);

        // Enable mapping by setting SubIndex 0x00 to the number of mapped objects.
        // (RXPDO_MAP_OBJECT RPDO1 mapping parameter).
        dType_U8 = 6;
        ret += ec_SDOwrite(slave, obj[RPDO1_MAPPING].index, 0x00, FALSE, sizeof(dType_U8), &dType_U8,
                           EC_TIMEOUTRXM);
        // printf("Data value: %x\n", dType_U32);

        if (ret != 7) {
            printf(" RXPDO1 configured incorrectly\n");
            printf("return rxpdo map: %d\n", ret);
            // //return 0;
        } else {
            check += 1;
        }

    } else {
        printf("Could not get write access(rxpdo map1), pdo already mapped?\n");
    }
```

ArthurKetels, 2020-08-09:
The problem is that you start downloading SDO data into the slave right after you set PRE-OP (0x0002) in register (0x0120). But you do not wait until the slave has reached PRE-OP (read in register 0x0130). So the SDO request in the mailbox of the slave gets cancelled at the slave transition to PRE-OP and SOEM waits for ever for a response that is never going to come. During the timeout the slave has transitioned to PRE-OP and subsequent request do get handled correctly.

Again use good programming practices, always check for a state transition after you have requested it. What gets you more confused is the error message that you have made up yourself "pdo already mapped?". You make assumptions that you have not checked. You put a question mark after the assumption, but that does not make it right. When programming you need to state only the things you know are true. And time and again your code shows the same type of flaws.

For example: 
> dType_U8 = 6;

You express "data type unsigned 8 = 6". But what you wanted to express is "data container of type unsigned 8 = 6".  Your code would be better when writing "dataU8 = 6". But even then the expression is not very strong nor robust. Because when you read the code you might think "what data?". And repeating the type in the identifier is dangerous, what if you later change the definition of the identifier to a struct or union?  

What about the following example?

> #define SDOsize_index 0x00;
#define SDOstandard FALSE;
#define SDOcompleteaccess TRUE; 
uint8_t   RXPDO1size;
RXPDO1size = 6;
ret = ec_SDOwrite(slave, obj[RPDO1_MAPPING].index[SDOsize_index], SDOsize_index, SDOstandard, sizeof(RXPDO1size), &RXPDO1size, EC_TIMEOUTRXM);

stevedanomodolor, 2020-08-09:
Thank you for the feedback to be honest(everything make sense now), I
always programmed like that, never knew it was wrong. Your suggestions are
way more clear than mine.

Thank tou again for your help

closed 2020-08-09 by stevedanomodolor