OpenEtherCATsociety/SOEM #429: Error reading datatype uint32_t

stevedanomodolor, 2020-08-13:
This might be a stupid question but I cannot seem to find the reason why it does not work. 
I get the a packet error when i try to read or write to varibales of type uint32_t even having making sure I am in the correct state. 

I tried it in safeop and preop state but got more or less similar values. There is something I am missing but I cannot seem to figure out what. 

This is the wire shark when I tried to read the value in safeop state after pdo has being mapped. 
[wireshark_data.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5070532/wireshark_data.zip)

```

            if (ec_slave[0].state == EC_STATE_PRE_OP) {
                uint32_t read_value_angle = 0;
                int size_read_value_angle;
              

                for (cnt = 1; cnt <= ec_slavecount; cnt++) {
                    int read_ret = ec_SDOread(cnt, 0x21F4, 0x00, FALSE, &size_read_value_angle, &read_value_angle, EC_TIMEOUTRXM);
                    printf("Velocity window time, slave %d: %d\n Ret: %d\n", cnt, read_value_angle, read_ret);

                    read_ret = ec_SDOread(cnt, 0x2388, 0x00, FALSE, &size_read_value_angle, &read_value_angle, EC_TIMEOUTRXM);
                    printf("Digital encoder 1 - Resolution, slave %d: %d\n Ret: %d\n", cnt, read_value_angle, read_ret);

                    read_ret = ec_SDOread(cnt, 0x2389, 0x00, FALSE, &size_read_value_angle, &read_value_angle, EC_TIMEOUTRXM);
                    printf("Digital encoder 1 - Polarity, slave %d: %d\n Ret: %d\n", cnt, read_value_angle, read_ret);

                    read_ret = ec_SDOread(cnt, 0x2102, 0x00, FALSE, &size_read_value_angle, &read_value_angle, EC_TIMEOUTRXM);
                    printf("Peak current time, slave %d: %d\n Ret: %d\n", cnt, read_value_angle, read_ret);
                }
            }

```
This is in safe-op state

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
1 slaves found and configured.
Calculated workcounter 3
Velocity window time, slave 1: 0
 Ret: 0
Digital encoder 1 - Resolution, slave 1: 0
 Ret: 1
Digital encoder 1 - Polarity, slave 1: 0
 Ret: 0
Peak current time, slave 1: 4800
 Ret: 1
Time:   33469.638 PACKET slave:1 index:21f4.00 error:3
Time:   33469.640 PACKET slave:1 index:2389.00 error:3

Slave:1
 Name:Servo Drives
 Output size: 88bits
 Input size: 88bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  11 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000D.0] 0x607A:0x00 0x20 INTEGER32    Position set-point
  [0x0011.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
  [0x0015.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
End slaveinfo, close socket
End program
```
WHen i try reading it in preop-state, i get the same error 

```
OEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
Velocity window time, slave 1: 0
 Ret: 0
Digital encoder 1 - Resolution, slave 1: 0
 Ret: 0
Digital encoder 1 - Polarity, slave 1: 0
 Ret: 0
Peak current time, slave 1: 0
 Ret: 0
Time:   34333.826 PACKET slave:1 index:21f4.00 error:3
Time:   34333.826 PACKET slave:1 index:2388.00 error:3
Time:   34333.826 PACKET slave:1 index:2389.00 error:3
Time:   34333.827 PACKET slave:1 index:2102.00 error:3
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Servo Drives
 Output size: 88bits
 Input size: 88bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  11 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:1
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000D.0] 0x607A:0x00 0x20 INTEGER32    Position set-point
  [0x0011.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
  [0x0015.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
End slaveinfo, close socket
End program
```

THis only happens when I am reading a uint32 type value,

nakarlsson, 2020-08-13:
Try setting the size of the variable you’ll read.

nt size_read_value_angle = sizeof(read_value_angle);

ArthurKetels, 2020-08-13:
To elaborate a bit more on previous remark.

The prototype is:
> int ec_SDOread(uint16 slave, uint16 index, uint8 subindex, boolean CA, int *psize, void *p, int timeout);

ec_SDOread can read any number of bytes. Because *p is void the size of the pointer to the data is unknown. Therefore *psize is a pointer to an int that gives the number of bytes to read. The difference with ec_SDOwrite is that now the size of the data read from the slave can be different (smaller or equal) from the buffer you provided. For scalar types this will never happen but there are also array and string types that can have variable size.

You give an uninitialized value for *psize. It can be any size. Again you assume things without actually checking them. The header files of SOEM give very valuable information about how the API is supposed to work.

stevedanomodolor, 2020-08-14:
I tried that initially but i still got the error,  it worked with the others variable type though.

velocity window time, slave 1: 0
 Ret: 1
Digital encoder 1 - Resolution, slave 1: 4800
 Ret: 1
Digital encoder 1 - Polarity, slave 1: 0
 Ret: 1
Peak current time, slave 1: 0
 Ret: 0
Time:    2883.191 PACKET slave:1 index:2102.00 error:3
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Servo Drives
 Output size: 88bits
 Input size: 88bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  11 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000b Ll:  11 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:1
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6040:0x00 0x10 UNSIGNED16   Control Word
  [0x000D.0] 0x607A:0x00 0x20 INTEGER32    Position set-point
  [0x0011.0] 0x60FF:0x00 0x20 INTEGER32    Velocity set-point
  [0x0015.0] 0x6060:0x00 0x08 INTEGER8     Operation mode
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   Status Word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    Actual position
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    Actual velocity
  [0x000A.0] 0x6061:0x00 0x08 INTEGER8     Operation mode display
End slaveinfo, close socket
End program

ArthurKetels, 2020-08-14:
I am sorry, but you really have start thinking about your coding style. 

Above is a simple coding bug, "reusing variables with unknown content". size_read_value_angle is not a constant and you are treating it as one. After the SDOread of 0x2389 it has value 2. For you to figure out why. And this is of course the wrong value for SDOread 0x2102. 

There is a distinction between "efficient" coding and "lazy" coding.

stevedanomodolor, 2020-08-14:
It more sense, in my real code, i only initialized the value itself not the size of the value, i seriously thought that the size was a constant value per variable type that was why i never though of checking the value. Thank you very much, everything with soem now works perfectly, sorry for the inconveniences so far,  learnt a lot so far

```
    (data_indentifier##readval) = 0;                                                                                                                                                       \
    if (((sdo_configuration_check[sdo_position]) == DATA_NOT_MATCH)) {    
```

closed 2020-08-14 by stevedanomodolor