OpenEtherCATsociety/SOEM #430: timeout in ec_receive_processdata(int timeout) and dropped packet

fmros, 2020-08-14:
Hy, 

i have a working ethercat-communication with some drives working on an application PC (preempt-rt @ 5.4.54-rt32). I am measuring with wireshark with a second PC (connected via a switch to the ethercat line) the traffic. 

The issue:
On some really rare cases, I am able to measure on the 2. PC that the ethercat_master(running on 1. PC) send a package and it returned from the drives (4. packet in the image ):
[image]

But the application PC does not recieve the answer and reacts as follows on packet 464090:
[image]

The two PCs have different hardware and OS, wherefore I assume that the application PC is not able to get the response from the drives. Or that the NIC (Realtek RTL8111/8168/8411)  is dropping that packet?

However having the timeout of 500us and calling ec_receive_processdata(500) seems to not be respected as the application PC reacts to a non response always after >~5ms.

Am I understanding the timeout wrong?

Thanks in advance.

ArthurKetels, 2020-08-14:
Not enough raw data to give a clear answer. Could you post both wireshark traces (minimum 100 line before and after the "event")?

fmros, 2020-08-20:
I did a new testrun and recorded the requested traces:
[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5102917/wireshark.zip)
you can find the issue on each trace on packet 200-201.
On this test the timeout was even ~10ms.

PC1 runs the ethercat master
PC2 is just used for listening.

Thanks!

ArthurKetels, 2020-08-20:
Thanks for the wireshark trace, very helpful.

About timeouts.  Soem uses two functions that need support from the OS. The socket receive function should be non-blocking, and the timer/clock function that should have enough granularity. If one or both are too course the programmed timeout can not be honoured. In most implementations we have rather precise performance. One notable exception is windows.

On your missing packet. In the wireshark traces it is easy to see that the master is accurate in transmit timing of the PDO packets. However sometimes there is no return packet recorded. On the log PC this packet is visible.

Possible causes:
1) Switch does not return packet to PC1 (packet dropped or link is lost).
2) Packet gets corrupted between switch and PC1 (fail of CRC will cause NIC to drop packet).
3) NIC driver in PC1 drops packet.

Just to verify, there are multiple reports that the Intel I210 NIC has problems, do you have this type in PC1?

fmros, 2020-08-20:
Thanks for the feedback.

"Soem uses two functions that need support from the OS. The socket receive function should be non-blocking, and the timer/clock function that should have enough granularity.  If one or both are too course the programmed timeout can not be honoured." 
Is there anything that can be done to prevent this? 

PC1 has a Realtek RTL8111/8168/8411 NIC.

ArthurKetels, 2020-08-20:
Dirty way is to provide ec_receive_processdata() with zero timeout. It will return immediately (and will have most likely not captured the incoming packet). But you can check the timing, it should not take longer than 10us to return. You can loop the function until a capture has succeeded.

Well, now we are sure the packets are returned from the slaves. So we have a downstream issue. Some drivers can be tweaked to pass packets with a crc error.

Another item to look at is the proper NIC driver configuration with ethtool. You do not want packet coalescing.

fmros, 2020-08-26:
Unfortunately also having:
```
$sudo  ethtool -k enp5s0
...
rx-fcs: on
rx-all: on
...
```
the issue returned, no return for over 4ms. Se here the wireshark trace:
[timeout.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5128384/timeout.zip)

ArthurKetels, 2020-08-26:
I am starting to think it is either something related to your hardware or a kernel bug. I did experience similar issues due to System Management Interrupts (SMI). The PC can become unresponsive for multiple ms. 
https://rt.wiki.kernel.org/index.php/HOWTO:_Build_an_RT-application

Another thing to test is to downgrade to an older linux kernel.

The only way to block ec_receive_processdata()  is to hit the mutex. And you can only do that by having other tasks requesting packet transfer on the same raw socket. I do not see any evidence for this in the wireshark captures. And even when, then the time penalty is no more than a few 10s of us.

An other suggestion is to run cyclictest to find other causes of system latency.

fmros, 2020-08-28:
To exclude the hardware i did some tests on a totally different computer, having a intel i219 NIC. Additionally i used a preempt-rt 4.14.146-rt67 to see how it behaves on another kernel version.

Was also able to record here similar behaviour: 
[test_2.pcapng.tar.gz](https://github.com/OpenEtherCATsociety/SOEM/files/5140254/test_2.pcapng.tar.gz)

The latencies of the OS tested with cyclictest(4hours) are below 50us, the highest SMIs (test https://www.osadl.org/Create-a-frequency-plot-from-data-of-har.bash-script-for-hwlat-plots.0.html) as well.