OpenEtherCATsociety/SOEM #433: Assessment of how well the RPI 4 is suited as a master

AlSa-91, 2020-08-14:
Dear SOEM Team
i work with SOEM on a RPI4 with CIA 402 engines and have xenomai installed for it. 
@ArthurKetels answered in another question, why not just use a preemt-rt on Linux (surely another master device)...
But I ask myself the same question at RPI. As far as I know the problem is that the NIC driver is not realtime capable and therefore adjustments are made like at simplerobot.net (RTnet Driver).
1) Do you think these are necessary for hard realtime or slave synchronization?
Red_test shows an example how to synchronize times or slaves. But if the communication is not determnistic, does it work at all?

2) Is the network driver in SOEM deterministic by default or do any adjustments have to be made by the user?

3) Do you think that the standard NIC of the RPI4 will also serve its purpose without any adjustments and that the way via Xenomai and the external Ethernet module is overrated?

Thanks a lot

ArthurKetels, 2020-08-14:
The RPI4 should be ok from a NIC perspective. It is not connected via USB as on RPI3.
For how to setup preemptrt on RPI4:
https://lemariva.com/blog/2019/09/raspberry-pi-4b-preempt-rt-kernel-419y-performance-test
Latency is not too bad at approx 70us. With some tuning this could be even improved. 
I would suggest to use the RPI4 headless. And to remove any service that is not needed.

To answer your questions:
1. Your master needs to be as deterministic as your slave configuration allows. You can calculate the maximum time deviation that a packet from the master is allowed to be send in order to stay within the bounds of the slave. Again this depends on the slave configuration.
2. SOEM has no network driver, it attaches to a standard RAW socket. SOEM gets what the OS (and NIC driver) supports. If you really need the best of best performance you have to build your own driver (not THAT hard).
3. Depends on your requirements (that you have not stated BTW).

AlSa-91, 2020-08-17:
Many thanks for the quick response @ArthurKetels!

closed 2020-08-17 by AlSa-91