OpenEtherCATsociety/SOEM #437: SOEM worked on ESTUN servo unnormally

ilovewall, 2020-08-24:
hello ,everyone. I used SOEM on ESTUN servo，it run unnormally,I set motor velocity was 60r/min,but the servo dispalyed 3 r/min actually,and I changed the pulses in a period,the servo still run 3r/min. I used wireshark to captured the message,the status word and control word were correct,i did not know why?
I changed panasonic servo instead of ESTUN,the servo could work fine.I guess maybe something wrong on configuraton,i had remapped PDO according the ESTUN xml,please give me some suggestion.thank you!

ilovewall, 2020-08-24:
[wireshark and xml.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5116730/wireshark.and.xml.zip)

summerlotus513, 2020-08-31:
Hello, I have recently encountered some problems in using SOEM to try to control the inovance servo. Can you provide me with the code for controllingESTUN servo.  Thank you very much.

ilovewall, 2020-08-31:
Hi,I had not inovance servo,so  i am not sure this servo can  work in my board,I am very sorry that the code cannot send to you as my code was encrypted.

closed 2020-09-08 by ilovewall