OpenEtherCATsociety/SOEM #439: how to keep the DOs as "1" for Beckhoff EP2338-0002 IO box

LamMork, 2020-08-26:
Hi, I am now using the EtherCAT IO box EP2338-0002 from Beckhoff, and I want to use the SOEM to control it's IO state.

The problem is, when I send the process date to the slave, the corresponding io do switch to "1", but it will go back to "0" immediately. So what I can see is the LED blinks. My question is, how can I keep this io as "1"? Do I need to send the process data unendingly?

Moreover, I tried to use the `ec_SDOwrite()` to write the value to the object, it doesn't work neither.

I am new to the EtherCAT and SOEM, can somebody help me with this?

Here are more information about my testing environment and test case:
Hardware:
    EtherCAT master: PC(debian) + SOEM
    EtherCAT slave: Beckhoff EP2338-0002 IO Module

Slave Information:
```
lammork@debian:~/Desktop/soem_demo/SOEM/build/test/linux$ sudo ./slaveinfo/slaveinfo eno2 -map
[sudo] password for lammork: 
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eno2 succeeded.
ec_config_init 0
ec_config_map_group IOmap:0x5612f60f92c0 group:0
 >Slave 1, configadr 1001, state 12
  SII Isize:8
    SM1 length 8
  SII Osize:8
    SM0 length 8
     ISIZE:8 8 OSIZE:8
  SM programming
    SM0 Type:3 StartAddr:0f00 Flags:00090044
    SM1 Type:4 StartAddr:1000 Flags:00010000
  OUTPUT MAPPING
    FMMU 0
      SM0
    slave 1 Outputs 0x5612f60f92c0 startbit 0
 =Slave 1, INPUT MAPPING
    FMMU 1
      SM1
    Inputs 0x5612f60f92c1 startbit 0
IOmapSize 2
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:EP2338-0002
 Output size: 8bits
 Input size: 8bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000002 ID: 09224052 Rev: 00170002
 SM0 A:0f00 L:   1 F:00090044 Type:3
 SM1 A:1000 L:   1 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:7 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   1 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1600 Channel 9
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1601 Channel 10
     addr b   index: sub bitl data_type    name
  [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1602 Channel 11
     addr b   index: sub bitl data_type    name
  [0x0000.2] 0x7020:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1603 Channel 12
     addr b   index: sub bitl data_type    name
  [0x0000.3] 0x7030:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1604 Channel 13
     addr b   index: sub bitl data_type    name
  [0x0000.4] 0x7040:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1605 Channel 14
     addr b   index: sub bitl data_type    name
  [0x0000.5] 0x7050:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1606 Channel 15
     addr b   index: sub bitl data_type    name
  [0x0000.6] 0x7060:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1607 Channel 16
     addr b   index: sub bitl data_type    name
  [0x0000.7] 0x7070:0x01 0x01 BOOLEAN      Output
  SM1 TXPDO 0x1A00 Channel 1
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x6000:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A01 Channel 2
     addr b   index: sub bitl data_type    name
  [0x0001.1] 0x6010:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A02 Channel 3
     addr b   index: sub bitl data_type    name
  [0x0001.2] 0x6020:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A03 Channel 4
     addr b   index: sub bitl data_type    name
  [0x0001.3] 0x6030:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A04 Channel 5
     addr b   index: sub bitl data_type    name
  [0x0001.4] 0x6040:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A05 Channel 6
     addr b   index: sub bitl data_type    name
  [0x0001.5] 0x6050:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A06 Channel 7
     addr b   index: sub bitl data_type    name
  [0x0001.6] 0x6060:0x01 0x01 BOOLEAN      Input
  SM1 TXPDO 0x1A07 Channel 8
     addr b   index: sub bitl data_type    name
  [0x0001.7] 0x6070:0x01 0x01 BOOLEAN      Input
End slaveinfo, close socket
End program
```

The "blinking" test:
```
void simpletest(char *ifname)
{
        int i,k = 8,j=0;
        if(ec_init(ifname))
        {   
                printf("start ethernet at %s\n",ifname);
                if ( ec_config_init(FALSE) > 0 ) 
                {   
                        printf("found %d slave on the bus\n",ec_slavecount);
                        ec_config_map(&IOmap);
                        for(i=0;i<ec_slavecount;i++)
                        {   
                                printf("slave%d to op\n", i); 
                                slavetop(i);
                        }   
                        if(ec_slave[j].state == EC_STATE_OPERATIONAL)
                        {   
                                //===========================================
                                // uint16_t tmpVal = 0x1;
                                while(k--)
                                {
                                        ec_slave[0].outputs[0x0000] = 0b0000000000000100;
                                        // ec_slave[0].outputs[0x0000] = tmpVal;
                                        // tmpVal = tmpVal << 1;

                                        ec_send_processdata();
                                        ec_receive_processdata(EC_TIMEOUTRET);
                                        sleep(1);
                                }
                                //===========================================
                                /*
                                set_output_int16(1, 0x0000, 0x01);
                                k++;
                                ec_send_processdata();
                                */
                                //===========================================
                                // k++;
                                // set_output_with_sdo();
                                //===========================================
                        }   
                        else
                        {
                                slavetop(j);
                                printf("slave again to op\n");
                        }
                }
                else
                {
                        printf("no slave on the bus\n");
                }
        }
        else
        {
                printf("no ethernet card\n");
        }
}
```

The SDO write test function:
```
void set_output_with_sdo()
{
        printf("write SDO\n");
        uint8_t p8 = 0xFF;
        
        int retval = 0;
        retval += ec_SDOwrite(1, 0x7000, 0x01, FALSE, sizeof(p8), &p8, EC_TIMEOUTSAFE);

        while(EcatError) printf("%s", ec_elist2string());
        printf("EL7031 slave %d set, retval = %d\n", 1, retval);
}
```

ArthurKetels, 2020-08-26:
By first looks, nothing wrong. Yes you have yo keep the process data exchange with the slave running for as long as you are in Operational State. When a slave detects no updates for approximately 100ms it will go in fail safe mode and reset all outputs.

Have a look at the sample program red_test to see how that is done. Also I suggest you read the tutorial in the doc folder. This also handles the basic operation principles of SOEM.

LamMork, 2020-08-26:
> By first looks, nothing wrong. Yes you have yo keep the process data exchange with the slave running for as long as you are in Operational State. When a slave detects no updates for approximately [...]

Thanks for your reply.

You said if the slave cannot receive the updates for about 100ms, it will go in fail safe mode, where is this `100ms` comes from? Is it possible for me to set another value for it?

Moreover, when I ran the `./slaveinfo eno2 -map`, the return info told me there are some objects are mapped to the PDO, like:
```
  SM0 RXPDO 0x1600 Channel 9
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1601 Channel 10
     addr b   index: sub bitl data_type    name
  [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
```
But if I ran the `./slaveinfo eno2 -sdo`, I cannot see any thing about these objects. And I can see something like:
```
CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
```
And I cannot set these objects with my function `void set_output_with_sdo()`. Do you know why? Is that because the slave do not support the CoE?

ArthurKetels, 2020-08-26:
Timeout is sometimes a parameter that you can change in the slave, sometimes not. Read the manual. It is up to the manufacturer to decide. 100ms is a general timeout that seems to fit many slaves.

The slave does not support CoE (CoE details value is zero). Therefore SOEM retrieves the mapping from the slave eeprom per SII interface. The giveaway is:

> PDO mapping according to SII :

LamMork, 2020-08-27:
Great, I will go into the detail later. Thank you for your help.

closed 2020-08-27 by LamMork