OpenEtherCATsociety/SOEM #440: SM flag, odd number and random invalid output mapping

stevedanomodolor, 2020-08-28:
This is just out of curiosity, After trying two slaves, one works fine everytime(slave 2) but the the second(slave 3) , always gives an error (invalid iniput mapping) when I use it initially(after a power cycle) without doing any pdo mapping configuration to the slave(using the simple test example) when there are more than one slaves connected. This only happens for this particular slave(slave 3). (This issue is probably related to the slave it self but cannot seem find out where exactly)

Observation: 

1. The invalid output messages appear for random slave when this error occurs
This only occurs when I connect this particular slave with another of the same make or with another with a make but other make work perfectly together when connected with the same make with the simple test code. This is why I thought that the issue might be the slave. 

2. Odd SM value, after looking at the slaveinfo value, I noticed that the flags for the sm2 are different from the typical sm2 flag I have seen in the forum. Slave 2 is the working salve, while slave 3 is the faulty slave. 
Info on the faulty slave 
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  11 F:00010020 Type:4

The only information I could find with respect to the meaning of this flag are this link, which did not help much
[Link to SM info](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_io_intro/4981170059.html&id=5071830807384181920
)

Example 1
```
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x0025 : Invalid output mapping
Slave 2 State=0x12 StatusCode=0x0024 : Invalid input mapping
Slave 3 State=0x12 StatusCode=0x0024 : Invalid input mapping
Slave 4 State=0x12 StatusCode=0x0024 : Invalid input mapping
Slave 5 State=0x12 StatusCode=0x0024 : Invalid input mapping
Slave 6 State=0x12 StatusCode=0x0052 : Unknown

Request init state for all slaves
End simple test, close socket
End program
```

Example 2
Wireshark for this example
[wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5142233/wireshark.zip)
```

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth0 succeeded.
7 slaves found and configured.
Slave in preop state, initicializing data configuration
Slaves mapped, state to SAFE_OP.
segments : 1 : 132 0 0 0
Request operational state for all slaves
Calculated workcounter 18
Not all slaves reached operational state.
Slave 1 State=0x12 StatusCode=0x0025 : Invalid output mapping
Slave 2 State=0x12 StatusCode=0x0024 : Invalid input mapping
Slave 4 State=0x12 StatusCode=0x0024 : Invalid input mapping

Request init state for all slaves
End simple test, close socket
End program
```

Second try 
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on eth0 succeeded.
7 slaves found and configured.
Slave in preop state, initicializing data configuration
Slaves mapped, state to SAFE_OP.
segments : 1 : 132 0 0 0
Request operational state for all slaves
Calculated workcounter 18
Operational state reached for all slaves.
Processdata cycle 5074, WKC 18 , O: 00 00 00 00 00 00 00 00 I: 50 02 fe ff ff ff 00 00 T:3348293733058380195

**Slave:2**
 ```
Name:E/BOX
 Output size: 80bits
 Input size: 184bits
 State: 4
 Delay: 600[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000500 ID: 01107001 Rev: 00000004
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:  10 F:00010024 Type:3
 SM3 A:1180 L:  23 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  10 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000015 Ll:  23 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 17 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
```

**Slave:3**
 ```
Name:Servo Drives
 Output size: 88bits
 Input size: 88bits
 State: 4
 Delay: 1460[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1003
 Man: 0000029c ID: 02c30001 Rev: 00000003
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  11 F:00010064 Type:3
 SM3 A:1c00 L:  11 F:00010020 Type:4
 FMMU0 Ls:0000000a Ll:  11 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000002c Ll:  11 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program
```

ArthurKetels, 2020-08-31:
Sorry, but your wireshark captures do not show the problem clearly. Make a minimal setup that shows the problem, remove all non essentials (slaves, software, configuration etc.). Capture that and post. I have no time to wade through thousands of packets of fluff and than come to the conclusion you captured the wrong data.

nakarlsson, 2020-10-23:
@stevedanomodolor can this be closed?

stevedanomodolor, 2020-10-23:
Yes, for now, thank you very much

closed 2020-10-23 by nakarlsson