OpenEtherCATsociety/SOEM #441: SOEM on BeagleBone AI with linux 5.4 kernel not working

lowkin18, 2020-08-30:
Hello I am trying to get SOEM to work with a beaglebone AI and it seems like the slave never sends any information back.

I also notice that the BBAI adds hw VLAN 802.1Q onto eth0, just wondering if this is the root of my problem or if you have encountered this before.

./slaveinit ./simple_test 
all say ec_init worked but that there are no slaves on the line. 
The same slave will work from my PC using SOEM just not with the Beaglebone AI.

Any insight would be appreciated, Cheers.

ArthurKetels, 2020-08-30:
Perhaps there is a filtering rule active? You could capture a simple_test data transfer between pc and slave. Then remove the slave and connect the BBAI, replay the packets on your PC and capture on your BBAI. All packets from the PC should arrive.

nakarlsson, 2020-08-30:
Is the link realy EN10MB, mentioned in the tcpdump log? Might indicate some HW/cable issue. Should be auto-negotiated to 100MB full duplex in a standard setup.

ArthurKetels, 2020-08-30:
@nakarlsson , no EN10MB just indicates it is sniffing from an Ethernet connection.

lowkin18, 2020-08-30:
The link on the BBAI is capable of 1gb, so I don't know why it says 10Mb in tcpdump, the computer also says 10mb so maybe just a tcpdump thing. 

sudo dmesg shows this from the BBAI 
```
[  152.857089] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  153.874339] cpsw 48484000.ethernet eth0: Link is Down
[  155.921386] cpsw 48484000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[  155.929211] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  186.733134] device eth0 entered promiscuous mode
```

I ran simple_test  from my PC to the BBAI, and it seems like the packets where changed.

This is outbound from the PC 
```
08:10:14.294720 01:01:01:01:01:01 (oui Unknown) > Broadcast, ethertype Unknown (0x88a4), length 29:
0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
0x0010:  0801 0000 0301 0100 0000 0000 00         .............
08:10:14.302588 01:01:01:01:01:01 (oui Unknown) > Broadcast, ethertype Unknown (0x88a4), length 29:
0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
0x0010:  0802 0000 2001 0100 0000 1100 00         .............
08:10:14.310561 01:01:01:01:01:01 (oui Unknown) > Broadcast, ethertype Unknown (0x88a4), length 29:
0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
0x0010:  0803 0000 2001 0100 0000 1100 00         .............
08:10:14.318727 01:01:01:01:01:01 (oui Unknown) > Broadcast, ethertype Unknown (0x88a4), length 30:
0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0e10  ................
0x0010:  0704 0000 0000 0200 0000 0000 0000       ..............
08:10:14.326501 01:01:01:01:01:01 (oui Unknown) > Broadcast, ethertype Unknown (0x88a4), length 30:
0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0e10  ................
0x0010:  0704 0000 0000 0200 0000 0000 0000       ..............
08:10:14.334511 01:01:01:01:01:01 (oui Unknown) > Broadcast, ethertype Unknown (0x88a4), length 30:
0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0e10  ................
0x0010:  0704 0000 0000 0200 0000 0000 0000       ..............
```

this is inbound on the BBAI```
15:10:14.291131 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 60:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
        0x0010:  0801 0000 0301 0100 0000 0000 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000            ............
15:10:14.298969 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 60:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
        0x0010:  0802 0000 2001 0100 0000 1100 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000            ............
15:10:14.306952 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 60:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
        0x0010:  0803 0000 2001 0100 0000 1100 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000            ............
15:10:14.315128 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 60:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0e10  ................
        0x0010:  0704 0000 0000 0200 0000 0000 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000            ............
15:10:14.322939 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 60:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0e10  ................
        0x0010:  0704 0000 0000 0200 0000 0000 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000            ............
15:10:14.330912 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 60:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0e10  ................
        0x0010:  0704 0000 0000 0200 0000 0000 0000 0000  ................
        0x0020:  0000 0000 0000 0000 0000 0000 0000 0000  ................
        0x0030:  0000 0000 0000 0000 0000 0000            ............
```

lowkin18, 2020-08-30:
Seems like simple test on the BBAI will send out 6 packets and on the PC it sends out only 5 packets when running simple test between the BBAI and the PC.

Is this normal that one would be generating an additional packet, the first packet on the BBAI is not seen on the PC
```
15:37:10.301377 01:01:01:01:01:01 > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x88a4), length 29:
        0x0000:  ffff ffff ffff 0101 0101 0101 88a4 0d10  ................
        0x0010:  0801 0000 0301 0100 0000 0000 00         .............
```

lowkin18, 2020-08-30:
The zip contains the sent packets, and the recieved packets.
There are some differences between the sent and received, in bytes 40-44. which may be normal I'll have to look at that later tonight.
[BEAGLEBONE.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5147077/BEAGLEBONE.zip)

ArthurKetels, 2020-08-30:
Looks like the packet reception on the BBAI is just fine. 

I start wondering if the BBAI to slave connection is indeed 100Mb/s?

You could check on the BBAI with the slave connected:
`$ ethtool eth0`

If you have a network switch you can try to put that in-between. The broadcast destination of the packets should make sure it will pass the switch in both directions.

lowkin18, 2020-08-30:
ethtool -eth0

Looks like the link gets setup properly between the slave and bbai

```
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                1000baseX/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                1000baseX/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  100baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 4
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: g
        Wake-on: d
        Current message level: 0x00000000 (0)

        Link detected: yes
```

Kk, I'll try a switch later tonight.
Thanks For the help

lowkin18, 2020-09-01:
I bought a switch and placed the PC, beaglebone and the slave all on it and tested the beaglebone with the slave and got the same behavior.

The PC is able to get the data from the slave.
I am lost at this point. Could it be something isn't being built correctly cause I don't have the right linux headers or dependencies for SOEM?

lowkin18, 2020-09-01:
It also seems like issue 344 was very similar to what I am experiencing. In that dialog there was mention of a bug in the kernel stack stopping a device from correctly reading or sending messages through SOEM.

Just wondering if there is a link for this or more info.

Thanks.

nakarlsson, 2020-09-01:
Post a Wireshark taken from the PC when running SOEM on the beaglebone against the LAN 9252 slave. Eg use the PC as a sniffer.

EtherCAT slaves can handle VLAN, they'll not strip the VLAN tag. If your interface add a VLAN tag 0, I guess it will also strip it and pass the stripped frame to the correct interface.

lowkin18, 2020-09-02:
This is what the bbai outputs for simple test to the slave, PC  used as suggested to read the packets on the switch.

[bbai_simple_test_sniff.zip](https://github.com/OpenEtherCATsociety/SOEM/files/5159546/bbai_simple_test_sniff.zip)

lowkin18, 2020-09-02:
I was able to start the slaveinfo transaction on a RPI4 4gb running 4.19 LTS but somewhere along the way it hung and kicked me out of ssh.

I downloaded the latest image of raspbian which is 5.4LTS and it works perfectly. 

Now I'm wondering what in the BBAI would cause the slave to not respond. Very interesting, any idea whats int he 4.14-4.19 kernel that might be causing issues with SOEM. I know the BBAI 5.4 kernel is a work in progress.

I'll muck about with the kernel config, see if anything sticks.

nakarlsson, 2020-09-02:
or that the BBAI include the VLAN tag while the PC strip it?

nakarlsson, 2020-09-02:
If the PC have captured , bbai_simple_test_sniff.pcap, you should expect to see the response frame from the slave even though the BBAI don't "read it " OK,  if the transmitted frame was indeed OK. 

@ArthurKetels might know more how erroneous frames in slaves are handled, if this one would get discarded. 
Simple CRC errors I think get flagged and forwarded as "broken". But this one might get dropped?

ArthurKetels, 2020-09-02:
Thanks again for the captured data. The last one finally showed the issue. As you mentioned the packet length is different. A minimum ethernet packet is 64 bytes total (as you mentioned) but the NIC strips the FCS (crc) so wireshark would see 60 bytes. 

Your BBAI kernel adds 4 bytes too many. Most OS are not to critical on length but ESCs are. Padding bytes are counted precisely and any deviation will trigger a packet invalidation. 

A similar (but different from this one) issue is described in #203 

So now the hunt is on where in the kernel is the bug?

ArthurKetels, 2020-09-02:
I think I have found the culprit :
https://patchwork.kernel.org/patch/10697191/
No idea how someone could think that simply extending the minimum frame size from 60 to 64 bytes would make sense....
But your escape might be to set the proper module boot parameter "tx_packet_min" to 60.

closed 2020-09-03 by lowkin18

ArthurKetels, 2020-09-03:
Looking a bit further in the history of the TI CPSW driver I think it is still a mess.

For reference a link to the offending patch:
https://git.ti.com/cgit/ti-linux-kernel/ti-linux-kernel/diff/drivers/net/ethernet/ti/cpsw.c?id=0040198aee4e6132cef405621f193ea31d7cb1bc&dt=1

In the current TI git repo in cpsw.c:
```
static netdev_tx_t cpsw_ndo_start_xmit(struct sk_buff *skb,
				       struct net_device *ndev)
{
	struct cpsw_priv *priv = netdev_priv(ndev);
	struct cpsw_common *cpsw = priv->cpsw;
	struct cpts *cpts = cpsw->cpts;
	struct netdev_queue *txq;
	struct cpdma_chan *txch;
	int ret, q_idx;

	if (skb_padto(skb, CPSW_MIN_PACKET_SIZE)) {
		cpsw_err(priv, tx_err, "packet pad failed\n");
		ndev->stats.tx_dropped++;
		return NET_XMIT_DROP;
	}
```

`if (skb_padto(skb, CPSW_MIN_PACKET_SIZE)) {`
Is the offending line where the packet is padded to 64 bytes instead of 60. The argument is that after this there could be an automatic vlan tagging removal causing 4 bytes to be removed from the packet. The resulting packet of 54 bytes would not be ethernet compatible. To me this is a totally broken argument. Frame padding should be done only at the last moment before transmission, not somewhere upstream.

The definition of CPSW_MIN_PACKET_SIZE is in cpsw_priv.h :
`#define CPSW_MIN_PACKET_SIZE	(VLAN_ETH_ZLEN)`

Replace by :
`#define CPSW_MIN_PACKET_SIZE	60`
To kill this nonsense. But it still leaves the bug when removing vlan tags.